micropython/micropython #2485: Prepare for v1.8.5

dpgeorge, 2016-10-07:
It's about time for the next minor release.  @pfalcon anything you want to get in before releasing, #2315 perhaps?

pfalcon, 2016-10-07:
I'm trying to deliver "upip running natively on esp8266" stretch goal, https://github.com/micropython/micropython-lib/issues/105 . It's not read yet, and fairly speaking, it's not clear when it will be "ready", as that task requires integration of many pieces of functionality and deep refactors, like https://github.com/micropython/micropython/issues/2489 or re-uploading all of micropython-lib. I can only propose to postpone 1.8.5 to get it "more ready" rather than "less ready" (a good realistic aim is apparently being able to demonstrate that a single specific package can be installed natively on esp8266).

#2315 is important, and would be next thing to work on (beyond 1.8.5 release).

dpgeorge, 2016-10-10:
We can wait a few weeks for "working" upip, but if it's going to take longer than that then I'd rather tag 1.8.5 now, and 1.8.6 in a month with bare-metal upip support.

pfalcon, 2016-10-10:
I'd rather "soft-launch" it now (i.e. in 1.8.5, without "official" announcement of support) to let interested parties to try it and report experiences.

Basic Zephyr support is merged, though there're more which can be done for it. I added couple of other tickets to the description.

I guess, deadline for the 1.8.5 should be set up and then let people do as much as possible before it.

pfalcon, 2016-10-12:
@dpgeorge : Ping, would be nice to have a deadline.

dpgeorge, 2016-10-13:
@pfalcon did you want to update upip to use ujson.load(stream)?

Otherwise, realistically I won't have time to do the release until early next week, maybe Monday.

pfalcon, 2016-10-13:
@dpgeorge : Nope, sorry, ujson refactor wasn't in the list of dependencies for 1.8.5, and I already spent a lot of time on testing and retesting upip for 1.8.5.

> Otherwise, realistically I won't have time to do the release until early next week, maybe Monday.

Ok, sounds good.

dpgeorge, 2016-10-17:
1.8.5 was tagged in c8d31585a00616b39839d112b7c696dafed6b08d, and esp-extra-scripts was rebased against it.

closed 2016-10-17 by dpgeorge