micropython/micropython #2487: stmhal: SD Clock Always Clocking

ryannathans, 2016-10-07:
During EM compliance testing it was found that the SD Card Clock is always clocking at 24MHz, even whilst just idling on the REPL. 

Radiation peak was at 264Mhz (11th harmonic) on the board I'm working with.

Problem originally occurred on STM32F439 but after testing it seems to also occur on the PYBLITEV1.0 and I suspect all chips covered by stmhal.

Working around it for now by changing the mode of the clock pin before and after SD access.

dpgeorge, 2016-10-07:
Is this only when an SD card is actually inserted?

ryannathans, 2016-10-07:
@dpgeorge apologies for the ambiguity, the line is at 24MHz continuously when SD is inserted prior to powering the device on and after removal of SD whilst device is running. If the device boots with no SD then the clock will sit at 400KHz instead of 24MHz but it still continuously clocks

ryannathans, 2016-10-09:
stmhal/hal/f4/inc/stm32f4xx_ll_sdmmc.h

Line 75

```
  uint32_t ClockPowerSave;       /*!< Specifies whether SDIO Clock output is enabled or
                                      disabled when the bus is idle.
                                      This parameter can be a value of @ref SDIO_Clock_Power_Save           */
```

Line 580

```
/* SDIO Intialization Frequency (400KHz max) */
#define SDIO_INIT_CLK_DIV ((uint8_t)0x76)

/* SDIO Data Transfer Frequency (25MHz max) */
#define SDIO_TRANSFER_CLK_DIV ((uint8_t)0x0)
```

ryannathans, 2016-10-10:
A combination of increasing SDIO_TRANSFER_CLK_DIV and enabling SDIO_CLOCK_POWER_SAVE_ENABLE resolved my issue, I've created a PR for disabling the SD clock when SD isn't in use as a result of my findings.

dpgeorge, 2016-11-07:
Fixed by above patch.

closed 2016-11-07 by dpgeorge