micropython/micropython #2488: stmhal: Easy way how to disable internal Flash storage

prusnak, 2016-10-07:
I would like to have an easy mechanism (build-time option) to disable internal Flash storage so CCM RAM could be used for something else, I'd like to keep SD card storage working.

Thinking more about this I think there could be 3 options altogether:

* (default) both /flash and /sd storages are enabled
* just /sd storage is enabled, CCM RAM could be used for something else
* both /flash and /sd storages are disabled, one has to freeze the modules and compile them into firmware in order to use them (saving lots of space, as filesystem related stuff can be removed from the firmware completely)

Ideas? Thoughts? Pointers how to do this?

chrismas9, 2016-10-07:
+1, but for a different reason. Because it is relatively enormous and has wear levelling SD is more reliable than internal flash. This is important for commercial or industrial systems that do regular files writes. The Flash endurance spec of most MCU's (100k cycles) is one erase per hour for 11 years, or one every 5 minutes for 11 months.

Could I suggest a fourth option? Disable on chip flash and use external SPI Flash (like ESP8266), EEprom or FRAM. These have small erase blocks and only need 256 to 4096 bytes of erase buffer, not 64k. Note also that newer MCU's including STM32L476 have much smaller erase blocks and don't need a 64k buffer.

ryannathans, 2016-10-09:
See #1915  and #1997

prusnak, 2016-10-10:
TBH I don't like stuff in mentioned issue/PR, because they are more hacks than a systematic approach to this wider problem. But thanks for pointing to them as a reference for further research.

dpgeorge, 2016-10-10:
One option is to follow the esp8266 port and have a `_boot.py` frozen script that is executed at the very start of the boot cycle.  This script can be used to configure the filesystem(s) however you like (and do other things if needed).  The drawback with this approach, compared with a compile-time option, is that you still need all the C code for SD card support and internal flash filesystem.

ryannathans, 2016-10-10:
How about ability to disable running code from SD whilst still being able to use it for storage? Or is this a separate issue? This is what I've hackily implemented in main.c and have been using for a while...

dpgeorge, 2016-10-11:
> How about ability to disable running code from SD whilst still being able to use it for storage?

That's related and would be encompased by having a `_boot.py` script to do the first-level initialisation.

dpgeorge, 2021-04-30:
Configuration of flash (both internal and external SPI flash) and SD storage is now fully configurable via:
- `MICROPY_HW_ENABLE_INTERNAL_FLASH_STORAGE`
- `MICROPY_HW_ENABLE_SDCARD`
- `MICROPY_HW_BDEV_IOCTL`
- `MICROPY_HW_FLASH_MOUNT_AT_BOOT`
- `MICROPY_HW_SDCARD_MOUNT_AT_BOOT`

closed 2021-04-30 by dpgeorge