micropython/micropython #2490: esp8266: socket accept() does not always accept

MarkR42, 2016-10-07:
If there is a queue of sockets to accept, accept() does not always clear it, some sockets seem to get "lost", they are still open but accept() will not accept them, it just blocks.

---
Steps to reproduce

I'm attaching two programs - main.py and client.py. Run client.py on a connected system, under normal python3. Run main.py on an esp8266. Supply the address of the esp8266 as a command line parameter.

---
Expected results

Despite the sleep(0.25), the sockets should queue and be accepted eventually.

Output should be, for example:

Socks connected: 4
b'Accept 0'
b'Accept 1'
b''
b''

And running the program >1 time should also succeed.

Actual results:

Socks connected: 4
b'Accept 0'
b'Accept 1'
b''
b''

MarkR42, 2016-10-07:
[socketbug0.zip](https://github.com/micropython/micropython/files/516312/socketbug0.zip)

MarkR42, 2016-10-07:
A little background info - I've written a web server which only serves one request at once. The web browser makes several sockets, and some of them just hang forever, or are closed without being served. So the web browser assumes these are broken and does not load the files.

pfalcon, 2016-10-07:
I may suggest trying to reproduce this issue with the official reference HTTP server example: https://github.com/micropython/micropython/tree/master/examples/network .

MarkR42, 2016-10-07:
Yes, I can reproduce it using the reference HTTP server, with this client program:

`#!/usr/bin/env python3
import socket
import sys

def main():
    if len(sys.argv)<2:
        print("Please supply the address of the ESP8266 as sys.argv[1]")
        raise Exception("Parameter missing")
    socks = []
    for n in range(4):
        sock = socket.socket()
        sock.connect( (sys.argv[1], 8080) )
        socks.append(sock)
    
    print("Socks connected:", len(socks))
    for s in socks:
        s.send(b'GET / HTTP/1.0\r\n\r\n')
        junk = s.recv(1024)
        print(junk)
        s.close()
        
if __name__ == '__main__':
    main()
`

jrobeson, 2016-10-08:
did you try it with micropython unix vs regular python 3?

MarkR42, 2016-10-08:
I can get this example to work with python3 under Linux:

```
try:
    import socket
except ImportError:
    import usocket as socket

try:
    import time
except ImportError:
    import utime as time

def mainloop():
    lsock = socket.socket()
    lsock.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
    lsock.bind( ('', 4242))
    lsock.listen(5)
    counter = 0
    while True:
        asock, addr = lsock.accept()
        print("accepted socket", counter)
        time.sleep(0.25) # Simulate doing some work...
        asock.send(bytes("Accept {}".format(counter), 'ascii'))
        asock.close()
        counter += 1

mainloop()
```

But I can't seem to get socket.bind() to work under Unix-Micropython

pfalcon, 2016-10-20:
> TypeError: object with buffer protocol required

You need to use "socket" module (from micropython-lib) on unix port, not usocket.

Please retry with the latest master build based on SDK2.0, that's a new baseline for anything now. And if you want to have progress with this, feel free to start looking at how lwIP handles your stuff, why "some sockets seem to get "lost", they are still open but accept() will not accept them, it just blocks". (How do you know that sockets are still open - do you judge by client side? Server side (lwIP) may have different opinion.)

MarkR42, 2016-10-21:
Ok, great, I rebuilt it using the latest version of Micropython and the latest SDK.

Results are exactly the same, bug behaves identically. 

The first two sockets accepted, work fine.

The third one, (and any more) gets lost.

Program to reproduce the bug is in socketbug0.zip

The sockets are still open on the client side, a network trace shows no FIN, RST packet etc, but the server doesn't respond and accept() won't accept them.

pfalcon, 2016-10-21:
What import port_diag says regarding PCBs? Look up which lwip module function it calls for quicker access. Then feel free to build lwip in esp-open-sdk yourself and look what debugging settings to enable to get more insight into what happens. That's again if you're interested to have progress of this bug in your hands ;-). Otherwise, it'll have to wait its turn. (And I somehow don't understand if any (basic) real-world functionality affected by this, actually, I know that nothing should be affected, like running http_server sample should work well with a real browser, etc.)

pfalcon, 2016-11-05:
> Yes, I can reproduce it using the reference HTTP server, with this client program:

And I cannot reproduce, using the standard HTTP testing tools, not adhoc clients. Slightly modified http_server.py example is used (with no output, which makes it much harder for esp8266 - slow UART output otherwise serves as a natural (slow again) flow control).

Here's result of running apache bench:

~~~
$ ab -n20 -c2 http://192.168.1.36:8080/
This is ApacheBench, Version 2.3 <$Revision: 1528965 $>
Copyright 1996 Adam Twiss, Zeus Technology Ltd, http://www.zeustech.net/
Licensed to The Apache Software Foundation, http://www.apache.org/

Benchmarking 192.168.1.36 (be patient).....done

Server Software:        
Server Hostname:        192.168.1.36
Server Port:            8080

Document Path:          /
Document Length:        24 bytes

Concurrency Level:      2
Time taken for tests:   2.315 seconds
Complete requests:      20
Failed requests:        0
Total transferred:      820 bytes
HTML transferred:       480 bytes
Requests per second:    8.64 [#/sec] (mean)
Time per request:       231.457 [ms] (mean)
Time per request:       115.728 [ms] (mean, across all concurrent requests)
Transfer rate:          0.35 [Kbytes/sec] received

Connection Times (ms)
              min  mean[+/-sd] median   max
Connect:        3    5   1.7      5      10
Processing:    11  220  67.6    245     287
Waiting:       11  220  67.6    245     287
Total:         14  225  68.2    249     292

Percentage of the requests served within a certain time (ms)
  50%    249
  66%    252
  75%    253
  80%    254
  90%    268
  95%    292
  98%    292
  99%    292
 100%    292 (longest request)
~~~

What you can see here is that single-threaded http_server.py is bombarded with requests of concurrency 2. That means that 2 connections are opened concurrently, and while is accepted immediately, one necessarily waits in listen queue to be accepted later. As you can see, this works reliably.

Trying to increase concurrency to 3 will throw ab off due to its tendency to die on first error:
~~~
$ ab -n20 -c3 http://192.168.1.36:8080/
Benchmarking 192.168.1.36 (be patient)...apr_socket_recv: Connection reset by peer (104)
~~~

The outcome is however expectable: yet another request doesn't fit into queue, and client gets connection reset. This queue length is not what is being passed to listen (5), but lwIP, just as Linux, can have its own opinion on what listen queue should be and ignore user's figures (that all is tweakable in config).

siege is better tool in this regard, also because it has delays between requests (increading -n for ab will make esp8266 due to well-know TCP TIME_WAIT issue). let's try 100 request with concurrency 5. We'll need to increase delay dispersion to be in range 1-3s to avoid running out of memory:

~~~
$ siege -r100 -c5 -d3 -v http://192.168.1.36:8080/
...
Transactions:		         494 hits
Availability:		       98.80 %
Elapsed time:		      160.98 secs
Data transferred:	        0.01 MB
Response time:		        0.03 secs
Transaction rate:	        3.07 trans/sec
Throughput:		        0.00 MB/sec
Concurrency:		        0.11
Successful transactions:         494
Failed transactions:	           6
Longest transaction:	        1.01
Shortest transaction:	        0.00
~~~

As you can see, under conditions which for a tiny device like esp8266 can be called "crushing", it (together with MicroPython's lwIP socket implementation) do pretty well, merely 6 requests of 500 were dropped.

@MarkR42 , to proceed, you'd need:

1. Reproduce my "lack of problems" setup.
2. See what are the differences between my and your setup and which of those differences cause problems.

I will post the exact version of test server script. Please remind me if I don't.

belyalov, 2018-01-06:
Confirm, does not always accept. Moreover, instead of accepting - ESP8266 issues RST.
I've got stable repro of this issue:

### Steps to reproduce
1. Use latest official esp8266 image (1.9.3 at this moment)
2. Run [official client](https://github.com/micropython/micropython/blob/master/examples/network/http_server.py) on ESP8266
3. Open few TCP connections simultaneously, e.g:
```bash
wget http://192.168.1.63:8080/ &
wget http://192.168.1.63:8080/ &
wget http://192.168.1.63:8080/ &
wget http://192.168.1.63:8080/ &
```

### ESP output:
```
Bind address info: [(2, 1, 0, '', ('0.0.0.0', 8080))]
Listening, connect your browser to http://<this_host>:8080/
Client address: ('192.168.1.33', 60097)
Client socket: <socket state=2 timeout=-1 incoming=3fff9360 off=0>
Request:
b'GET / HTTP/1.1\r\n'
b'User-Agent: Wget/1.19.2 (darwin16.7.0)\r\n'
b'Accept: */*\r\n'
b'Accept-Encoding: gzip\r\n'
b'Host: 192.168.1.63:8080\r\n'
b'Connection: Keep-Alive\r\n'

Client address: ('192.168.1.33', 60098)
Client socket: <socket state=2 timeout=-1 incoming=3fff93a0 off=0>
Request:
b'GET / HTTP/1.1\r\n'
b'User-Agent: Wget/1.19.2 (darwin16.7.0)\r\n'
b'Accept: */*\r\n'
b'Accept-Encoding: gzip\r\n'
b'Host: 192.168.1.63:8080\r\n'
b'Connection: Keep-Alive\r\n'

Client address: ('192.168.1.33', 60101)
Client socket: <socket state=2 timeout=-1 incoming=3fff9340 off=0>
Request:
b'GET / HTTP/1.1\r\n'
b'User-Agent: Wget/1.19.2 (darwin16.7.0)\r\n'
b'Accept: */*\r\n'
b'Accept-Encoding: gzip\r\n'
b'Host: 192.168.1.63:8080\r\n'
b'Connection: Keep-Alive\r\n'

Client address: ('192.168.1.33', 60102)
Client socket: <socket state=2 timeout=-1 incoming=0 off=0>
Request:
b'GET / HTTP/1.1\r\n'
b'User-Agent: Wget/1.19.2 (darwin16.7.0)\r\n'
b'Accept: */*\r\n'
b'Accept-Encoding: gzip\r\n'
b'Host: 192.168.1.63:8080\r\n'
b'Connection: Keep-Alive\r\n'
```
So at a glance we've connections from 60097, 60098, 60101, 60102.

### Client side
However, client was tried to make more connections and a few of them have been rejected, see attached picture:

[image]

Bottom line: connections from ports 60099, 60100 have been rejected.

curlyz, 2018-08-03:
socket.accept() will block.
The correct way to use it is 
setup : 
socket.settimeout(0)

now , when the accept() is called , after a period of time , if nothing connect , it will raise ETIMEOUT
catch it like nothing happen 
.
note that socket.accept() only return after a device connected to its AP and also after access a local address !

pacmac, 2018-11-08:
Hi;

I can verify that there does still appear to be a problem on the esp8266 that does not appear to be present on the esp32.

https://forum.micropython.org/viewtopic.php?f=16&t=5505&p=31747#p31745

See the post, I have created a simple web server with index.htm file and included css files, on the esp8266 only 2 or 3 of these attachments are ever loaded and the rest are reported by chrome as "net::ERR_CONNECTION_RESET", however on the esp32 all of them are loaded every time, and this problem does not exist.

It does in fact appear like not all requests are reaching accept() and that the esp8266 socket only accepts up to 3 connections and then drops the rest instead of queueing them ?

Just to reiterate, identical code running on the esp32 does not exhibit this problem.

dpgeorge, 2018-11-28:
It's possible this is fixed by #4322

dpgeorge, 2018-12-01:
> It's possible this is fixed by #4322

The very first socktebug0.zip posted above is now fixed, but only superficially: it now works because the backlog of 5 set in main.py is honoured and the 4 connections made by the client are now queued and eventually processed.  But increasing the number of simultaneous connections made by the client to more than the backlog setting in main.py will again show the error that occurred before (eg backlog of 2, with 10 simultaneous connections).

The error does not happen happen on esp32, which uses the top-level lwIP socket API.  So that points to a problem with `extmod/modlwip.c` and its implementation of listen/accept on top of the raw TCP API.  And it looks like it is tied to the lack of use of the `TCP_LISTEN_BACKLOG` option, and `tcp_backlog_delayed()` and `tcp_backlog_accepted()` functions.  The client seems to think that connections are accepted immediately, when in fact the host just drops them when the backlog queue is full.