micropython/micropython #2491: add global symbol `__port__` for conditional compilation

blazewicz, 2016-10-07:
Consider an IoT software targeted to stmhal, cc3200 and esp8266 devices. They all need different code for network initialization and for some peripheral actions, some may lack a feature that others have etc.

Currently we have to either maintain separate initialization scripts for different ports or use `ImportError` tricks - first solution isn't too handy and the second one results in garbage bytecode in device's memory.

Another example may be a test suite which we would like to run on our machine or in some CI tool.

My proposition is to add a special symbol named `__port__` which would hold current (or target when cross-compiling) ports name (ex. `'__stmhal__'`, `'__unix__'`, `'__esp8266__'` etc.)

Then we could write something like:

```Python
if __port__ == '__stmhal__':
    # this part will be compiled only for stmhal port
elif __port__ == '__esp8266__'
    # only for esp8266
else:
    # and this for any other
```

For this to work we would also need to add expression evaluation for constant operands in compile time so unused blocks aren't compiled.

Same conditional compilation technique is used with `__debug__` symbol:
```Python
if __debug__:
    # compile only when optimizations are disabled
```

Or with constant expressions:
```Python
if False:
    # skip this block
elif 0:
    # this one too
else:
    # this block will be compiled
```

deshipu, 2016-10-07:
What's wrong with ``sys.platform``?

blazewicz, 2016-10-07:
> What's wrong with `sys.platform`?

My lack of knowledge about it apparently.

Despite the fact that such symbol already exists, mentioned `if-elif-else` tree would still require runtime evaluation rather than provide conditional compilation like with constants.

blazewicz, 2016-10-07:
What I mean here is some form of equivalent to C's `#if` blocks.

deshipu, 2016-10-07:
I can't help but wonder... before we reimplement the whole C preprocessor, why not run your python code through the original C preprocessor before you upload it to the board?

dhylands, 2016-10-07:
And a quick google search found a python preprocessor:
http://stackoverflow.com/questions/482014/how-would-you-do-the-equivalent-of-preprocessor-directives-in-python

blazewicz, 2016-10-07:
> I can't help but wonder... before we reimplement the whole C preprocessor, why not run your python code through the original C preprocessor before you upload it to the board?

Because `cpp` looks for its directives on every line which first non-white character is '#' which also happens to be Python comment.

> And a quick google search found a python preprocessor:
http://stackoverflow.com/questions/482014/how-would-you-do-the-equivalent-of-preprocessor-directives-in-python

Yes, you see the first answer? `if __debug__:` or more generally `if <constat expression which evaluates to false>:` are only, native ways to achieve what I'm talking about.

This is not such an issue in "big" Python implementations because no one gets ever concerned about size of scripts bytecode in the memory. I am not looking for a general preprocesor-like solution but a way to define target specific blocks in a least invasive way.

I do understand that this is not very pythonic and I'll be just fine if you disagree with me on this RFC because this is what this issue realy is.

dpgeorge, 2016-10-10:
> Consider an IoT software targeted to stmhal, cc3200 and esp8266 devices. They all need different code for network initialization and for some peripheral actions, some may lack a feature that others have etc.

Ideally we'd have a consistent API (ie the machine module, network module, etc) that works the same across all ports.

And I'd say that in most cases it's probably better to do runtime selection based on features that a module/class does or does not support (eg does the SPI class have the readinto method or not?), rather than on select based on which port/board you're on.

For things that are really port specific you could either put it in a separate script/module (eg `import port_setup`), or create a wrapper/abstraction layer (eg around the network module) that's tailored to your application and provides the same API for all your targets.

Otherwise, if you want compile-time selection of code, uPy already has good support for constant folding and dead-code elimination in the parser/compiler.  Eg you can already do this:
```python
from micropython import const

# available boards
_BSTMHAL = const(1)
_BESP8266 = const(2)

# current board
_BOARD = const(_BSTMHAL)

if _BOARD == _BSTMHAL:
    # stmhal specific code
elif _BOARD == _BESP8266:
    # esp8266 specific code
```
It's not perfect, you'll still need to manually edit the `_BOARD` variable, but it does do proper dead-code elimination.

As pointed out above, sys.platform already exists, along with sys.version, sys.version_info, sys.implementation and os.uname() to tell you what platform/port/board you are on.  It's possible to make the compiler recognise sys.platform as a constant, and get it to do constant folding on strings (eg str1==str2), and if this was done then `if sys.platform == 'pyboard': ...` would be the best way to do compile-time selection of code (and would even be compatible with CPython).

blazewicz, 2016-10-17:
Thanks @dpgeorge, that is one satisfying answer.

> It's possible to make the compiler recognise sys.platform as a constant, and get it to do constant folding on strings (eg str1==str2), and if this was done then if sys.platform == 'pyboard': ... would be the best way to do compile-time selection of code (and would even be compatible with CPython).

And this is exactly what I was thinking about, but I'm just throwing the idea.

dpgeorge, 2022-06-23:
Closing due to inactivity.

closed 2022-06-23 by dpgeorge

DrSensor, 2023-01-20:
Does micropython now recognize `sys.platform` as a constant and perform constant folding?

jimmo, 2023-01-23:
> Does micropython now recognize `sys.platform` as a constant and perform constant folding?

@DrSensor No. e.g. you cannot write `if sys.platform == 'linux':` and have the if optimised out. (That would be neat though!)