micropython/micropython #2492: ESP8266 List Comprehensions: TypeError: 'int' object is not an iterator

dothacker495, 2016-10-08:
I've been experiencing this _(TypeError: 'int' object is not an iterator)_ weird error using List comprehensions.

I used this [`usched.py`](https://github.com/peterhinch/Micropython-scheduler/blob/master/usched.py)  for cooperative multi-tasking.

I used this ['example'](https://github.com/peterhinch/Micropython-scheduler/blob/master/subthread.py) to try the scheduler out but unfortunately it raises

> TypeError: 'int' object is not an iterator

on list comprehensions like  `self.lstThread = [thread for thread in self.lstThread if thread[STATE] != DEAD] ` where self.lstThread is a List initialized with `self.lstThread = []`

peterhinch, 2016-10-08:
A number of people have recently reported issues with comprehensions and frozen bytecode. These have been fixed by pulling and compiling the latest version of the cross-compiler. I pulled the latest firmware, rebuilt the cross-compiler and the ESP8266 firmware, erased the ESP flash and installed. My application runs with no problems around these list comprehensions.

pfalcon, 2016-10-22:
I assume suggestion to rebuild mpy-cross helped. Please not that frozen bytecode is experimental feature, and users are expected to be prepared to learn inner detail of its working and be prepared to face and resolve issues. Closing, feel free to reopen if issue persists.

closed 2016-10-22 by pfalcon