micropython/micropython #2493: EPS8266: No automatic reconnect after WiFi activation

puuu, 2016-10-08:
Assuming the ESP8266 is connected to an AP. Then I do:

```
wlan = network.WLAN(network.STA_IF)
wlan.active(False)
# some tasks
wlan.active(True)
```

After that, there is no automatic reconnect to the AP.  Running `wlan.connect(ESSID, PASSWD)` work as expected, but requires the WiFi credential.  An other posibillity is resetting the ESP8266. During boot a reconnect to the AP is performed. Therefore, I wonder if there is a possibilty to reconnect to the AP without providing the Wifi credentials.

I think, it would be nice to have a automatic reconnect after `wlan.active(True)` or the possibility to call `wlan.connect()` without parameter.

I need this for a solar charged battery powered Device. Before entering deep-sleep mode, I deactivate WiFi. On every wake up, I check the battery state before 	possibly activating WiFi. Unfortunately, this requires to run `wlan.connect(ESSID, PASSWD)` with WiFi credential.

pfalcon, 2016-10-11:
Connection and reconnection works the way Espressif designed it. We aren't in the position to change that - yet, or at all.

puuu, 2016-10-14:
@pfalcon I can understand, why you are not willing to change the behaviour of `wlan.connect()` (#2506), but I do not understand, why you mention:
> We aren't in the position to change that - yet, or at all.

As my pull request in #2506 shows, the missing part is a possibility to run the C function `wifi_station_connect()` from python.

So please remove the “wontfix” label and let discuss how to force a reconnection from python.

Also with a look to #2510, there is even one more reason to have the possibility to reconnect (after `wlan.active(False);wlan.active(True)` or `wlan.disconnect()`) to the previous AP without running `wlan.connect(ESSID, PASSWD)` which overwrite the previous credentials in flash.

commit b97c17e1 (2016-11-01): esp8266/modnetwork.c: Allows AP reconnection without WiFi credentials

There is no automatic reconnect after wlan.active(False);
wlan.active(True). This commit provide the possibility to run
wlan.connect() without parameter, to reconnect to the previously
connected AP.

resolve #2493
closed 2016-11-01 by pfalcon