micropython/micropython #2504: Lazy * operator for bytes

deshipu, 2016-10-11:
I'm just throwing a crazy idea here for discussion. It's something that is definitely not compatible with Python, but could possibly be very useful in MicroPython. I am not entirely sure how feasible it is in practice, though. Any comments welcome.

The "*" operator (multiplication), when applied to strings and bytestrings, will normally create a new object with the string or bytes repeated. I wonder if it would be possible to instead return a proxy object, similar to the "reversed" or "memoryview" objects, that wouldn't actually use much extra memory, and instead pointed to the original string repeatedly, otherwise behaving like a normal string.

The idea came to me initially when I was writing code for clearing the screen for one of the display drivers I wrote, but it could actually be generally useful in many cases. There is a thread on the forum with one example: http://forum.micropython.org/viewtopic.php?f=2&t=2480

peterhinch, 2016-10-11:
Use a generator?

deshipu, 2016-10-11:
Well, that would solve half of the problem, if only slice assignment worked with generators, I guess. I even proposed that in that forum thread. But looking at my example, you can see it's still quite a mouthful, and you are possibly still allocating a range object (not sure if MicroPython optimizes that away in generator expressions too).

pfalcon, 2016-10-11:
I'm not sure how useful to open such speculative/forward-looking tickets at the current state of the project, when we have 250+ open "real" tickets. We have "development" forum just for such discussions: http://forum.micropython.org/viewforum.php?f=3 , and indeed, similar matters were discussed more than 2 years ago there: http://forum.micropython.org/viewtopic.php?f=3&t=103&p=351 .

deshipu, 2016-10-11:
I'm very sorry for not reading the complete archives of all the forums.

deshipu, 2016-10-11:
Should we also move all the "RFC" issues from this bug tracker to the forum, if that's a better place for them?

pfalcon, 2016-10-11:
I doubt that (why waste energy on bureaucracy if there're useful things which can be done instead). I personally just refrain from posting far-fetched RFC which, based on 3-years experience, aren't going to be acted on anytime soon. Don't feel that you need to close this ticket or something, what I'm trying to communicate is that, paraphrasing a known aphorism, "all good things were already discussed and considered, one just needs to implement them". But 2 years after those discussions, we're at the stage when Hardware API isn't finalized and not implemented consistently across supported ports. So, something's needed to be done first, something next. You don't need to agree on prioritization either, but then if it comes to actual patches, don't be surprised that a maintainer gives -1 because "we aren't yet ready for that" or "this goes sideways to the current direction of development, maybe later". (I can also refer to slide 27 of http://www.slideshare.net/linaroorg/las16108-jerryscript-and-other-scripting-languages-for-iot)

deshipu, 2016-10-11:
That's exactly the reason why I posted a question first, before I actually began to write any code. And I posted it here, because I have seen those other RFCs posted here, and assumed it's the right place for a development-related discussion. Of course I was meaning to work on it, if the discussion had any positive outcomes.
I don't have your three years of experience, and this particular feature looked like a relatively small feature, that could nevertheless open the way to more such optimizations in the future -- so I decided to ask, both to see if it indeed would make sense, and to confirm my assessment of its difficulty, which, due to lack of experience, could possibly be way off. Unfortunately so far I got neither.
As for the Hardware API, I'm not sure if this is the reason why others don't want to work on it, but for me personally many details in it just feel wrong, and since I don't really have much power to change them, I will just keep away and let people who feel this design better do it. I suspect that this wrong feeling is the result of my lack of experience and connection with this project, as some of it started to feel slightly better over time, but I'm still feel like I would just twist the design and corrupt it while trying to implement it despite my feelings of wrongness, just as I did with the hardware SPI api for the esp8266. Thus, I will leave that part to the originators of the design, and instead work on things that I think I understand better.

dpgeorge, 2016-10-12:
A lazy * operator would break CPython compatibility, so it's a no-go.  Instead it would be better to propose a function (but I'm not endorsing such a thing).

> As for the Hardware API, I'm not sure if this is the reason why others don't want to work on it, but for me personally many details in it just feel wrong

Are your concerns about the overall design, or small details of various functions/methods?  I know you'd like i2c.start to send the address.  What are other problems?  [Answer to this should probably go in the hardware API issues.]

closed 2021-04-02 by deshipu