micropython/micropython #2508: Introduce machine.soft_reset()

pfalcon, 2016-10-12:
As discussed in https://github.com/micropython/micropython/pull/2495

dpgeorge, 2016-10-13:
It's not that easy to implement.  Simplest thing would be to do what was previously done with SystemExit: machine.soft_reset() would set a special flag saying that there should be a soft reset, and then it would raise SystemExit.  Assuming the script does not intercept the exception, it would propagate to the pyexec function, which needs to then return another special value (like it did before) to indicate to the caller that SystemExit was raised.  Then finally the caller would check the soft reset flag and do a soft reset.

pfalcon, 2016-10-13:
So, do we want to leave this for after 1.8.5?

dpgeorge, 2016-10-13:
I'd rather not have this feature missing (at least for pyboard/stmhal) in a release.  I'll try to add something that at least works, and we can have further discussion later if needed.

dhylands, 2016-10-13:
It isn't clear to me what the desired sys.exit() behaviour should be (aside from what it currently is) for something like the pyboard. I think that under unix, it should quit the app.

pfalcon, 2016-10-13:
Same for any other port.

dhylands, 2016-10-13:
> Same for any other port.

which is what exactly?

There are 2 different classes of machines here, one where you have the ability to exit the process (i.e. unix) and one where you don't (i.e. the pyboard doesn't have the notion of a process).

pfalcon, 2016-10-13:
https://github.com/micropython/micropython/pull/2495#issuecomment-252588226

dhylands, 2016-10-13:
> #2495 (comment)

Actually, under linux, sys.exit() doesn't "drop to command line" unless the caller happens to be a shell.

And since there is no command line on the pyboard what exactly does "drop to the command line" mean?

pfalcon, 2016-10-13:
It means the behavior of the current micropython master.

dhylands, 2016-10-13:
Now I'm totally confused. You want to change the behaviour of sys.exit() to be what it currently is?

dpgeorge, 2016-10-17:
The point is that sys.exit() (and hence raise SystemExit) should just "stop the running code" / "quit the app", and shouldn't lead to extra behaviour like restarting the board and rerunning start-up scripts, for example.  This distinction was not really a problem with pyboard, but with the esp8266 it can't properly handle a soft reset, and soft-resetting in WebREPL doesn't work at all (because the start-up scripts initialise the WebREPL).

As such, we need to make the behaviour of SystemExit more fine-grained / decomposed: the exiting of the script/app should be independent to an actual soft reset.  So SystemExit will now just finish the script (and drop to a REPL on bare-metal, since there's not much else it can do).

See ad3724e0bc87305f9280f65226066a199042d736 and 57226a2b7f91bd53f0bf392de713be937ede3f5f for implementation of machine.soft_reset() in stmhal.  We can factor that out to all ports when we agree on the behaviour (and keep this ticket open until then).

gengshenghong, 2018-12-20:
Agree to introduce soft_reset for esp32. 👍

Thanks,
Shenghong

dpgeorge, 2020-01-12:
`machine.soft_reset()` was added to the docs in 1caede927ab946e989946d78878934a15ed3d487, and implemented on esp32 in 6632dd3981cbdca8e70c0bf19e36338101e9ce0e

closed 2020-01-12 by dpgeorge