micropython/micropython #2509: machine.SPI(): add CS/NSS pin

dmartauz, 2016-10-12:
It would be nice if CS/NSS pin for SPI could be driven by machine.SPI() itself. Manual toggling in user application slows it down.

deshipu, 2016-10-12:
I know that ESP8266 has support for it a single CS pin on gpio15. The problem is that with multiple devices on the bus, you want to have multiple CS pins, different for every device.

Davidus55, 2018-11-19:
I agree with @dmartauz. Can you add the CS configuration on the machine.SPI() itself.
I am using a DAC without data buffer. A flag is set when the data is ready and rewrite.
My problem is that between the CS and the read data I have around 100µs and the flag data is activated four times.

peterhinch, 2018-11-19:
I am with @deshipu here. On the Pyboard CS is user defined for the reason he gives. I can see merit in doing it in C, but the user should be able to disable it for the common case of multi-device use.

Davidus55, 2018-11-19:
I agree with you, but we can change the CS at all moment like this "spi.init(CS=Pin(xx))".

With these two lines code :
`
cs.value(0)
buf = global_variable.spi2.read(12)
`

I have this result:

[image]

As I explained, my problem is that the data ready flag could occur when I am reading the data.
And without buffer inside of the DAC the value could be wrong.
The DAC is an ADS131A02 but there is nothing for disable the conversion during the read time.

deshipu, 2018-11-19:
How does using hardware CS solve that? It only makes the time shorter, but doesn't make it disappear completely, so you will still get random wrong values.