micropython/micropython #2515: donating/sponsoring micropython (query from irc)

jrobeson, 2016-10-14:
``<sorseg> Hey! I fail to find any donation button. Is bying a pyboard the only way to sponsor micropython development?```

kamikaze, 2016-10-15:
you could buy all the PyBoards and then send to others for free if you don't need them ) As I see - we need to spread micropython by spreading pyboards. Not esp8266 or whatever, but a real PyBoard. This is a go in a long-term. Then there will be more involved people. Also you could create shields for PyBoard. Also you could contribute your code to help the development. As I see it now - there is a lack of hands.

dpgeorge, 2016-10-18:
I'll try to get a "donate" button on https://micropython.org very soon!

jrobeson, 2016-10-18:
too bad I don't know how to contact sorseg to point them to this ticket :(  I hope I'll see them again on irc.

dpgeorge, 2020-06-11:
MicroPython now has GitHub sponsoring enabled, see here: https://github.com/sponsors/micropython

closed 2020-06-11 by dpgeorge