micropython/micropython #2517: Unable to freeze modules for PyBoard

kamikaze, 2016-10-15:
I've added modules dir with files to freeze, launched make command... and there is nothing frozen.

kamikaze, 2016-10-15:
I'm tired of that I can't even figure out why Tony D from Adafruit is able to freeze modules for esp8266 and I'm not (for PyBoard). This is not even developer friendly to have missing part of Makefile (as I understand - because there is no such modules in the default build). 
So here it is: #2518

robert-hh, 2016-10-16:
You have to place your scripts in a subdirectory of stmhal, e.g. modules. Then call:
`make FROZEN_MPY_DIR=modules` 
You can also add a line to Makefile like:
`FROZEN_MPY_DIR ?= modules`
which sets the default to modules. Doing that, you can also set the board type default to what you have at you desk, like PYBV11.

dhylands, 2016-10-16:
Rather than edit Makefile, you can instead create a file called GNUmakefile and put the following in it:
```
BOARD = PYBV11
$(info BOARD = $(BOARD))
FROZEN_MPY_DIR = modules
$(info FROZEN_MPY_DIR = $(FROZEN_MPY_DIR))
include Makefile
```

Since GNUmakefile isn't under source code control, there won't be any issues when trying to update your source tree.

I find the prints to be useful to let me know that I've overriden defaults.

kamikaze, 2016-10-16:
all this is fine, but still takes time to figure out what extra actions to take. And to get this working - everyone should do it. I find esp8266 way to be more "user" friendly. So they can concentrate on their tasks, not on fighting with black magic

kamikaze, 2016-10-16:
ok. I will add an override support too, but later.

kamikaze, 2016-10-16:
ok, added defaults for FROZEN_DIR and FROZEN_MPY_DIR if not set. Also printing out actual values.

kamikaze, 2016-10-16:
btw. What about an automatic cross-mpy build when there are modules/scripts to be frozen?

robert-hh, 2016-10-16:
Thanks Dave. The hint about GNUmakefile is very helpful. It saves me from keeping an own branch just for setting my favorite defaults in Makefile.

robert-hh, 2016-10-16:
> What about an automatic cross-mpy build when there are modules/scripts to be frozen?
That was also a thread in the ESP8266 branch of the MicroPython board. No conclusion so far as I can tell.

kamikaze, 2016-10-16:
but will it rebuild each time? I thought make command will just pass this step if it already has been built and not changed

robert-hh, 2016-10-16:
If nothing it depends on changed, then it will indeed not be rebuilt. I'll try to put it in this magic GNUmakefile.

pfalcon, 2016-10-20:
Per https://github.com/micropython/micropython/pull/2518#issuecomment-255247036 , frozen modules handling for stmhal was made consistent and verified to work.

closed 2016-10-20 by pfalcon