micropython/micropython #2526: Hard to find the appropriate source code for documentation

kamikaze, 2016-10-17:
As an example: there is an [outdated SPI documentation](http://docs.micropython.org/en/latest/pyboard/library/machine.SPI.html) and turns out that is an outdated WiPi doc. Isn't it a better idea to provide a link to the source code file/files right in documentation? There are at least two pros:

1. even if a document is "wrong" - you can figure out this by opening a wrong source code for a wrong port.

2. you can find a right source code fast, find a difference and update documentation also fast.

deshipu, 2016-10-17:
You have a "View page source" link in the top right corner of every page.

kamikaze, 2016-10-17:
this is about PAGE source, but I'm talking about SOURCE code )

deshipu, 2016-10-17:
There is no 1-1 correspondence between the source code and the documentation.

kamikaze, 2016-10-17:
ok, but at least if there is a class name in documentation - isn't it possible to link it to the correct source file?

deshipu, 2016-10-17:
This is the class definition for SPI: https://github.com/micropython/micropython/blob/master/stmhal/spi.c#L1125-L1132

Not very helpful, I'm afraid.

deshipu, 2016-10-17:
By the way, I created a number of display drivers for MicroPython, perhaps my code will be useful for you: https://bitbucket.org/thesheep/micropython-display/src

kamikaze, 2016-10-17:
wow, that's a lot :) don't you want to make a fork and place them into drivers directory and then ask for PR? :)

deshipu, 2016-10-17:
No, I don't.

pfalcon, 2016-10-17:
> As an example: there is an outdated SPI documentation

What's exactly outdated? If you'd like to assist with it being fixed, please submit a separate ticket with exact and objective information about it. (Some background information on effective bug reporting: http://www.chiark.greenend.org.uk/~sgtatham/bugs.html).

For general discussions and opinion exchanges, we have a forum instead: http://forum.micropython.org

kamikaze, 2016-10-17:
That is what exactly outdated: http://forum.micropython.org/viewtopic.php?p=14826#p14826

pfalcon, 2016-10-17:
Ok, that at least partially addressed by https://github.com/micropython/micropython/commit/a0b2f48c2f501fb60ec312756fe4b17e81f94cc5 . It will be on docs.micropython.org tomorrow.

closed 2016-10-17 by pfalcon