micropython/micropython #2527: ESP Build broken?

mkarliner, 2016-10-17:
I've just tried a new ESP build for the first time in a couple of weeks.
I'm getting the following error:

xtensa-lx106-elf-ld: build/firmware.elf section `.text' will not fit in region `iram1_0_seg'
xtensa-lx106-elf-ld: region `iram1_0_seg' overflowed by 152 bytes
make: *** [build/firmware.elf] Error 1

Reading previous issues, the last time this occurred, it was caused by an
out of data compiler, so to be sure, I've just pulled down a fresh clone of
the compiler and built that, and I'm still getting the error.

I have the following list of frozen modules, but removing say,
uasyncio has not changed the size of the overflow.

MPY modules/_boot.py
MPY modules/ds18x20.py
MPY modules/encoder.py
MPY modules/flashbdev.py
MPY modules/logging.py
MPY modules/onewire.py
MPY modules/robust.py
MPY modules/sh1106.py
MPY modules/simple.py
MPY modules/ssd1306.py
MPY modules/test.py
MPY modules/uasyncio.py
MPY modules/webrepl.py
MPY modules/webrepl_setup.py
Creating build/frozen_mpy.c
CC build/frozen_mpy.c

mkarliner, 2016-10-17:
Just tried with a newly checkout version of MPY, with no additional modules. Same result
Build platform OSX - El Cap

pfalcon, 2016-10-17:
Yes, upgrade to SDK2.0 started: http://forum.micropython.org/viewtopic.php?f=16&t=1689#p14850 , the port will be unstable for some time.

pfalcon, 2016-10-18:
master is buildable again. The latest esp-open-sdk is mandatory.

closed 2016-10-18 by pfalcon