micropython/micropython #2528: More finegrained section control for esp8266 and in general

pfalcon, 2016-10-17:
We're running (have run out with SDK2) of IRAM space. At the same time, a lot of code can be moved out of it actually. But doing that on a per-file basis doesn't scale. So, I'd like to introduce generalized MP_FASTCODE modifier, along the lines of existing MP_WEAK, MP_NOINLINE, etc.

However, if it's defined just as:
~~~
#define MP_FASTCODE __attribute__((section(".iram0.text")))
~~~

The ability to GC unused MP_FASTCODE functions is lost. To retain that ability, MP_FASTCODE should be defined as function name modifier:

~~~
#define MP_FASTCODE(n) __attribute__((section(".iram0.text." #n))) n
void MP_FASTCODE(write_buf)(void) {}
~~~

This makes function definition look a bit funky, but given the functionality it provides, should be ok. Alternative opinions?

pfalcon, 2016-10-17:
@dpgeorge: ?

dpgeorge, 2016-10-18:
> The ability to GC unused MP_FASTCODE functions is lost.

Do we need that ability?  Isn't it enough to just have good #if control around selective code, like we already do?

If you do want to go for a macro that wraps the function name, how about `MP_FASTFUN`?

pfalcon, 2016-10-18:
> Do we need that ability? Isn't it enough to just have good #if control around selective code, like we already do?

We have code complexity where we don't already, and shouldn't need to have, that for every function. For example, if some utility function can be used by MICROPY_PY_STR_FIND, MICROPY_PY_UZLIB, and can be used any time by another MICROPY_PY_STR_*, there's no point of deducing, and maintaining, complex #if conditions if that can be automatically done by the linker.

> If you do want to go for a macro that wraps the function name, how about MP_FASTFUN?

If you really think that makes anything better. I'd however find it more confusing, because it's code of the function is "fast", not a function for some reason "fast".

dpgeorge, 2016-10-18:
Is this macro intended for use in just esp8266 directory/files, or other files as well?  If the latter then it seems that use of the macro is going to be very specific to esp8266.

pfalcon, 2016-10-18:
The macro is purposely named in generic and understandable manner to be reusable across ports. Indeed, many (almost all?) MCUs have a similar situation when Flash at extended CPU frequencies is too slow to fetch instructions, and known workaround is to put timing-critical code into SRAM (Cortex-M has this stuff, etc.). The current work is intended for esp8266, and in most cases, MP_FASTCODE would be applied to port-specific code. (Btw, I tried to move VM execution loop to IRAM and didn't see any effect on pystone; that's another example when it might be used).

dpgeorge, 2016-10-18:
> Btw, I tried to move VM execution loop to IRAM and didn't see any effect on pystone; that's another example when it might be used

But that's exactly the problem with such a macro when used globally: for one port it might be good to have a certain function like VM loop in "fast instr memory", but for another it either makes little difference or overflows the "fast memory" segment size.

I guess the only way to make it generalise to all ports would be to tag all candidate functions with MP_FASTCODE, and then in a given port's linker script, name explicitly only those function that you actually want to have in "fast memory".

pfalcon, 2016-10-18:
> for one port it might be good to have a certain function like VM loop in "fast instr memory", but for another it either makes little difference or overflows the "fast memory" segment size.

Then we don't cover that usecase for now, but just per-port optimizations and research.

pfalcon, 2016-10-18:
Ok, turned out that for this iteration this wasn't needed - I was able to send whole .o's to iROM without immediate adverse effects. But they may appear, and to not come back to this matter later, merged in https://github.com/micropython/micropython/commit/9514d847feb70c330d310a366b92013960729bdc.

closed 2016-10-18 by pfalcon