micropython/micropython #2530: Strange problem with memory

djolez, 2016-10-18:
Hi,
I have the following problem on my ESP8266 12e chip. These are the files with stripped down code:

_main.py_
```python
import sensor_implementation as sensors
```

_sensor_implementation.py_
```python
from sensor import Sensor
```

_sensor.py_
```python
import machine
import logging
import mqtt_client

class Sensor:
    #some_code
```

_mqtt_client.py_
```python
from umqtt.simple import MQTTClient
```
When I try to run this I get `MemoryError: memory allocation failed, allocating 299 bytes` in mqtt_client.py. Strange thing is that on soft reboot (using Ctrl+D from WebREPL) I don't get this error, while on hard it happens every time, yesterday when I wanted to make a bare minimum example of this problem I got the opposite, error on soft reboot, which was OK since I found an issue here that RAM is not cleared properly using soft reboot, but this one doesn't seem logical. Now, if I change in _sensor.py_ `#some_code` with `pass` it will work. Another thing is that after every import I wrote  `gc.mem_free()` and the output was never below 10k bytes.

deshipu, 2016-10-18:
It's difficult to say anything without actually seeing the code, or being able to reproduce it.

The 10k of free memory might be all fragmented in chunks smaller than the 299 bytes needed, and then you would get a memory error too. But that's just one possible explanation.

djolez, 2016-10-18:
The only piece of code missing is `#some_code` from _sensor.py_ but I don't think that it makes much difference, I will post it anyway. The rest is commented out so what I posted here is the actual code that I am running to get the error.  

This is the missing part from _sensor.py_:
```python
class Sensor:
	def __init__(self, name, pin, read_func, init_func=None, is_analog=False):
		self.name = name
		self.pin = pin
		self.pin_object = machine.Pin(pin) if not is_analog else machine.ADC(pin)
		self.read_func = read_func
		self.init_func = init_func
		self.read_func = read_func
		self.mqtt_client = None
		
		logger.debug('Initialized {}'.format(self))
		
		self.__setup_communication()
		
	def __str__(self):
		return '[name: {}, pin: {}]'.format(self.name, self.pin)
	
	def __setup_communication(self):
		self.mqtt_topic = '{}/sensor/{}'.format(config.CONFIG['client_id'], self.name)
		self.mqtt_client = mqtt_client.make(self.mqtt_topic)
		
	def __send_msg(self, payload):
		logger.debug('{} sending msg: '.format(self.name, payload))
		self.mqtt_client.publish(self.mqtt_topic, payload)
	
	def read(self, send_response=True):
		result = self.read_func(self)
		
		if(send_response):
			msg = b'{{"action": "result", "value": {} }}'.format(result)
			self.__send_msg(msg)
		return result```

robert-hh, 2016-10-18:
The combination of both modules is about 200 loc. That's about the size the ESP8266 can handle as yource code. Try to pre-compile the modules with mpy-cross and upload the resulting .mpy files. Don't forget to remove the .py files from the esp8266, because these get precedence over .mpy files when importing.

djolez, 2016-10-19:
Thanks for the suggestion, I will try that.

djolez, 2016-10-20:
I tried with the pre-compiled modules but the problem is the same, only the number of allocated bytes in the error message changes.

robert-hh, 2016-10-21:
Are you sue that not copies of the .py files are left, for instance in the scripts directory, which will then take precedence over the compiled .mpy files.

djolez, 2016-10-21:
Yeah I am sure, i checked also with `os.listdir()`

robert-hh, 2016-10-21:
The files which are imported from the scripts directory are not visible with os.listdir(). But unless you copied it to that place, it should not be there.

djolez, 2016-10-21:
Ok, I didn't know that but in any case I didn't add anything to that folder. 

I am testing now using this code:
```python
import dht
import onewire
import ds18x20
import logging
import machine

from umqtt.simple import MQTTClient
```
And I discovered that if I put `from umqtt.simple import MQTTClient` to be before everything else then it doesn't throw an error, but all the time free heap is more than 10k so I don't know what could be the reason for this.

peterhinch, 2016-10-22:
The problem may be heap fragmentation - see the discussion of the topic [here](http://docs.micropython.org/en/latest/esp8266/reference/constrained.html). Try issuing
```python
import gc
```
Then issue
```python
gc.collect()
```
after each import statement. This may help reduce fragmentation.

djolez, 2016-10-22:
I tried it now and it doesn't help. I ran `micropython.mem_info()` in the case where it throws memory error and in the case where it doesn't (when I move import of umqtt to the top) and these are the outputs, maybe it can help.

```python
import gc
import micropython
micropython.mem_info()

import dht
import onewire
import ds18x20
import logging
import machine
micropython.mem_info()

from umqtt.simple import MQTTClient
```
```
stack: 2640 out of 8192
GC: total: 28160, used: 8544, free: 19616
 No. of 1-blocks: 55, 2-blocks: 9, max blk sz: 263, max free sz: 1048

stack: 2640 out of 8192
GC: total: 28160, used: 12784, free: 15376
 No. of 1-blocks: 142, 2-blocks: 25, max blk sz: 263, max free sz: 936

MemoryError:
```
```python
import gc
import micropython
micropython.mem_info()

from umqtt.simple import MQTTClient
micropython.mem_info()

import dht
import onewire
import ds18x20
import logging
import machine
micropython.mem_info()
```
```
stack: 2640 out of 8192
GC: total: 28160, used: 8560, free: 19600
 No. of 1-blocks: 55, 2-blocks: 9, max blk sz: 263, max free sz: 1048

stack: 2640 out of 8192
GC: total: 28160, used: 12800, free: 15360
 No. of 1-blocks: 118, 2-blocks: 22, max blk sz: 263, max free sz: 270

stack: 2640 out of 8192
GC: total: 28160, used: 16528, free: 11632
 No. of 1-blocks: 185, 2-blocks: 36, max blk sz: 263, max free sz: 350
```

robert-hh, 2016-10-22:
The suggestion of @pythoncoder was to insert a gc.collect() statement after each and every import statement. That reduces heap fragmentation, like:
`from umqtt.simple import MQTTClient`
`gc.collect()`
`import dht`
`gc.collect()`
`import onewire`
`gc.collect()`
`import ds18x20`
`gc.collect()`
`import logging`
`gc.collect()`
`import machine`
`gc.collect()`

djolez, 2016-10-22:
Yes that's what I did, sorry maybe I wasn't clear, the code that I posted is the next thing I tried after that didn't help.

djolez, 2016-10-22:
I am trying now different variations of import order to see what happens, this is the latest output of `micropython.mem_info(1)`
```
stack: 1952 out of 8192
GC: total: 28160, used: 17632, free: 10528
 No. of 1-blocks: 261, 2-blocks: 40, max blk sz: 263, max free sz: 646
GC memory layout; from 3ffef4f0:
00000: MDhhhhhMh=DhhBDhB=BBhBh===h====hhh==h===========================
00400: ================================================================
00800: ================================================================
00c00: ================================================================
01000: ===========================================BhhMDMhh=======h=====
01400: ================================================================
01800: ========hDBSBBBhBh===SB=h===hSh===B=hhhBhShhhSh========MDBM.....
01c00: BDAhhShSAhh=h=h==hhhhBBDh=h=======hShhhh======SSSh=Sh=Sh=Shh==h=
02000: h======h=========h===BBh=Dhh==========================hh=======h
02400: ===Bh=h==h=h=Sh=Sh==Shh==========h======h===hh=====hBhSBh===hh=S
02800: hh===hh==Sh=ThhMDDBDhh=h======Sh===BBh=TMhh==============h======
02c00: ==========h====h======Shh==hh===Dhhh==hh=======hh====h=Sh=hhh=hh
03000: h=hhh=hh=h===hhh==hhh==hBh====h=h===Sh===Sh==h==BDBBBBBDBBBhBBBh
03400: ===h====h=B=Bhh========hDDBhB=BTBh==B=h===h===MDhBh==h===h=DBTBB
03800: BBB=BBh========BBBhBBDhBBhhh===h====BBBBhBh===h===BBBBBhhShShShh
03c00: AhSSh========h===h=SSh=h==h==STTh===ShhhSh==h=hSShh=h==TTh====h=
04000: =h==h==h====h==h==h==hSDTBhAh===LhAhh===LhhLhShAhTSh=.hSSh=h==h=
04400: =STTh===Sh=......h========......................................
       (9 lines all free)
06c00: ................................
Traceback (most recent call last):
  File "main.py", line 15, in <module>
  File "/home/djolez/git-projects/hydroponic_2.0/esp8266/mqtt_client.py", line 1, in <module>
MemoryError: memory allocation failed, allocating 72 bytes
```
Max block size is 263 and max free 646 yet it fails to allocate 72 bytes, maybe I don't understand something but there shouldn't be problems.

pfalcon, 2016-10-22:
> Max block size is 263 and max free 646 yet it fails to allocate 72 bytes, maybe I don't understand something but there shouldn't be problems.

Note that values and 263 and 646 are in allocation units (16 bytes), not bytes. But yes, the error shouldn't happen if you use frozen bytecode. And the obvious explanation is that you don't use it, even despite having been suggested above. And there's no point to make good people like @robert-hh and @peterhinch to try to guess what you have on your side - actually, your task as a bugreporter is to make all the information as complete and as correct as possible (you can read more about effective bugreporting at http://www.chiark.greenend.org.uk/~sgtatham/bugs.html).

So, the next step at his point is providing recursive file directory listing for subdirectories `scripts/` and `modules/` of your MicroPython `esp8266/` directory.

djolez, 2016-10-22:
Sorry but this is literally my first issue that I made, I get your point. 

In the last thing that I posted I was experimenting with the modules that I wrote and trying to include them in different ways so I can see if I can get it to work (the problem is always in importing umqtt, and I froze all of my modules, as suggested) so I don't think that it would be useful to post exactly the same code since I can reproduce the same error with much simpler example that is contained in only one module.

Files that I have on esp8266 are:
```
>>> os.listdir()
['boot.py', 'port_config.py', 'main.py', 'logging.mpy']
``` 
I don't know how to upload to scripts folder as @robert-hh mentioned so I guess that it is empty. The reason why only **_logging_** is frozen is that when I tried to freeze **_main_** I got an error "_Module not found_" so I guessed that you shouldn't freeze **_boot_** and **_main_**.

_boot.py_
```python
import network
import webrepl

ap_if = network.WLAN(network.AP_IF)
ap_if.active(False)

def wifi_connect():
    import network
    sta_if = network.WLAN(network.STA_IF)
    if not sta_if.isconnected():
        print('connecting to network...')
        sta_if.active(True)
        sta_if.connect(..., ...)
        while not sta_if.isconnected():
            pass
	webrepl.start()
    print('network config:', sta_if.ifconfig())
	
wifi_connect()
```

_main.py_
```python
import gc
import micropython

import dht
import onewire
import ds18x20
import logging
import machine

from umqtt.simple import MQTTClient
```
_Output_
```
Traceback (most recent call last):
  File "main.py", line 10, in <module>
MemoryError:
```
`MemoryError` is like that, empty, but in other cases I get `MemoryError: memory allocation failed, allocating [some_number] bytes`, I didn't understand when it throws it empty and when not.

If I missed some important information please tell me.

@pfalcon I didn't understand from where I need to provide the listing of directories, esp or the ones where I keep my files on the pc? Also how can I list all the directories and files on esp, not using `os.listdir()` but something that would show also micropython libraries? Sorry if I ask stupid questions but I am fairly new to this and I am trying to learn.

robert-hh, 2016-10-22:
Paul is talking about the relevant directories on your host PC, which are micropython/esp8266/scripts and micropython/esp8266/modules. In order to put frozen bytecode into flash, you have to 
1. clone the micropython repository to your PC 
2. clone the esp-open-sdk to your PC (see  microypthon/esp8266/README.md for instructions)
3. put your files into the micropython/esp8266/modules on the host PC, 
4. rebuild the flash image for your esp8266 device (command make) and 
5. upload it to your device. (command: make deploy) 
All this assumes a Linux host OS. A virtual machine works as well for step 1 to 4. Step 5 might get a little bit tricky with virtual machines. Getting an USB device to run is sometimes hard.

djolez, 2016-10-22:
I am running Linux in a virtual machine, and the steps that I take to upload the code to esp are:
1. Cloned micropython repo
2. Use mpy-cross to create .mpy files
3. Upload them to the esp using webrepl_cli

Now it seems to me that I was doing it wrong since there is quite a difference between my steps and the ones that you suggested, I was thinking that since the .mpy modules are loaded without any errors it is working as it should.

Does this mean that what I did didn't have any effect since I was just uploading the files and not rebuilding the whole image every time?

robert-hh, 2016-10-22:
The steps you did are right, but the .mpy files will still be loaded and executed in RAM. That's good for testing, and I do that all the time. The other steps I suggested will place the .mpy files into the device binary image. The code will then be executed from flash and not use RAM for the code itself. Obviously, the data will still be in RAM. Files placed into the image will not be visible with os.listdir().

peterhinch, 2016-10-22:
To augment the points made by @robert-hh, you are trying to import quite a lot of code. My approach to a similarly large body of code was as follows. Put only the file under development in the filesystem on the ESP. Ensure none of the other modules are located there. Place the other debugged and working modules in micropython/esp8266/modules and build and deploy the firmware. This will minimise RAM use while still keeping the module under development in a form where it can be modified and run without a firmware rebuild.

djolez, 2016-10-22:
@robert-hh OK I understood now, thanks. 

@peterhinch I am trying now to follow the rebuild path and I put all my files in the /modules directory including _main.py_ but on start up I get an error saying `could not open file 'main.py'  for reading` and I couldn't find in which directory I need to put the files that are going to be on the file system, if _main.py_ is required to be there..

djolez, 2016-10-22:
Also I tried to connect over WebREPL to upload the file manually but I just get message `Disconnected`

robert-hh, 2016-10-22:
Just leave main.py and the code under test in the normal file system.
You might have told that somewhere, but which kind of ESP8266 board are you using?

djolez, 2016-10-22:
I am using esp8266 12e.

Sorry but I didn't understand what does it mean to leave it in the normal file system, I tried putting _main.py_ and _boot.py_ directly in the _esp8266/_ directory but I get the same error.

robert-hh, 2016-10-22:
No. I meant, leaving main.py and boot.py on the filesystem of the esp8266, not into the filesystem of esp8266/modules of your PC.

djolez, 2016-10-22:
Ah ok, well I ran `erase_flash` so there is nothing there and now cannot get the WebREPL to connect to the board.. At least now I have a little better understanding how everything works, I will continue tomorrow with testing and post the results here, thank you for your help.

peterhinch, 2016-10-23:
@djolez This is really a support request rather than a bug report. The place for support is in the [forum ESP8266 topic](http://forum.micropython.org/viewforum.php?f=16).

djolez, 2016-10-23:
Ok, I will make a thread there if I continue to have problems.

closed 2016-10-23 by djolez

gampam2000, 2017-08-20:
Somehow it makes a difference which module you import first. I had a similar problem and moved the problematic  import to the top.

one-quaker, 2018-03-26:
Have same problem:
`>>> micropython.mem_info()
stack: 2112 out of 8192
GC: total: 35968, used: 16128, free: 19840
 No. of 1-blocks: 128, 2-blocks: 33, max blk sz: 264, max free sz: 231
>>> micropython.mem_info()
stack: 2112 out of 8192
GC: total: 35968, used: 16192, free: 19776
 No. of 1-blocks: 130, 2-blocks: 34, max blk sz: 264, max free sz: 231
>>> gc.collect()
>>> micropython.mem_info()
stack: 2112 out of 8192
GC: total: 35968, used: 15968, free: 20000
 No. of 1-blocks: 123, 2-blocks: 33, max blk sz: 264, max free sz: 231
>>> a = bytearray(4000)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
MemoryError: memory allocation failed, allocating 4000 bytes
>>> micropython.mem_info(1)
stack: 2128 out of 8192
GC: total: 35968, used: 16144, free: 19824
 No. of 1-blocks: 127, 2-blocks: 34, max blk sz: 264, max free sz: 231
GC memory layout; from 3ffef4e0:
00000: MDShhhMhSDSShDSB=BBBh===h====hh=hh==h===========================
00400: ================================================================
00800: ================================================================
00c00: ================================================================
01000: ============================================h=hShhhhh=MDBh=Sh===
01400: ====h===========================================================
01800: ================================================================
01c00: ================================================================
02000: =================hSSh=h=ThDBh==MDhAh=======hBBMDh=.Sh===hBh=DDh.
02400: h===h==BDh=B==BhB.B=B=B.B=B.B=BSSSh=h==Bh===h=..A.h========DDDhh
02800: ==T=h==h====h=DhB.Sh=======h===....hh=======B=BBBSh..hSBBBBBBhhB
02c00: BB.BBBBBhBSB=h=..B=....hh..h=...................................
03000: .....................................h==========================
03400: ======================...............h=======..........h=h====.S
03800: h=......h=........hh=======.............S..............h====....
03c00: ......................D.Lh..h=======............................
04000: ................................................................
04400: SSh=.....................Sh=....Sh.............................h
04800: =======............................................h==........h=
04c00: .......................................................h==......
05000: .....Sh=======.........h=...........Sh..........................
05400: ............................................h======h............
05800: ....h=................................h==.......................
05c00: ..................Sh=..............Sh==.........................
06000: h=========......................................................
06400: .....hh...h=....hh....hh........................................
06800: ..................................hh....h=......................
       (3 lines all free)
07800: .................h====h========h==h==h==h====h=====h==h======h==
07c00: h======h========h===h=========h==h===h====h===h===h==========h==
08000: =======h======..................................................
       (2 lines all free)
08c00: ........`

peterhinch, 2018-03-27:
I suggest you read [the section on the heap](http://docs.micropython.org/en/latest/pyboard/reference/constrained.html) in the docs.

one-quaker, 2018-03-27:
I readed it, but question still here. I have about 19KB free memory, but I can use only 3KB, why?

peterhinch, 2018-03-28:
Heap fragmentation, as described in the doc I referenced. A 4000 byte bytearray requires 4000 bytes of contiguous memory. Large arrays should be instantiated early, before doing anything which uses significant amounts of RAM. It's best to run gc.collect() before instantiating large arrays.

one-quaker, 2018-03-28:
Thanks Peter,
I found function who fragmented all free memory

zhujisheng, 2018-04-26:
I found:
1. if I turn off the webrepl, it runs ok
2. if I turn on the webrepl, and not login from webrepl, the memory is running out
3. if i turn on the webrepl, and login into the webrepl, it works ok
4. if i turn on the webrepl, and login then logout the webrepl,  it works ok
It's strange. 
The webrepl uses less memory  after first login?

one-quaker, 2018-04-26:
mpy-cross solve my heap fragmentation problem. Other cases, described in documentation, doesn't help