micropython/micropython #2534: unix/micropython_coverage: "with" causes segfault on emit=native

hosaka, 2016-10-19:
I've spotted this looking at Travis-CI logs in #2511.
I have compiled the `micropython_coverage` and ran it with `./micropython_coverage -X emit=native.`
Forgive the lengthy example, but I couldn't get it to fail any other way. This is basically a vfs_fat testcase.
```python
import uos

class RAMFS:

    SEC_SIZE = 512

    def __init__(self, blocks):
        self.data = bytearray(blocks * self.SEC_SIZE)

    def readblocks(self, n, buf):
        #print("readblocks(%s, %x(%d))" % (n, id(buf), len(buf)))
        for i in range(len(buf)):
            buf[i] = self.data[n * self.SEC_SIZE + i]

    def writeblocks(self, n, buf):
        #print("writeblocks(%s, %x)" % (n, id(buf)))
        for i in range(len(buf)):
            self.data[n * self.SEC_SIZE + i] = buf[i]

    def ioctl(self, op, arg):
        #print("ioctl(%d, %r)" % (op, arg))
        if op == 4:  # BP_IOCTL_SEC_COUNT
            return len(self.data) // self.SEC_SIZE
        if op == 5:  # BP_IOCTL_SEC_SIZE
            return self.SEC_SIZE

bdev = RAMFS(48)
uos.VfsFat.mkfs(bdev)
vfs = uos.VfsFat(bdev, "/ramdisk")

with vfs.open("large_file.txt", "wb") as f:
    f.write(bytearray(512 * 15))
```
Which then segfaults round about stream.c#L259 when trying to mp_raise_OSError(). The bytearray size is going beyond the available size on purpose. The OSError should be MP_ENOSPC (28) and returned from the stream write in vfs_fat_file.c#L105. 

Not using the `with` keyword works okay:
```python
bsize = vfs.statvfs("/ramdisk")[0]
free = vfs.statvfs("/ramdisk")[2] + 1
f = vfs.open("large_file.txt", "wb")
f.write(bytearray(bsize * free))
```
Exactly the same example works as expected without emit=native bot using and not using `with`.

dpgeorge, 2016-10-20:
"with" is not fully supported by the native emitter.  Unless you really need "with" (eg you're testing `__exit__` on the file object) please use try/except/finally instead so that it'll work with the native emitter.

hosaka, 2016-10-20:
Thanks for the clarification. The problem doesn't go away with try/except, however. If I surround the file write block, with try/catch, I still get a segfault, rather than an OSError and throw back to REPL (try/except). 

I stepped through the code briefly, it appears to fail upon nlr_jump(), but I don't have enough knowledge to dig further. I expect at least not segfaulting would be preferable?

As it is not supported, I can close the issue, but just wanted to bring the hard crash to your attention :)

pfalcon, 2016-10-20:
Please provide a small testcase (in a separate ticket apparently) and we'll look into it.

hosaka, 2016-10-20:
Sure thing, I will look into it a bit further and try to narrow down the problem, hopefully excluding the VFS.

closed 2016-10-20 by hosaka