micropython/micropython #2535: esp8266: Architecture and implementation of OTA update

pfalcon, 2016-10-19:
I have a WIP prototype of OTA updates support for esp8266 port. The architecture is:

1. First-stage bootloader. Decides which of 2 components below to boot.
2. OTA server application.
3. MicroPython application.

The distinctive difference from existing OTA solutions for esp8266 is that it doesn't require 2 areas for the main application, with a new OTA update being flashed to "inactive" area, and "active area" switch occurring only integrity of new flashed application is confirmed. With proposed scheme, OTA update overwrites area of the main application immediately. This is based on the requirement to utilize well resources of 1MB modules: support both file system and OTA.

Of course, such scheme is much more risky, so all measures to preventing "bricking" by incomplete OTA update should be provided.

mkarliner, 2016-10-19:
Fantastic!

pfalcon, 2016-10-19:
I now have a prototype of 1st-stage bootloader (capable of just chain-loading a single app store elsewhere). My plan would be to make it a separate project, for reuse by other projects. Then OTA server app would be another separate component (OTA app is just a normal application for the most part). So, it should be considered how these components would be brought into micropython. With plan above, there would be 2 new git submodules. Potentially, bootloader and OTA app can be combined into a single repo. Finally, we can bring up just precompiled binaries into upy repo.

pfalcon, 2016-10-19:
All in all, I'd like to move fast with adding these features, even before full scheme works completely - there will be substantial changes to how upy starts, so the sooner people will start to test it, the sooner issues will be known.

Of course, we need to support OTA-less deployment, apparently as backlogged 512K-only firmware (otherwise, I wouldn't like to support both OTA-ful and OTA-less variants for other flash sizes, even though 1MB's FS size will likely be reduced considerably :-( ).

pfalcon, 2016-10-21:
Ok, with SDK2.0, minimal application size is 0x30d00 iROM + 0x7xxx iRAM. I.e. OTA app would be 250+K, which kinda kills an idea of having a useful filesystem...

pfalcon, 2016-10-21:
One stubborn approach would be to follow the same idea as bootrom lib uses - link SDK at specific address and use it as a binary blob, reusing between upy and ota app. But of course, then SDK itself can't be updated.

dpgeorge, 2016-10-21:
> Ok, with SDK2.0, minimal application size is 0x30d00 iROM + 0x7xxx iRAM. I.e. OTA app would be 250+K,

I assume that's because "minimal app" includes a lot of wifi etc libraries.

> One stubborn approach would be to follow the same idea as bootrom lib uses - link SDK at specific address and use it as a binary blob, reusing between upy and ota app. But of course, then SDK itself can't be updated.

Well, one could indeed argue that uPy will be updated much more often than the SDK.  But it does seem rather restrictive to need to flash via a wire when the SDK changes, and also it might lead to many user difficulties knowing when OTA can proceed and when it can't.

mkarliner, 2016-10-21:
Sorry to be slightly off topic, but what security measures do you envisage OTA to have?

pfalcon, 2016-10-21:
> I assume that's because "minimal app" includes a lot of wifi etc libraries.

Yes, just consider that it should support WEP, WPA with all the crypto variants, etc.

> Well, one could indeed argue that uPy will be updated much more often than the SDK. But it does seem rather restrictive to need to flash via a wire when the SDK changes, and also it might lead to many user difficulties knowing when OTA can proceed and when it can't.

So, there would be a way to update it all: "SDK blob" part would be logically grouped with OTA app, and an OTA app would be able to update SDK blob + OTA app (it would flash-download it to the main app area, then once complete and verified, copy using an iRAM routine to the final place). Complexity and error-proneness of that would depend on smartness and complexity of a protocol (versioning of different parts, etc.)

Anyway, that's a possible optimization for future.

pfalcon, 2016-10-21:
> Sorry to be slightly off topic, but what security measures do you envisage OTA to have?

There will be public key based verification of OTA update.

pfalcon, 2016-10-23:
Ok, I just finished first integration test of all this stuff. OTA isn't too fast, took 2min for me (AP is 1-2 m away across brick wall). Of ~500 packers, there was 50+ retransmits (10%). Missing feature of the core functionality is checksumming the main app. Culprit is where to store complete size of the area to checksum. Currently it's stored in padding between iRAM and iROM blocks, and exact location is hardcoded into Python code. For a generic solution, calculating this location is quite boring, especially that for initial bootloader, memory-mapping of flash isn't yet set up.

So, instead I propose just to prefix complete firmware image with its size, and suffix with md5sum.

pfalcon, 2016-10-23:
> So, instead I propose just to prefix complete firmware image with its size, and suffix with md5sum.

Well, "complete firmware" doesn't work here. It causes iROM to shift too, trying to set iROM start address to 0xXX004 causes it be wrapped to 0xXX010. So, I guess will have to hardcode it like before.

pfalcon, 2016-10-23:
Checksumming 550KB takes 250ms :-I .

pfalcon, 2016-10-23:
Going from 4K to 8K block for reading/checksumming saves 28380 CPU cycles (19419961 vs 19391581). Times seem to be cycle-perfect, even though they involve I/O with an external device (SPI flash). Now sidechannel (timing) attacks on crypto don't sounds very unrealistic ;-).

19419961 19391581

pfalcon, 2016-10-25:
Ok, so based on the above, I'm choosing this:

> bootloader and OTA app can be combined into a single repo.

There're still more questions: where to host OTA client script? Where to host helper scripts for key generation, etc.?

@dpgeorge : I'd appreciate comments or at least acks.

dpgeorge, 2016-10-27:
> So, instead I propose just to prefix complete firmware image with its size, and suffix with md5sum.

Could you have the first machine instruction of the firmware a "jump", followed by the size (which would then be at a fixed location), then the destination of the jump?

> Checksumming 550KB takes 250ms

Given the download time is about 2 minutes, I'd say .25 seconds for checksum calc is fine.

> bootloader and OTA app can be combined into a single repo.

That sounds good.  The micropython organisation doesn't feel like the right place to put this repo, but I'd be fine putting it there if you think otherwise.

> where to host OTA client script?

You mean the OTA server script, the one that serves up the new firmware?  I guess it would go in the same repo as the bootloader and OTA app.  Same with the helper scripts, just put them together.

pfalcon, 2016-10-27:
> Could you have the first machine instruction of the firmware a "jump", followed by the size (which would then be at a fixed location), then the destination of the jump?

Given that the original idea was to make this fully generic and reusable, the question is actually "could everybody else do that?". The first byte of firmware is signature 0xe9, followed by some flags, followed by segment header, then segment data, then another header, etc. Order of segments is random, or ordered based in .ld file, or something. Summing up, no, everybody won't be able to do that, and, I having  given a quick thought to it, decided not to bother. I guess, that's part is already resolved.

>> Checksumming 550KB takes 250ms-
> Given the download time is about 2 minutes, I'd say .25 seconds for checksum calc is fine.

No, those 250ms happen on each reset, and thus a non-trivial boot delay. Beyond the initial implementation, which is now done, there will be need for a number of tweaks and optimizations, like skip all this on wake up from deep sleep, etc.

> That sounds good. The micropython organisation doesn't feel like the right place to put this repo, but I'd be fine putting it there if you think otherwise.

Well, I'd put under my account, later we can move it if makes sense.

> You mean the OTA server script, the one that serves up the new firmware? I guess it would go in the same repo as the bootloader and OTA app. Same with the helper scripts, just put them together.

I mean OTA client, per common naming that server is a part which waits for requests, and client which issues them ;-). Again, push model of OTA is implemented (wireless analog of normal wired UART upgrade). If we put everything into a single repo, an end user will need to download all this "cruft" just to use OTA. I'm all for that (disseminating open-source), just wonder if there're any concerns (and I'm guess I'm starting to err on the side of "make it really simple", to avoid extra questions). Other option is to make self-contained script (currently there're 2 modules).

pfalcon, 2016-10-28:
Ok, I started to spool the code to https://github.com/pfalcon/yaota8266.

drasko, 2016-11-23:
@pfalcon how functional is your OTA at this point? Can you please add a small REAME.md over here: https://github.com/pfalcon/yaota8266 with instructions so we can test it / use it?

gesaleh, 2016-12-08:
@pfalcon any update on the readme and usage of the OTA over the ESP8266

pfalcon, 2017-01-02:
@drasko, @gesaleh : yaota8266 is in development, and there's no other information besides what's available in the github project (and which you can watch unfold using means github provides).

pfalcon, 2017-01-02:
@dpgeorge : With today's changes to https://github.com/pfalcon/yaota8266, I guess its basic functionality is there. As usual, would love to hear your feedback.

drasko, 2017-01-02:
@pfalcon thanks for adding README. On a quick look I have spotted that RSA encryption is here - great! Just a small remark - I think that ECC would be a better candidate for embedded devices, due to the key size: https://casecurity.org/2014/06/10/benefits-of-elliptic-curve-cryptography/

pfalcon, 2017-01-02:
I'm using what's available in the existing esp8266 toolset. axTLS (version used) supports only RSA, and that's what I use.

drasko, 2017-01-02:
[esp-open-rtos](https://github.com/SuperHouse/esp-open-rtos) has a port of [mbedTLS](https://github.com/SuperHouse/esp-open-rtos/tree/master/extras/mbedtls), which supports ECC.

pfalcon, 2017-01-08:
Ok, while I'm waiting @dpgeorge's feedback and thoughts on next steps, I'll dump my thoughts on that. @dpgeorge , as a next step, http://micropython.org/download#esp8266 should start to provide following versions of firmwares: 1) non OTA enabled firmware; 2) OTA enabled firmware; 3) OTA upgrade image. With 512K version, that will be 4 firmware types.

pfalcon, 2017-02-05:
@dpgeorge : It would be nice to keep this going. The above is purely an organizational decision question, so should not take much effort to answer. Though if you answer "yes", the next step is indeed starting to provide all those downloads. Providing OTA downloads ahead of OTA feature announcement may be backwards, but should be a motivation to come out with such announcement (after the testing required, the stage it's currently at).

From my side, I considered of what features I'd like to have from OTA bootloader, and implemented them all a month ago, except https://github.com/pfalcon/yaota8266/issues/10, which itself depends on https://github.com/micropython/micropython/issues/2779.

I was about to test this with a real OTA-requiring device (like, something nailed to a ceiling where accessing it physically is a real chore), but then NY holidays suddenly ended. I can't concentrate on something hardware and physical bound now, and have no idea when I will be. But in my list, OTA works. So everything depends on you confirming it works for you, or providing feedback what needs to be fixed. (It's clear there's not enough docs, but the idea is that what's available should be enough. The uPy core lacks docs, and that's of higher priority. But again, if you something blocking, please let me know.)

dpgeorge, 2017-02-07:
> as a next step, http://micropython.org/download#esp8266 should start to provide following versions of firmwares: 1) non OTA enabled firmware; 2) OTA enabled firmware; 3) OTA upgrade image. With 512K version, that will be 4 firmware types.

Yes, sure, that's a good idea to provide OTA firmware at the download site.  But please remind me the difference between (2) and (3)... I can easily add the "make ota" target to the download site, but to have a full working firmware doesn't one need to compile with their specific RSA key?

pfalcon, 2017-02-07:
> difference between (2) and (3)

There're: 1) non OTA enabled firmware, complete flash image, which thus cannot be upgraded using OTA; 2) complete "seed" OTA firmware which needs to be flashed once and then can be upgraded OTA (until OTA bootloader change is made); 3) signed OTA image using OTA upgrade (not a flash image).

> but to have a full working firmware doesn't one need to compile with their specific RSA key?

Right, so it's about time to set it up if otherwise you confirm that the OTA infrastructure and process work for you. This key would belong to you (George Robotics). So, just "make ota" won't work, you'd need to prepare the bootloader with your key.

puuu, 2017-02-09:
@pfalcon wouldn't it be possible to provide a function in micropython that allow to change the public key of the ota-server?
Then it is easier for the user to flash a compiled OTA firmware signed with his own private key, without having to reflash the OTA bootloader. As far as I understood it, only the 512 bits modulus of the RSA public key must be overwitten in the flash. 

For people that may think of a security risk: Everyone who has access to the REPL or scripts, has already all rights to do everything with the firmware.

dpgeorge, 2017-02-27:
@pfalcon I was testing the OTA code.  I got ota-server to run, but ota-client is unable to send the data via UDP.  It seems that any UDP packets that are 549 bytes or larger are just ignored by the esp8266.  I checked this using the non-OTA firmware as well (by creating a simple UDP client and server, server running on esp8266, client on PC) and indeed 548 bytes is the largest UDP packet that the esp8266 will receive.  This number matches the known MTU limit before packets are fragmented.  I wonder how you managed to get packets as large as 1108 bytes (the size ota-client sends) through?

pfalcon, 2017-02-27:
Apparently, I have a proper MTU set for an Ethernet-like link (which WiFi is) ;-). So, what MTU ifconfig reports for your WiFi connection?

pfalcon, 2017-02-27:
(Mine is obviously 1500).

pfalcon, 2017-02-27:
Also, did you test directly between a Linux box and esp8266, or cute vendor routers are involved?

dpgeorge, 2017-02-27:
> Apparently, I have a proper MTU set for an Ethernet-like link (which WiFi is) ;-). So, what MTU ifconfig reports for your WiFi connection?
> Also, did you test directly between a Linux box and esp8266, or cute vendor routers are involved?

I connected directly to esp8266's AP.  The IP address of the Linux PC was 192.168.4.2 and the MTU was 1500:
```sh
$ ip addr
...
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    inet 192.168.4.2/24 brd 192.168.4.255 scope global wlan0
       valid_lft forever preferred_lft forever
...
```

pfalcon, 2017-02-27:
Weird. And you of course use current esp-open-sdk with inlwip_open?

Anyway, ota-server doesn't care about specific packet size, it doesn't even care packets be of the same size. It's just packets should add up to 4K flash page buffer. I even thought about sending 1500, 1500, 1096 sized packets, to have 3 packets per page instead of current 4. Well, we can take a hint and make that 8, with the default packet size of 512, and that's how you'll provide officials images, while adding a command line option to override the size.

But it would be nice if you figured what's wrong, because it surely affects other people, and then any UDP-related report like "multicast doesn't work" can be actually this underlying issue.

dpgeorge, 2017-02-27:
> Weird. And you of course use current esp-open-sdk with inlwip_open?

What's "inlwip_open"?  Did you mean "esp-open-lwip"?  Yes I'm using the latest esp-open-sdk with "git submodule update --init" all current.

I just tried it again, with different configurations, and the results are:
- PC connected to esp8266's AP, sending UDP packets from PC to esp: anything above 548 bytes doesn't get through
- PC connected to esp8266's AP, sending UDP packets from esp to PC: packets above 548 bytes *do* get through
- PC and esp8266 connected through a local wifi router: packets above 548 *do* get through from PC to esp

> Well, we can take a hint and make that 8, with the default packet size of 512,

If the overhead of a packet is 84 bytes (1108-1024) then the default packet size needs to be 256 bytes of data, and there needs to be 16 for a 4k flash page.

dpgeorge, 2017-02-28:
Ok, I managed to get the OTA update working.  With PC and esp connected through a local wifi router I can use a packet size of 1024 without a problem.  With PC directly connected to esp's AP I need to use 256 for the packet size (512 is still too big with packet overhead added).  In the latter case the update took about 40s, which is roughly 14kbytes/sec.

dpgeorge, 2017-02-28:
OTA initial images (.bin) and update files (.ota) are now served at https://micropython.org/download/#esp8266

pfalcon, 2017-02-28:
Cool, thanks! So, what blocks size the images there use, and what we should have as the default in the tool?

dpgeorge, 2017-02-28:
> what blocks size the images there use

256 bytes.  I think that's the one we should use as default.  It doesn't have too much of an impact on overall flashing time.

pfalcon, 2017-03-02:
Ok, 256 was set as the default in ota-client, and I opened a separate ticket there, https://github.com/pfalcon/yaota8266/issues/15.

pfalcon, 2017-04-09:
As it doesn't make sense to wait more (I can't make further progress on it for the foreseeable future), I posted a "soft launch" announcement on the forum: https://forum.micropython.org/viewtopic.php?f=16&t=3262

msedv, 2017-05-20:
@pfalcon So far OTA works fine, but I have a general problem: I would like to use it to update ESP8266-devices which are NOT easy to access and thus not possible to press a button or something like that. Is there a way which I didn't find (or will there ever be) to start the OTA process e.g. from WebREPL?

Thank you!

Lisa999, 2017-09-29:
@msedv: https://github.com/ulno/yaota8266

jedie, 2019-12-30:
> I would like to use it to update ESP8266-devices which are NOT easy to access and thus not possible to press a button or something like that.

There is a idea in: https://schinckel.net/tags/yaota/

For me, it seems that the yaota8266 project is not really alive.
as @pfalcon write in https://github.com/pfalcon/yaota8266/issues/25 :

> ...are no longer developed at this time by me

There are some open pull requests since may 2018: https://github.com/pfalcon/yaota8266/pulls/schinckel

And a tutorial is also missing, see: https://github.com/pfalcon/yaota8266/issues/26

Only https://schinckel.net/tags/yaota/ contains helpful information how to put everything together...

My suggestion would be to merge yaota8266 into micropython repository and describe in the official documentation how to build an OTA firmware and how to perform the OTA update.
I also think it will be a good idea to implement a pure python OTA server, isn't it?

jedie, 2020-01-01:
> My suggestion would be to merge yaota8266 into micropython repository and describe in the official documentation how to build an OTA firmware and how to perform the OTA update.

Any interests in this? Then i would try to create a pull request ;)

dpgeorge, 2020-06-09:
> Any interests in this? Then i would try to create a pull request ;)

It seems there is some interest (by the thumbs up).  If you open a PR here please be mindful of licensing (nothing GPL).

dpgeorge, 2022-06-23:
Closing due to inactivity.

closed 2022-06-23 by dpgeorge