micropython/micropython #2539: esp8266: Publishing .elf and .map files for official downloads

pfalcon, 2016-10-20:
To investigate issues like #2340, #2537, it would be nice to have prompt access to ELF/map files for official releases. Trying to reproduce a particular build locally takes extra time and may still be in vain (no guarantee toolchain produce identical binaries on different systems, plus local system may have various patches and changes which are hard  to track completely).

dpgeorge, 2016-10-21:
.elf and .map files are now retained for daily builds, and the html page now links to them, but they'll take a few days to populate.

pfalcon, 2016-10-21:
Well, my idea was to make them available for download, but not crowd space on the page by linking to them. Just adding a URL from .bin to .elf or .amp would work really well. Sorry for not making that explicit.

dpgeorge, 2016-10-21:
> Just adding a URL from .bin to .elf or .amp would work really well.

Yes, that bit is done as part of it.  They have the same root URL, just different extension.

> but not crowd space on the page by linking to them

It doesn't add much, just `(elf, map)` for each item.  Without the link I think people (myself included) will forget that those files exist.

pfalcon, 2016-10-21:
Ok, thanks!

closed 2016-10-21 by pfalcon