micropython/micropython #2541: ESP8266: make deploy shows incorrect flash filesize in printed esptool.py command

jrobeson, 2016-10-21:
```sh
make PORT=/dev/ttyUSB0 deploy                                    46.3s  2016-10-20 22:40:55
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Generating build/genhdr/mpversion.h
GEN build/genhdr/qstrdefs.collected.h
QSTR not updated
CC ../py/modsys.c
CC ../py/../extmod/modwebrepl.c
CC moduos.c
CC ../lib/utils/pyexec.c
LINK build/firmware.elf
   text	   data	    bss	    dec	    hex	filename
 553780	   1084	  56352	 611216	  95390	build/firmware.elf
Create build/firmware-combined.bin
esptool.py v1.2-dev
('flash    ', 36144)
('padding  ', 720)
('irom0text', 518760)
('total    ', 555624)
('md5      ', '5e615515229ad27c3cb651a721476e95')
Writing build/firmware-combined.bin to the board
esptool.py v1.2-dev
Connecting...
Auto-detected Flash size: 32m
Running Cesanta flasher stub...
Flash params set to 0x0040
Writing 557056 @ 0x0... 557056 (100 %)
Wrote 557056 bytes at 0x0 in 48.3 seconds (92.3 kbit/s)...
Leaving...
Verifying just-written flash...
Verifying 0x87a78 (555640) bytes @ 0x00000000 in flash against build/firmware-combined.bin...
-- verify OK (digest matched)
#@esptool.py --port /dev/ttyUSB0 --baud 115200 write_flash --flash_size=8m 0 build/firmware.elf-0x00000.bin 0x9000 build/firmware.elf-0x0[1-f]000.bin
```

8m is hardcoded in the output. I tried changing it to `$(FLASH_SIZE)` which changes the output to `--flash_size=detect`. If that's what is wanted, I can submit a PR.

dpgeorge, 2016-10-21:
That line is old and commented out and was originally used to flash separate parts of the firmware.  But the firmware is now combined into 1 file.  Use `make V=1` to see the actual esptool command that is executed.

@pfalcon I guess we should just delete this old line?

jrobeson, 2016-10-21:
ah, I thought it was there on purpose (due to the way Makfile handles comments)

pfalcon, 2016-10-22:
Indeed, removed in 1b76614d4154dbe41d79d3e750b1e34964f5eaf1.

closed 2016-10-22 by pfalcon