micropython/micropython #2543: streams: Support binary-encoded I/O

pfalcon, 2016-10-21:
There's a well-known paradigm of reading/writing binary encoded primitive values from/to stream. This paradigm became well-known with Java, and used in Android, Node.JS, etc. Due to Java heritage, there's big collection of method, one for each data type. Some environments lead it to extremes, for example Node.JS has methods like write32UInt32BE(), read32UInt32BE().

All in all, it's useful paradigm which doesn't have good footing in Python. Instead Python's approach is based on encoding/decoding whole structures with struct module. However, when working with primitive data types that leads to verbose code, and for us, has a drawback of requiring extra memory allocation.

I propose to add binary-encoded I/O support to MicroPython stream classes, byte avoid bloated method space pitfall using Python's native approach to data type encoding, using character codes, as used in standard modules struct and array. Examples:

~~~
s.writebin("B", 100)  # write a byte
val = s.readbin("<I")  # read little-endian int32
s.writebin("H", 0xaa55) # write using native system byte order, as expected
~~~

This extension is intended to be used in standard stream classes (files, sockets) and in stream-like classes of HardwareAPI (I2C, SPI, etc.)

dpgeorge, 2016-10-21:
That's certainly a nice idea and seems quite general.  In particular `readbin` would be most welcome, to save doing things like `s.read(1)[0]`.

But it's obviously very biased towards byte streams.  For example, how would it work for something like UART when in 9-bit mode?  stmhal has `uart.readchar()` and `uart.writechar(int)`.  These could become `uart.readbin('c')` and `uart.writebit('c', 0x1ff)`.

Also with SPI in 16-bit word mode, that would need special consideration.

pfalcon, 2016-10-21:
> That's certainly a nice idea and seems quite general.

Glad you like it!

> But it's obviously very biased towards byte streams. For example, how would it work for something like UART when in 9-bit mode?

As any general mechanism, it may not cover niche cases. As the examples you give show, those can be covered too if desired.

pfalcon, 2016-10-21:
> A note about naming: I think it would make sense to call those methods "write_struct" and "read_struct", giving a hint that they use the same format as the struct module.

Not more than the "array" module should be named "array_struct" just because it uses (almost) the same codes as "struct".

pfalcon, 2016-10-21:
> A more "pythonic" approach to avoiding memory allocation would be to make those stream methods accept any iterable, not just things that fit in memory in a single chunk -- but I understand it would be much harder to implement.

The proposed methods specifically work with a single scalar value (number). Working with an arbitrary iterable is as easy as:

~~~
for b in in:
    s.writebin("B", b)
~~~

dpgeorge, 2016-10-22:
I also wondered about reading/writing multiple values.  The problem with reading multiples is that then you should always return a tuple of the results, and reading of a single value would necessarily need to return a tuple with 1 item.  Not great, since it needs to do an allocation.

Regarding adding these methods SPI: SPI reading requires sending data out on mosi, and the API allows you to specify the output byte.  So it seems you'd also need such an (optional) argument to readbin:
```
val = spi.readbin('<H', 0xff) # read a 16-bit int, while writing 0xffff on mosi
```
And for I2C you need to specify the device address, something like:
```
i2c.writebin_to(addr, 'B', 123)
val = i2c.readbin_from(addr, 'B')
```

deshipu, 2016-10-22:
> As it stands, .writeInt() - like methods are an orthogonal approach to classical Python's using struct module. So, what I propose is to add this orthogonal approach. What you propose is to go step further and make a "dot product" of Python's API paradigm and write-single-value API. The resulting API is more complex, if not heavy-weight or bloated.

I'm not proposing anything. I just wrongly interpreted your initial proposal, because it wasn't entirely clear to me that it was about single values.

dpgeorge, 2021-05-18:
Closing, see https://github.com/micropython/micropython/pull/3382#issuecomment-842807006

closed 2021-05-18 by dpgeorge