micropython/micropython #2548: WiPy: OSError or Memory Error after file read & list operations

robert-hh, 2016-10-22:
This little piece of code runs into a OSError or Memory Error:
```
import gc
def get_file(fname):
    with open(fname) as f:
        content = f.readlines()
####### this is the critical section
    if True:
        for i in range(len(content)): 
            content[i] = content[i].rstrip('\r\n\t ')
#######
    return content
    
t = get_file("getfile.py")
print("1", t)
gc.collect()
print("2", t)
```
The first print statement is executed, but the OSError comes up. If I change the start of the "critical section" to: `if False`, everything runs well.
Wipy, build from today's sources:
MicroPython v1.8.5-35-gf184139 on 2016-10-22; WiPy with CC3200

dpgeorge, 2016-10-24:
Thanks for the bug report.  The problem was that `file.close()` was being called multiple times and only the first time succeeded, the rest failed with OSError.  Furthermore, the finaliser `file.__del__()` was being called from the `gc.collect()` and raising an exception within the garbage collection run.  This lead to the heap remaining locked at the REPL, hence the MEmoryError.  This issue also affected stmhal (and probably esp8266).

So, in 56942019309645781d330312f5944db2d4cb5cd7 I fixed file.close() so it doesn't try to close the file more than once.  It fixes the problem described above.

BUT, there seems to be a separate problem with cc3200 (wipy).  If the code example in the OP above is saved to a file "getfile.py", then the following is run at the REPL, it eventually crashes (REPL becomes unresponsive):
```
>>> import getfile
>>> getfile.get_file('getfile.py')
>>> getfile.gc.collect()
[use up/down arrow keys to go through history]
[use tab completion]
>>> getfile.get_file('getfile.py')
>>> getfile.gc.collect()
[continue doing these things]
```
I checked that this bug is there as far back as v1.8.  I don't have resources to fix this so will defer to @danicampora .

robert-hh, 2016-10-24:
Hello @damien. Thanks for fixing. But I cannot reproduce the REPL lock.

closed 2021-03-07 by robert-hh