micropython/micropython #2551: Cleaning up file naming in esp8266/

pfalcon, 2016-10-23:
In esp8266/ we have (among others) file named like:

* modpybpin.c
* modmachinewdt.c

So, first name is based on historical naming in another port and clearly incorrect. 2nd name is attempt to fix that, but still not too right.

I think both name should be:

* modmachine_pin.c
* modmachine_wdt.c

If there're concerns of too long names, these can be named

* machine_pin.c
* machine_wdt.c

Note that in extmod/ , we have machine_spi.c, etc. But there's fine distinction that stuff in extmod/ is inventory of functions which can be used for modmachine, wheres stuff in esp8266/ and other port directories is actual implementation of a particular modmachine.

I really like to proceed with renaming soon, because this starts to be a bad example for other ports (e.g. there's a Zephyr port patch pending which has modpybpin.c).

pfalcon, 2016-10-23:
@daniel-thompson : FYI

dpgeorge, 2016-10-24:
Note that extmod/machine_i2c.c does actually include a full implementation of machine.I2C class, so should it be renamed to extmod/modmachine_i2c.c?  I'd say having everything (in extmod/ and in port-specific dir) just machine_XXX.c is fine and means you don't have to think/know if a particular file implements the class or not.

OTOH, extmod/machine_i2c.c doesn't allow a port to supply hardware accelerated versions of I2C functions (unlike extmod/machine_spi.c, which does).  So that i2c code needs to be refactored to allow this (hence it may no longer have an implementation of machine.I2C).

pfalcon, 2016-10-24:
> I'd say having everything (in extmod/ and in port-specific dir) just machine_XXX.c is fine and means you don't have to think/know if a particular file implements the class or not.

So, is it just for machine_*, and not other modules? ;-) Rules getting complicates, and inconsistency rampages ;-).

In extmod, there's moduos_dupterm.c. Should it be renamed to uos_dupterm.c? (My answer is yes.)

dpgeorge, 2016-10-25:
> In extmod, there's moduos_dupterm.c. Should it be renamed to uos_dupterm.c?

Definitely.  Otherwise it's a "dupterm implementation of full moduos", like "modussl_axtls" is axtls impl of modussl.

dpgeorge, 2016-10-25:
> So, is it just for machine_*, and not other modules? ;-) Rules getting complicates, and inconsistency rampages ;-).

I'd say general rule (for extmod/ and ports) is: anything beginning with "mod" should contain the actual module object and local dict with all entries for that module.  So there should only be one modXXX.c for any given module (any given XXX) in any given build.  Any partial implementation of a module (whether it contains some functions or full class) should start with just the module name.

pfalcon, 2016-10-25:
>> In extmod, there's moduos_dupterm.c. Should it be renamed to uos_dupterm.c?
> Definitely. 

Done.

pfalcon, 2016-11-05:
Majority of changes for esp8266/ done in https://github.com/micropython/micropython/commit/8bc3fc20feca9ebe9e9972aeb6b68a81d792c5df . Smaller cleanups may be needed still, so keeping open.

dpgeorge, 2020-06-09:
Lots of clean up has been done now, including moving all ports to `ports/`.  So closing this issue.

closed 2020-06-09 by dpgeorge