micropython/micropython #2553: ds18b20: negative temperatures not correctly reported

mutax, 2016-10-23:
Hey,

I have to use the following code to get negative temperatures from my DS18B20 sensor:

        data = ds.read_scratch(rom)

        temp_lsb = data[0]
        temp_msb = data[1]

        if (temp_msb & 0x80)==0x80:
            # convert negative number
            temp = (((~temp_msb & 7) << 8)|~temp_lsb) / 16.0
        else:
            temp = (((temp_msb & 7) << 8)|temp_lsb) / 16.0

The default implemenatation returns 4077.5625 instead of -13.625 °C

mutax, 2016-10-30:
My code above has some issues, e.g. not correctly reporting all negative values. Working on a fix.

mutax, 2016-10-30:
for negative temperatures the code has to look like this:

 `  temp = -((((temp_msb & 7)^7)<<8 | (temp_lsb^255))+1)/16.0 `

Because in python using ~variable interprets the result already so one has to use variable^x to invert the bits. Additionally as there is no negative zero you have to add 1 to the inverted bitmask in 2 complement, which I forgot as well, leading to slightly wrong values.

I checked all examples from Table 1 in the datasheet http://datasheets.maximintegrated.com/en/ds/DS18B20.pdf
and the results are correct for the negative values now. Sorry I did not check all values with that table in the first place when reporting the issue.

deshipu, 2016-10-30:
Why don't you use the `ustruct` module to properly convert a signed number, instead of doing it manually?

mutax, 2016-10-31:
The original code is using the manual approach, I have to divide anyway to get the comma values and I do not know the additional memory footprint of the module. On my ESP memory is an issue so I did not want to import that module just for one simple calculation.

turbinenreiter, 2016-10-31:
@deshipu how do you do two's complement with struct? the struct doc doesn't seem to mention it.

deshipu, 2016-10-31:
The module is built-in, so there is no additional memory wasted.

You can do:
```
temp = struct.unpack('<h', data)[0]
```

dpgeorge, 2016-11-02:
Should be fixed by #2582.

closed 2016-11-02 by dpgeorge