micropython/micropython #2556: vfs_fat/fsusermount: mkfs, mount, umount inconsistencies.

hosaka, 2016-10-24:
While writing the tests for fsusermount, I notices some oddities with the vfs_* functions that I'd like to mention.

1) Fsusermount mkfs:
```python
uos.vfs_mkfs(bdev, "/ramdisk")
```
takes a blockdevice and a path, the latter is not necessary but required. Comparing this to:
```python
uos.VfsFat.mkfs(bdev)
```
which writes the filesystem and nothing else, as the function name suggests. Defined as a CONST_FUN_OBJ_1 in vfs_fat.c (#L55)

2) We have 3 ways of unmounting the device:
```python
uos.vfs_umount("/ramdisk") # by calling umount
uos.vfs_mount(None, "/ramdisk") # by mounting None
uos.vfs_mkfs(None, "/ramdisk") # why?
```
While first 2 calls are logical, the last call is a side effect from having the same underlying call to `fatfs_mount_mkfs()`. We duplicate the umount functionality in fsusermount.c (L#54, #L166)

Mkfs thus also can take a readonly=True parameter, which will obviously fail.

3) In vfs_fat.c both `fat_vfs_make_new()` and `fat_vfs_mkfs()` call the internal function `fatfs_mount_mkfs()`, instead of `fatfs_mount()` and `fatfs_mkfs()` respectively, which should be internal to fsusermount.

What I would like to propose is:
1) Have the mkfs take a block device only, it should not be concerned with mount path at this stage.

2) Umount functionality should belong to it's `fatfs_umount()` on L#166 only, and not be repeated in `fatfs_mount_mkfs()`. Mounting a None case can then make a call to umount instead of being duplicated, but I don't see a good reason for such call in favour of regular umount. Readonly=True argument should be relevant to mounting only.

3) Just like we have a new "_internal" function for removing files/dirs, `fatfs_mount_mkfs()` should be renamed and not called directly by VfsFat. Mounting and mkfs will use the internal function as they do at the moment.

This is just my observations, please share your thoughts and counter-arguments!

pfalcon, 2016-10-24:
VFS system is at its infancy, and currently a hack. Further progress depends e.g. on https://github.com/micropython/micropython/pull/2140.

@hosaka , I wanted to write for a while, but can't find enough time. So, let me be short - you did a great work on improving coverage of VFS-related things, but don't go *too deep* with it. As I mentioned, it's WIP and a hack currently anyway. Besides, improving coverage work should be fun - you hack something here, then hack something there, never stuck at one area for a long time and get bored ;-). YMMV

hosaka, 2016-10-24:
That's very interesting! Somehow I missed that PR :)
You're right of course, what I outlined here largely depends on what's underneath in FatFs and further changes to VFS. I do like filesystem work though, that's why I wanted to drive existing coverage to it's maximum and help with other related work.

I will follow your advise and put on hold any work after #2542. As that adds the last set of tests. I am also happy to maintain it in the future if FatFs gets updated and the tests need to be adjusted.

There's plenty of other coverage work in the meantime.

hosaka, 2016-10-24:
If you and Damien can agree that what I outlined in the issue description, is valid and can be used as a sidenote for future fsusermount changes, I will keep this open.

pfalcon, 2016-10-24:
Yes, please keep it open.

dpgeorge, 2016-10-25:
@hosaka the tests you've written are very valuable because when #2140 finally lands we will have an extensive test-suite to test it against.

jms19, 2016-10-28:
I agree some names and locations of filesystem functions are wrong and need fixing if nothing else to prevent further head scratching.

dpgeorge, 2017-01-27:
These suggestions are now implemented by the new VFS sub-system.  In particular you create and mount devices like this:
```python
bdev = BlockDevice() # create a block device
uos.VfsFat.mkfs(bdev) # create a filesystem on the block device
vfs = uos.VfsFat(bdev) # creat a filesystem object
uos.mount(vfs, "/mnt") # mount the filesystem object on the VFS
...
uos.umount("/mnt") # unmount
uos.umount(vfs) # same as above
```
There is currently no `uos.mkfs` but it can be added later if needed/wanted (use uos.VfsFat.mkfs instead).

The C code still needs some cleaning but only after all ports are converted to use the new VFS code.

hosaka, 2017-01-27:
This is great, thanks for your work on the new FatFS version! I am going to close this.

closed 2017-01-27 by hosaka