micropython/micropython #2557: esp8266: Support for 512K modules

pfalcon, 2016-10-24:
I pushed initial support in https://github.com/micropython/micropython/commit/2550d7dfd2e9032474a33d8208868f42680c5863 . Builds, untested. There's very space left still, so hardly can add a big app as a frozen module/bytecode, nor can store user data in raw flash. So, we need to keep disabling more features. This ticket is open to discuss these matters.

kfricke, 2016-10-25:
What are you estimates savings for disabling float support? Or is it off already?

hosaka, 2016-10-25:
Is onewire protocol crucial for this?

pfalcon, 2016-10-25:
> What are you estimates savings for disabling float support? Or is it off already?

No, 512K doesn't offer advanced (e.g. DB modules) and fringe features (slice module attributes), but all core MicroPython features are there. I don't think that disabling float is realistic, but if someone will be serious about that, I'm sure they will present numbers and argumentation.

pfalcon, 2016-10-25:
> Is onewire protocol crucial for this?

Should be nice topic to discuss on the forum. Though I may guess that the answer will be "yes" or "no" depending on whom you ask. That's the problem of 512K port - there's unlikely a single configuration possible which will suit everyone, so this port will be relevant to advanced users who will make/maintain their own config. All in all, the 512K variant is given out for maintenance to its users. If they are interested to cooperate and find consensus, then various things are possible.

dpgeorge, 2020-06-09:
The 512k board target for esp8266 is now built by `make BOARD=GENERIC_512K`.  It works and the intention is to maintain this target.

closed 2020-06-09 by dpgeorge