micropython/micropython #2560: FreeDos build needs update

pohmelie, 2016-10-25:
Tried to make FreeDos build on dev version and got some issues:

- DJGPP crashes on linking with extra gcc [LDFLAGS_ARCH](https://github.com/micropython/micropython/blob/master/unix/Makefile#L69) for FreeDos build. Is it good enough to add FreeDos extra check?
```makefile
ifeq ($(UNAME_S),Darwin)
CC = clang
# Use clang syntax for map file
LDFLAGS_ARCH = -Wl,-map,$@.map -Wl,-dead_strip
else ifneq ($(PROG), micropython_freedos)
# Use gcc syntax for map file
LDFLAGS_ARCH = -Wl,-Map=$@.map,--cref -Wl,--gc-sections
endif
```
- [FatFS](https://github.com/micropython/micropython/blob/master/unix/Makefile#L163) should be disabled for FreeDos build. Is there a better solution?
```makefile
ifneq ($(PROG), micropython_freedos)
# FatFS VFS support
LIB_SRC_C += $(addprefix lib/,\
	fatfs/ff.c \
	fatfs/option/ccsbcs.c \
	)
endif
```
- Some minor changes to disable incompatible modules (will be pulled after discussion about above issues).

After fixing this everything build fine.

pfalcon, 2016-10-25:
> FatFS should be disabled for FreeDos build. Is there a better solution?

Most definitely it should be. The main purpose of presence of FatFs is unix build is unit testing, and it's enabled only in coverage build.

pfalcon, 2016-10-25:
> DJGPP crashes on linking with extra gcc LDFLAGS_ARCH for FreeDos build. Is it good enough to add FreeDos extra check?

So what exactly it crashes on? On producing map file? Please have an upstream bug reference ready and include it as a comment, to explain why suddenly freedos build doesn't produce a map.

pohmelie, 2016-10-25:
```
...
LINK micropython_freedos
i586-pc-msdosdjgpp-gcc -o micropython_freedos build-freedos/py/mpstate.o build-freedos/py/nlrx86.o build-freedos/py/nlrx64.o build-freedos/py/nlrthumb.o build-freedos/py/nlrxtensa.o build-freedos/py/nlrsetjmp.o build-freedos/py/malloc.o build-freedos/py/gc.o build-freedos/py/qstr.o build-freedos/py/vstr.o build-freedos/py/mpprint.o build-freedos/py/unicode.o build-freedos/py/mpz.o build-freedos/py/lexer.o build-freedos/py/lexerstr.o build-freedos/py/lexerunix.o build-freedos/py/parse.o build-freedos/py/scope.o build-freedos/py/compile.o build-freedos/py/emitcommon.o build-freedos/py/emitbc.o build-freedos/py/asmx64.o build-freedos/py/emitnx64.o build-freedos/py/asmx86.o build-freedos/py/emitnx86.o build-freedos/py/asmthumb.o build-freedos/py/emitnthumb.o build-freedos/py/emitinlinethumb.o build-freedos/py/asmarm.o build-freedos/py/emitnarm.o build-freedos/py/formatfloat.o build-freedos/py/parsenumbase.o build-freedos/py/parsenum.o build-freedos/py/emitglue.o build-freedos/py/runtime.o build-freedos/py/runtime_utils.o build-freedos/py/nativeglue.o build-freedos/py/stackctrl.o build-freedos/py/argcheck.o build-freedos/py/warning.o build-freedos/py/map.o build-freedos/py/obj.o build-freedos/py/objarray.o build-freedos/py/objattrtuple.o build-freedos/py/objbool.o build-freedos/py/objboundmeth.o build-freedos/py/objcell.o build-freedos/py/objclosure.o build-freedos/py/objcomplex.o build-freedos/py/objdict.o build-freedos/py/objenumerate.o build-freedos/py/objexcept.o build-freedos/py/objfilter.o build-freedos/py/objfloat.o build-freedos/py/objfun.o build-freedos/py/objgenerator.o build-freedos/py/objgetitemiter.o build-freedos/py/objint.o build-freedos/py/objint_longlong.o build-freedos/py/objint_mpz.o build-freedos/py/objlist.o build-freedos/py/objmap.o build-freedos/py/objmodule.o build-freedos/py/objobject.o build-freedos/py/objpolyiter.o build-freedos/py/objproperty.o build-freedos/py/objnone.o build-freedos/py/objnamedtuple.o build-freedos/py/objrange.o build-freedos/py/objreversed.o build-freedos/py/objset.o build-freedos/py/objsingleton.o build-freedos/py/objslice.o build-freedos/py/objstr.o build-freedos/py/objstrunicode.o build-freedos/py/objstringio.o build-freedos/py/objtuple.o build-freedos/py/objtype.o build-freedos/py/objzip.o build-freedos/py/opmethods.o build-freedos/py/sequence.o build-freedos/py/stream.o build-freedos/py/binary.o build-freedos/py/builtinimport.o build-freedos/py/builtinevex.o build-freedos/py/modarray.o build-freedos/py/modbuiltins.o build-freedos/py/modcollections.o build-freedos/py/modgc.o build-freedos/py/modio.o build-freedos/py/modmath.o build-freedos/py/modcmath.o build-freedos/py/modmicropython.o build-freedos/py/modstruct.o build-freedos/py/modsys.o build-freedos/py/moduerrno.o build-freedos/py/modthread.o build-freedos/py/vm.o build-freedos/py/bc.o build-freedos/py/showbc.o build-freedos/py/repl.o build-freedos/py/smallint.o build-freedos/py/frozenmod.o build-freedos/py/../extmod/moductypes.o build-freedos/py/../extmod/modujson.o build-freedos/py/../extmod/modure.o build-freedos/py/../extmod/moduzlib.o build-freedos/py/../extmod/moduheapq.o build-freedos/py/../extmod/moduhashlib.o build-freedos/py/../extmod/modubinascii.o build-freedos/py/../extmod/virtpin.o build-freedos/py/../extmod/machine_mem.o build-freedos/py/../extmod/machine_pinbase.o build-freedos/py/../extmod/machine_pulse.o build-freedos/py/../extmod/machine_i2c.o build-freedos/py/../extmod/machine_spi.o build-freedos/py/../extmod/modussl_axtls.o build-freedos/py/../extmod/modussl_mbedtls.o build-freedos/py/../extmod/modurandom.o build-freedos/py/../extmod/modwebsocket.o build-freedos/py/../extmod/modwebrepl.o build-freedos/py/../extmod/modframebuf.o build-freedos/py/../extmod/fsusermount.o build-freedos/py/../extmod/vfs_fat.o build-freedos/py/../extmod/vfs_fat_ffconf.o build-freedos/py/../extmod/vfs_fat_diskio.o build-freedos/py/../extmod/vfs_fat_file.o build-freedos/py/../extmod/vfs_fat_lexer.o build-freedos/py/../extmod/vfs_fat_misc.o build-freedos/py/../extmod/utime_mphal.o build-freedos/py/../extmod/moduos_dupterm.o build-freedos/py/../lib/embed/abort_.o build-freedos/py/../lib/utils/printf.o build-freedos/main.o build-freedos/gccollect.o build-freedos/unix_mphal.o build-freedos/mpthreadport.o build-freedos/input.o build-freedos/file.o build-freedos/modmachine.o build-freedos/modos.o build-freedos/modtime.o build-freedos/moduselect.o build-freedos/alloc.o build-freedos/coverage.o build-freedos/fatfs_port.o build-freedos/build-freedos/_frozen_upip.o build-freedos/lib/mp-readline/readline.o build-freedos/lib/timeutils/timeutils.o   -Wl,-Map=micropython_freedos.map,--cref -Wl,--gc-sections -lm 
collect2: fatal error: ld terminated with signal 11 [Segmentation fault], core dumped
compilation terminated.
../py/mkrules.mk:115: recipe for target 'micropython_freedos' failed
make[1]: *** [micropython_freedos] Error 1
make[1]: *** Deleting file 'micropython_freedos'
make[1]: Leaving directory '/home/poh/tmp/upython/micropython/unix'
Makefile:251: recipe for target 'freedos' failed
make: *** [freedos] Error 2
```
But map file generated (don't really know if it is finished).
[micropython_freedos.zip](https://github.com/micropython/micropython/files/552007/micropython_freedos.zip)

pfalcon, 2016-10-27:
@pohmelie : So, to not leave this hanging on a half-word - as a freedos port maintainer, you would need to come up with the best way to resolve all this. Recommendation from a micropython maintainer is above (make sure it's reported upstream, then maybe in some half year it's - surprise - fixed), but we unlikely will be able to help much, as we're busy with esp8266 port, etc., so please use your best judgement, the aim is of course have a working freedos port. Thanks.

prusnak, 2016-11-13:
@pohmelie using the fix commit in my [freedos branch](https://github.com/prusnak/micropython/tree/freedos) I was able to build working binary of micropython that works in dosbox.

pohmelie, 2016-11-13:
@prusnak, I can't check it right now, but if so, then I think it is a good place for pull request. :+1:

prusnak, 2016-11-13:
@pohmelie I have no idea whether

```
MICROPY_PY_*=0
```

macro definitions belong to `unix/Makefile` `freedos` target or they should be put in `unix/mpconfigport_freedos.h`. I don't have time nor motivation to deal with this, so I think it would be better if you took the changes and tried to upstream them as you are the FreeDOS maintainer (at least that's what I learnt above).

Also the following:

```
#define MICROPY_ENABLE_EMERGENCY_EXCEPTION_BUF 0
```

is included in `unix/mpconfigport_freedos.h`, because there is broken strict-aliasing in `py/objexcept.c` on lines 336-338, which deserves a proper fix, but I don't have time for this either.

pohmelie, 2016-11-13:
@prusnak, ok, I will try your fix ASAP.

pohmelie, 2016-11-16:
@prusnak 
Ok, I tested your branch and have no success. Since on clean branch with djgpp 6.2.0 I have:
``` bash
$ PATH=/home/broomrider/tmp/upython/djgpp-6.2.0/bin/:$PATH make freedos
...
CC ../py/formatfloat.c
In file included from ../py/mpconfig.h:43:0,
                 from ../py/formatfloat.c:27:
/home/broomrider/tmp/upython/djgpp-6.2.0/lib/gcc/i586-pc-msdosdjgpp/6.2.0/include-fixed/math.h:210:15: error: expected identifier or '(' before 'double'
 extern double trunc(double);
               ^
./mpconfigport_freedos.h:83:27: error: expected ')' before '?' token
 #define trunc(X) ((X > 0) ? floor(X) : ceil(X))
                           ^
/home/broomrider/tmp/upython/djgpp-6.2.0/lib/gcc/i586-pc-msdosdjgpp/6.2.0/include-fixed/math.h:216:15: error: expected identifier or '(' before 'double'
 extern double round(double);
               ^
./mpconfigport_freedos.h:81:27: error: expected ')' before '?' token
 #define round(X) ((X > 0) ? floor(X + 0.5) : ceil(X - 0.5))
                           ^
/home/broomrider/tmp/upython/djgpp-6.2.0/lib/gcc/i586-pc-msdosdjgpp/6.2.0/include-fixed/math.h:240:14: error: expected identifier or '(' before 'float'
 extern float truncf(float);
              ^
./mpconfigport_freedos.h:84:28: error: expected ')' before '?' token
 #define truncf(X) ((X > 0) ? floorf(X) : ceilf(X))
                            ^
/home/broomrider/tmp/upython/djgpp-6.2.0/lib/gcc/i586-pc-msdosdjgpp/6.2.0/include-fixed/math.h:243:14: error: expected identifier or '(' before 'float'
 extern float roundf(float);
              ^
./mpconfigport_freedos.h:82:28: error: expected ')' before '?' token
 #define roundf(X) ((X > 0) ? floorf(X + 0.5) : ceilf(X - 0.5))
                            ^
../py/mkrules.mk:47: recipe for target 'build-freedos/py/formatfloat.o' failed
make[1]: *** [build-freedos/py/formatfloat.o] Error 1
make[1]: Leaving directory '/home/broomrider/tmp/upython/micropython/unix'
Makefile:225: recipe for target 'freedos' failed
make: *** [freedos] Error 2
```

It looks like there is something wrong with defines at [lines 81-84 of micropython_freedos.h](https://github.com/prusnak/micropython/commit/9d63033d0937f1f52f530ed21cc5d42461745db5#diff-6f88993b55d4389243952664f61dc129R81)

And I have same segfault on linker after I comment those lines. How did you build your binary?

prusnak, 2016-11-16:
I think I was using old version of djgpp. That's why I needed to define round[f]/trunc[f] stuff and others.I might look into upgrading it and looking into it, but no promises.

prusnak, 2016-11-16:
I was able to update DJGPP to 6.2.0 and reproduce the crash.

It is `-Wl,--gc-sections` that is causing it. Updated my commit to include the following for freedos target:

`LDFLAGS_EXTRA='-Wl,--no-gc-sections' `

pohmelie, 2016-11-16:
@prusnak 
So there will be both `--gc-sections` and `--no-gc-sections`? Is it ok?
https://github.com/prusnak/micropython/blob/b545dca3c6f45440f2dd12f869f673118210e3b7/unix/Makefile#L75
https://github.com/prusnak/micropython/blob/b545dca3c6f45440f2dd12f869f673118210e3b7/unix/Makefile#L230

prusnak, 2016-11-16:
Yes, latest on the command line wins. Feel free to fix more correctly if needed, but this works for me and does not increase complexity of makefile too much.

pohmelie, 2016-11-17:
pull #2644

prusnak, 2016-11-17:
Let's close this issue if there is a pull request for this ...

closed 2016-11-17 by pohmelie