micropython/micropython #2563: ESP8266: Unexpected empty packets

robert-hh, 2016-10-26:
Hello all,
I have a strange observation in my little task of making an ftp server on ESP8266, which runs in th background. If I send a file to the server, I receive an empty readline() on the command session, if the file exceeds a certain length. This would not be a problem, if not some ftp clients do not properly leave sending the QUIT command. Instead they just close the command session, which is signalled to the server as an empty readline(), and therefore the server cannot simply ignore empty readlines().
I have attached a version of the sever, which is reduced as much as possible. You have to test it with command line ftp, where it accepts 'open', 'put' (in passive mode) and 'bye'. The user name does not matter.
Obviously, the whole matter is not too critical.

[tftp.zip](https://github.com/micropython/micropython/files/553810/tftp.zip)

pfalcon, 2016-10-27:
The best quick suggestion I can give is - try first to write well-working "foreground" server. To give an idea what you're getting into otherwise, background server necessarily needs to work with non-blocking sockets, and .readline() method is not valid for non-blocking sockets (or more specifically, its behavior is different from a blocking version and not what an asynchronous networking programmer novice would expect).

pfalcon, 2016-10-27:
The bugreport is otherwise not very clear, what does for example "receive an empty readline()" means?

robert-hh, 2016-10-27:
Hello Paul. Thank you for responding. I have a foreground server which does not show that phenomenon. 
The 'background' server uses the callback mechanism you invented for weprepl, and the readline() call is the point where in the command sessions callback function the next data from the command session is read, usually a ftp command. If a ftp client just closes the ftp command session, it gets visible at that readline by receiving an 0 length data item. Unfortunately, this sometimes also happens when the data session used in the ftp PUT command is closed, after the data is transferred. Unfortunate, because it happens, and, because not always.
I looked also the actual web traffic and compared it to the traffic with an debian ftp server. Quite different in detail, but the general structure is the same.

robert-hh, 2016-10-29:
To be more clear: I have a server scrips on esp8266 which communicates to a PC client with a tcp session (command socket). The client send commands, which the server processes. The sever code is activated through the callback mechanisms. For one of the commands the server accepts or opens another tcp session for data xfer (data socket). If that session is closed by the client, the server scripts gets a signal on the command socket, that this one is closed. That's unexpected. I found a way now to avoid that by 'shooting first', meaning. closing the data socket before the client does so, but thats not robust and may only work in my set-up.
I do not know whether this is the behavior I should expect, or if that is a glitch.

closed 2016-11-07 by robert-hh