micropython/micropython #2566: The slice type is present but inaccessible

nickovs, 2016-10-28:
It seems that the `slice` type defined in objslice.c does not have it's name exported and as a result can not be referenced. This makes it hard to define an implementation of `__getitem__()` that properly handles slicing, since you can't easily test if the index passed to you is a slice or not. It is possible to kludge this with something ugly like:
```
    class __X:
        def __getitem__(self, x):
            return x
    slice = type(__X()[:]
```
but surely it would be easier to just add the existing class to the set of built in names.

As it happens, even when you do the hack listed above the resulting type can not be instantiated. If you try `s = slice(2,10,4)` you get the error `TypeError: cannot create 'slice' instances`. This is less of a problem (since it's rarely needed) but it is still a deviation from Cpython.

dpgeorge, 2016-10-28:
The slice builtin was exposed very recently, see bdb0d2d0bc146cc043d9bd0c066446d56266413a.

> the resulting type can not be instantiated

Yes that's a known limitation.

nickovs, 2016-10-28:
Thanks. That's very helpful.

Is the limitation about instantiation by design or a bug? `mp_type_slice` seems to have a valid `make_new` function pointer so I don't see why `type_call` would be complaining.

dpgeorge, 2016-11-06:
> Is the limitation about instantiation by design or a bug?

It's not a bug, more of an implicit design decision: it wasn't implemented initially and it was never found to be needed.

> mp_type_slice seems to have a valid make_new function pointer so I don't see why type_call would be complaining.

No, there is no make_new for slice.  If it existed it would be similar to range's make_new.

I'm closing this issue because slice type is present as a builtin.

closed 2016-11-06 by dpgeorge

pfalcon, 2016-11-06:
@nickovs : There's also a document which describes MicroPython philosophy ("zen" in Python terms ;-) ) and what kind of compromises MicroPython users have to accept to be able to enjoy it: https://github.com/micropython/micropython/wiki/ContributorGuidelines

nickovs, 2016-11-06:
Thanks. That makes sense. Given the exceedingly limited utility of being able to construct slices, and the work-arounds should anyone ever actually need to do it, it doesn't seem to be worth the few dozen bytes to put it in.