micropython/micropython #2568: unstable MQTT on ESP8266 (4+ days)

phieber, 2016-10-28:
Hi,

I was successful creating multiple publish and subscriber ESPs using the lib. However, when running the code for several days, the ESP seems to hang. I wonder if anyone was successful running MQTT code in a reliable manner with the current library for multiple weeks/month/..?
I am using the 32Mbit module, so memory should'n be an issue (anymore).

br
Patrick

peterhinch, 2016-10-28:
I've had the same experience. The MicroPython MQTT library uses blocking sockets with no timeout. On rare occasions probably caused by WiFi disruption these can block forever causing the application to hang. I've raised an issue. One workround might be to use a watchdog timer.

slzatz, 2016-10-28:
As I mentioned in a forum post ([http://forum.micropython.org/viewtopic.php?f=16&t=2525&hilit=slzatz&start=20#p15152)]) I am using MQTT and running into occasionally issues that produce an OSError that I catch in try/except section and I just reset and continue and this has run for days with occasional resets that appear to be related to the WiFi signal.  I assume everyone is pinging the MQTT broker because the other reason that things appear to hang is simply that the broker isn't communicating with the client because it hasn't been pinged within the timeout.

phieber, 2016-10-28:
@slzatz I am using the same approach when subscribing. However, it seems that in pub only cases, the ESP seems to hang. Timeout cannot be an issue as I am sending a heartbeat message every 5 seconds.
I've changed my code to qos=0 to see whether it is (more) stable...
Watchdog is a good idea, though ;)

slzatz, 2016-10-28:
> I've changed my code to qos=0 to see whether it is (more) stable...
Yes, should have added that everything I am doing is qos=0.

kfricke, 2016-10-28:
My Sonoff smart switches are acting the same. After several hours/days they do reboot caused by an hanging MQTT connection. I do only notice it because of the relay switching when doing so and based on the fac that they are of course not connected to mains and serial at the same time, i can not provide detailed console output.
I will try to set up a dummy module and keep an eye on the serial output using screen. May take some time...

torwag, 2016-10-28:
@kfricke why not connecting main and serial together ;) (Just kidding, this might be deadly) 
I did not check the power supply of the sonoff units in great detail but it looks like a flyback converter, you might get away by applying a DC voltage to the main plug (which is NOT connected to main). Usually flyback power supplies will start to run from a certain voltage and they do not care whether its is ac or dc. That could be useful for debugging without frying ourself.

torwag, 2016-10-28:
I also had hang-ups from time to time. My simple explanation was the fact that the MQTT might interferes with the housekeeping of the wifi on the esp. If the MQTT is blocking, would the wifi housekeeping still work? It might be just a very rare case of execution clashing. If a blocking MQTT starts at the same time as the wifi housekeeping, it might be that the wifi connection get lost, and the blocking never resolves. Just an idea

peterhinch, 2016-10-29:
I should add that this ticket should really have been raised against micropython-lib, where ticket 103 would have been spotted.

@phieber FWIW I think qos=0 will reduce but not fix the problem. It eliminates one of the failure modes outlined in the ticket which is a missing PUBACK. It will not fix the blocking sockets.

phieber, 2016-10-29:
Closing this one, although issue is not fixed. We should use the micropython-lib issue as mentioned by @peterhinch

pfalcon, 2016-10-29:
> I should add that this ticket should really have been raised against micropython-lib, where ticket 103 would have been spotted.

Well, if it goes beyond issues with mqtt module and "my signal is weak" (can't help much with the latter I'm afraid) into "port is unstable". However, there're only anecdotal reports so far. E.g.:

> the ESP seems to hang

What does that mean? What was the last output? Can you Ctrl+C it? If you can, what import port_diag tells?

> I tried this using example_sub_robust.py minimally modified to connect with my server. I published a couple of entries using mosquitto_pub on a PC to verify the code on the ESP8266 was working correctly, and then killed the broker process on the server. The ESP8266 immediately rebooted. This behaviour seems repeatable.

Is that all? Then I can tell that when I developed it I of course tested it and it worked as expected - reconnect, no reboots. Where do we go from here? We go to reporting the exact version with which it happened (and probably worth providing ESP8266 SDK version, not just uPy version, as we switched number of SDK for this 6+ months already). And then to exact, step by step description of hwo to reproduce issue. Because if we do it your way, me my way, we'll end with the same results - you get reboots, I normal work. (Then, say, in few months you'll report that you had a broken power supply).

> The MicroPython MQTT library uses blocking sockets with no timeout.

This was reported quite some time ago, I'd assume you already tried it, and if it helped, would have submitted a patch. Or would you prefer maintainers to do that? Ok, it's in our queue, just as all other issues (270+ tickets, 70+ pull requests).

Anyway, we've accumulated some bugreports regarding network functionality. My plan is to finish current feature work (specifically: leading OTA work to some milestone, and doing integration testing for possible heap size increase), and I'm going to work on bugreports for some time. But we really need help by getting high-quality bugreports, or otherwise it will be a lot of time spent walking blind-fold. And actually, there're enough qualified people he could go beyond that if they liked. Because well, it's 10% of networking code is in MicroPython and trust me, we with Damien already spent a lot of time debugging that, and 90% in vendor binary blobs, where there're hundreds of assertions leading to busy-loops (and eventually, WDT resets), and an issue you experience may be peculiar to you router, WiFi signal propagation, antenna, module and not reproducible by anybody else. So, if you experience issue, try to get to the bottom of it (all the output while it happens, how to reproduce it with the highest probability (exact steps), etc.)

Thanks for help!

peterhinch, 2016-10-30:
@pfalcon I would have loved to have submitted a reproducible test case, but after three months (intermittent) work towards that end failed to achieve it and reluctantly gave up. It's difficult artificially to create the conditions which cause a hang. The robust module works fine if you switch the access point on and off but it still is prone to hanging. You can sometimes provoke it by gradually moving out of range, but it's far from reproducible. Deliberate provocation aside, I failed to achieve reliable MQTT communications between my living room and the AP on the floor above, with the ESP8266 hanging every few hours. This on a WiFi network which is reliable in normal use. Timeouts helped but did not fix the problem.

I offered the best analysis I could of the failure modes in my original ticket. I'm no network expert, but I think these issues are best resolved by careful analysis of the problem by someone who is, and has an understanding of the limitations of WiFi compared to wired networks (you?). Of its nature it isn't going to yield a reproducible test case, in my opinion.

I appreciate you guys are busy and am happy to wait, but I've got as far as I can with it.

pfalcon, 2016-10-30:
> I'm no network expert, but I think these issues are best resolved by careful analysis of the problem by someone who is, and has an understanding of the limitations of WiFi compared to wired networks (you?).

You know there're server farms which serve high-profile sites like google.com, etc. They don't use WiFi, but a wired connection. Does that ring a bell? Let me approach it from another side: any network by definition non-reliable. Like, it's not some abstract "it may fail", it will and does fail regularly. In this regard, any wireless network is very unreliable by any reasonable definition of "reliable". And well, it's not designed to be reliable. It's assumed there's a user behind it who will reconnect, go reboot a router, power it off, wait 5min and power on again, phone IPS, etc.

Final story for today - I left my module doing downloads of large file (that's our standard integration test - downloading 20MB without errors is not too bad for a tiny piece of something esp8266 is) to see how far it can go. It downloaded 300MB, which isn't bad at all IMHO. But then WiFi connection dropped and esp went into connect-associate-immediate-disassociate-repeat loop. Am I surprised? Not at all. I get the *same* about every 10th time (that's very high probability!) I resume my laptop. Nothing helps then except suspending and resuming again. Neither trying to deactivate STA interface and then activate and reconnect helped esp. Reset cleared it.

> You can sometimes provoke it by gradually moving out of range

Sorry, but no wireless network can work with poor signal. It's like asking "I don't have electricity and my TV doesn't work, what should I do?" or "I put half of the voltage into TV and it doesn't work, what should I do?". The answer is obvious - get the electricity! The same with wireless networks - get the strong signal. I figure that you'd want your app not to hang in such conditions, we'll see what we can do, not everything depends on us. But mind that your app will never work well in the conditions above anyway.

Generally, my concern as a MicroPython maintainer is to separate issues into those pertaining to (caused by) MicroPython and those of the underlying platform (or just outside, because your WiFi performance depends on your router and your neighbours). This is inherently hard, and that's what I call for help of users and bug reporters with, otherwise, we will have to try to reproduce it all ourselves, which, as you know, hard, and there will be slow progress.

peterhinch, 2016-10-31:
I have some experience of radio communications and appreciate that radio links don't work when out of range. As a recent example I wrote a couple of drivers to simplify the usage of the NRF24L01+ under MicroPython. They use timeouts to guarantee a return and I would regard indefinite blocking under any circumstances as a bug. I used various test procedures to verify this behaviour.

You said elsewhere
> On a personal note, I'd wish you write not your cooperative scheduler module, but your mqtt module ;-).

While I get the smiley, I wrote the scheduler over two years ago. Recent minor improvements took little time. Alas MQTT, despite three months' fairly determined effort, defeated me. Alas I lack professional network experience or training; I even lack prior amateur experience. I reluctantly concluded that making MQTT truly robust needs someone with a solid theoretical background in networking. Sometimes in life you have to admit defeat.

I notice you have a umqtt-threaded branch which perhaps will provide nonblocking behaviour.

torwag, 2016-11-04:
Hi,
does this not condense down to the question whether the lib itself should take care of most of the possible failure scenarios (weak signal, router, hardware, etc.), or if the user itself should take care of it?
As I understood Peter, he does not want a fix of an (possible) internal bug, but something which deals with possible external errors in a sensible, reliable way. 
Thus, for me this is not an issue but more a feature request.

peterhinch, 2016-11-04:
It depends whether you view permanently blocking behaviour as a bug, or the absence of a feature. It certainly makes writing applications difficult...

mkarliner, 2016-11-04:
I'm hesitant to weigh in here, but it seems to me that neither Peter nor anyone else is suggesting that MQTT should magically fix all network problems, merely raise an exception in the case of an error or in the case of 'robust' attempt automatic reconnects.  As a good example, I have an open issue that is that a publish in 'simple' to a session where I have stopped the broker does not raise an exception.  As an application programmer, I cannot manage this case, as I simply don't know about it.

phieber, 2016-11-04:
Dealing with exceptions or other external failuers is one thing. Waiting for a blocking socket to return when it doesn't throw an error and when not beeing able to use a watchdog, in case of esp8266, is another thing.
How can a programmer catch those cases? It all comes down whether the lib is reliable (aka returns or throw something) or not
A lib which will work a few days is not practical for IoT devices
(My 2 cents)

pfalcon, 2016-11-04:
@phieber : Somehow I missed that you're from IBM. Welcome aboard. MicroPython project is always open to contracts from the commercial entities to fast-track any functionality or bugfixes. I'm at all available for hire for full-time work on MicroPython. Don't hesitate to contact us.

torwag, 2016-11-04:
@mkarliner Yep, I am aware that a lib can't fix stuff automagically. But raise exceptions are a way to deal with problems by just sending them upward.  
Regarding the blocking socket. I did not understand that fully, maybe @peterhinch @kfricke can give me some more info here. I understand that the socket blocks and hence the python instance. However, does this effect the wifi routines? I thought this micropython is running "semi-parallel" to the wifi stack based on the underlying RTOS. Hmmm, guess I have to understand better how micropython is integrated in the esp, infos on the net are not exactly easy to find. Maybe @pfalcon can point to some sources?

pfalcon, 2016-11-04:
> I understand that the socket blocks and hence the python instance.

It blocks because it's default socket behavior and how TCP is designed. Want *your* sockets unblock? Set timeout on them. We purposedly provide as much as possible written in Python so it was transparent both for reading and writing. When you have MQTTClient instance, say `foo`, your socket is `foo.sock`. Do with it whatever you want. Want somebody else to set timeout on *your* sockets? It's coming too, as soon as previous wishes will be granted - OTA, uasyncio, sensor drivers, video tutorials (even during the campaign, some cold-headed backers warned the other that "these guys are developers, don't bother them with that video stuff", gosh, were they true).

Sorry guys, but sometimes I'm really stumped. I thought we're programmers here, maybe some hobbyists, but programmers, not cartoon watchers. How it's supposed to work is that we lay the paths, users go by them as far as they want - guided by existing examples and documentation, but not each individually by hand. Ever since ESP8266 campaign started I never had a chance to work on any of my side projects no related directly to EPS8266. It's close to 2am here it's few days a week like that, week over week, month over month. And I probably drop it and won't sit by the screen 4th weekend in row, 3 is enough, don't you think? It won't be like that forever - the tentative project duration is 1 year, more than half is over, and we'll do as much possible with Damien for remaining time. But after that, esp8266 port is yours, community, for taking and maintaining. So you'd better start now to learn how to make it work like you want. And in large part, that's actually how it works (bunch of latest features of MQTT client were contributed by users). But sometimes - SOCKETS BLOCK!! OMG!!!11 http://docs.micropython.org/en/latest/esp8266/library/usocket.html#usocket.socket.settimeout 

> Maybe @pfalcon can point to some sources?

With pleasure I can. Sources are right here, on the github. Click "Code" tab at the top of the page.

peterhinch, 2016-11-05:
@mkarliner The socket blocks but the WiFi continues to work.

> I thought we're programmers here ... not cartoon watchers

@pfalcon I can only speak for myself. I'm a programmer, albeit one 12 years retired. I appreciate that blocking sockets block and radios aren't reliable. So I produced a version of mqtt robust with timeouts on sockets and with retransmission in the event of a PUBACK not being received within a timeout. If the outcome had been robust I'd have submitted a PR. It wasn't. Nor was it in anyone's interest to offer a half-baked solution. After a lot of work I realised I was making no further headway; and that the reason for this was a lack of network programming knowledge. Hopefully someone with the necessary experience will succeed where I failed.

My wife would tell you I'm a determined ('stubborn') SOB and don't give up on projects lightly :)

torwag, 2016-11-05:
@pfalcon I think it might be a good idea to take a weekend off and have some fun with family and friends. Sorry it was not in my attention to bother you. As you can see, my post was similar late then your reply. It was at the end of a very intense working day and this post was my "relax" time to come down a bit, I might have not explained my point correctly, I might have used some wrong terms. English is not my native language and I do not make a living with programming. Thus it might happen, that thinks sound strange.  Anyhow,  I followed this MQTT discussions on several paths for some time now and made my own experience. 
Heck, I know sockets usually block and I played around with timeouts, watchdogs, etc. My question was, does this blocking behaviour also interfere with the WIFI-stack on the esp? I was wondering if a unfortunate timing constellation of a socket request within mp and some operation of the wifi stack (which as I understood, is hard to get by,  as it is not well described?!) could result in loosing the wifi connection resp. setting the wifi stack in an unknown state and hence resulting in a hang-up, reboot, etc.
Thus my question was, could the socket operation block in some obscure way the wifi stack? Not whether a socket is blocking in general. 

As to answer this for myself to learn further and to understand, I simply asked you if you could give me some idea where I could read how exactly mp is embedded in the esp.  Well you pointed me to the code button ;) fair enough. I know its buried there. Simply was hoping that you might have some infos or webpages which e.g. you might have used when you started to work on the esp port. The Internet is  sometimes so full of infos, that you often lose sight of the wood for the trees.

You see, I can feel your frustration that you spend 24/7 for weeks on this and suddenly stupid people like me pop up and asking questions which sound like the very basics to you. Being a academic teacher by profession I stay very often on the same side as you do here.  I know you shout out for others to help. Thus, please let us try to understand, come into the topic and see how can we help. 
Otherwise, it will be hard to acquire new people to contribute. In summary, please don't get upset, take your time to relax, nobody would blame you, at the final end it should be a fun ride for you and not being frustrating.

Actpohomoc, 2016-11-08:
I'm using MQTT robust version with simple one. I have corrected it a little and put it to the module folder and create a custom firmware based on 1.8.5. Just add extra lines: [https://github.com/micropython/micropython-lib/pull/117](https://github.com/micropython/micropython-lib/pull/117)
I use client.check_msg() to check messages from MQTT and send info about water leakage and temp, humidity info from dht sensor and free memory value.
The only problem always was memory. So I have read carefully the docs and split my program to 3 or 4 files and use import. So my main.py:
```
import app

if __name__ == '__main__':
    app.setup()
    try:
        app.main_loop()
    finally:
        app.teardown()

```
It works with Home Assistant and no problem at all for the weeks. I use chinese Wemos D1 board.
 
Actually I don't really understand this issue. I think that many depends on the micropython program.  So it should be published to analyze it. Maybe problem is in the code?

peterhinch, 2016-11-09:
@Actpohomoc Robustness depends on the quality of the WiFi connection so is variable from site to site. Incidentally, running on a microcontroller, you don't need the
```python
if __name__ == '__main__':
```
test. This applies when running under an OS to run code only when a Python module is run from the OS. It won't run if the module is imported. This distinction doesn't apply on a microcontroller.

EddieParis, 2018-03-17:
Hi all, some hope in this discussion ! I use the robust mqtt and got some trouble, and found interresting things.

Today I faced the blocking issue when looping on MQTT check_msg() call. I noticed that it occured when the wifi was reconnecting which occurs a lot in this particular ESP module (really poor one ??). It always get stuct in the sock.recv() line 168.
The robust part detects the wifi issue and automatically reconnects which is fine :-)
But when we are in check_msg the socket has to be in nonblocking mode, and robust does not set the newly created socket he creates when reconnecting to non blocking mode. Hence the program freeze when it get back to the main loop of wait_msg().
So I added a call to unblock socket (needs to be done in a clean way) and now I have another issue. I don't get any new message. This is because from the mqtt broker it is a new connection and then the subscribtions are lost. I did not try yet with keping track of them but I will asap.

So to summarize :
1 the socket must be set to non blocking mode if reconnection occurs in the context of a check_msg()
2 the subscriptions must be re sent to broker to continue receiving messages.

peterhinch, 2018-03-18:
There are a number of limitations in the official libraries when it comes to coping with failing Wifi signals, especially when operating near the limit of reliable radio communications. You will find information and an asynchronous version of MQTT [here](https://github.com/peterhinch/micropython-mqtt.git).

jonnor, 2024-07-19:
The linked library is a good solution for this problem. I think we can close this issue, as there is not sufficient updated information to reproduce a particular issue here.

closed 2024-07-19 by dpgeorge