micropython/micropython #2572: extmod/modframebuf.c framebuf1_text() does not clear pixels

peterhinch, 2016-10-29:
This section of the code looks wrong:
```C
                    if (vline_data & 1) { // only draw if pixel set
                        if (0 <= y && y < self->height) { // clip y
                            uint byte_pos = x0 + self->stride * ((uint)y >> 3);
                            if (col == 0) {
                                // clear pixel
                                self->buf[byte_pos] &= ~(1 << (y & 7));
                            } else {
                                // set pixel
                                self->buf[byte_pos] |= 1 << (y & 7);
                            }
                        }
```
If the pixel is not set, it should surely clear down the pixel in the framebuf as we don't know its prior contents. Further, if we happen to be drawing with col==0 to a part of the framebuf which is already zero, then nothing will be drawn, because in that circumstance the code never sets a pixel.

I hope I'm right on this observation from reviewing the code: I haven't had the opportunity to try it since my SSD1306 was DOA.

mkarliner, 2016-10-29:
I can confirm this is the case. overwriting text requires a pixel by pixel erase first which is incredibly slow.

peterhinch, 2016-10-29:
Thanks for verifying that.

dpgeorge, 2016-11-02:
The idea is that the text rendering does very simple transparency.  If you need a blank background then you can clear a region before writing text over it.  But, point taken that there's not a fast way to fill a rectangular region.

mcauser, 2016-11-11:
I was experimenting with the text() method by adding a 2nd colour for the background colour, defaulting to transparent (current "skip" behaviour). When set, would fill all pixels in the loop. It was rather slow pixel-by-pixel. I've shelved the idea while people are reorganising the underlying framebuffer data structure. #2364 Rendering a rect beforehand to clear the region sounds like a better approach, as long as we can make them really quick.