micropython/micropython #2573: Windows port needs to switch to --gc-sections

pfalcon, 2016-10-29:
@stinos: FYI, it's impractical to try to ifdef everything any longer (well, not now), if it can be done automatically. Please clarify MSVC situation in that regard.

stinos, 2016-10-31:
Which ifdefs are you referring to? The msvc linker does have flags to remove unused code and they are on by default, is that what's needed?

pfalcon, 2016-10-31:
> Which ifdefs are you referring to? 

Of config vars from mpconfig.h, and in general.

> The msvc linker does have flags to remove unused code and they are on by default, is that what's needed?

Probably yes, if this removal happens before reporting undefined symbols (that's how --gc-sections work). I.e. if some module module.o contains 2 functions, one user by an application, and one not, then if not-used function depends on another function which is not defined by the app, then it's not an error.

stinos, 2016-11-03:
That's indeed how it appears to be working: for excample adding a static function which isn't used anywhere and depends on an undefined function, both are skipped at link time.

pfalcon, 2017-04-19:
@stinos : Thanks for checking, so let's rely on this feature full-on. E.g.: https://github.com/micropython/micropython/commit/58467709972724ac94f6488fefe6d066bfa1589e#commitcomment-21817258

@dpgeorge : fyi

closed 2017-04-19 by pfalcon

stinos, 2017-04-19:
Is this effectively in use already somewhere reachable by windows port so that we're sure it works?

pfalcon, 2017-04-19:
@stinos: Non conclusive. This ticket was exactly about my concern with usage of this pattern in the core code, where I tried to avoid it because of MSVC concerns, while already using e.g. for baremetal code which isn't compiled for windows. But I can't say how thorough I was at avoiding relying on it in generic code. With this valuation, I assume that hands are untied to rely on it everywhere (unless @dpgeorge wants to re-replay it now ;-) ).