micropython/micropython #2574: stmhal frozen bytecode: bytes instances consume RAM on import

peterhinch, 2016-10-29:
There has been much discussion around storing large volumes of data as bytes objects in frozen bytecode, on the basis that as immutable objects RAM could be conserved by ensuring they were accessed by reference. This behaviour seems to have been lost at some point: it used to work. I have a file ``freesans.py`` which is frozen as bytecode. It is not present on ``/flash`` and there is no SD present (Pyboard V1.0 running firmware built from today's source). The fact that the import succeeds proves that it is the frozen file which is being employed.

Prior to import ``micropython.mem_info()`` reports 368 bytes used. After issuing ``import freesans`` it reports 4784 bytes, an increase of 4416 bytes. The two bytearrays together use 3271 bytes so it seems clear that they are consuming RAM on import. The file contents (bytes definitions severely truncated) are:

```python
version = '0.1'
def height():
    return 19
def max_width():
    return 19
def hmap():
    return False
def reverse():
    return False
def monospaced():
    return False

_font =\
b'\x05\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'\
b'\x00\x07\x00\xfe\x31\x00\x7e\x30\x00\x00\x00\x00\x00\x00\x00\x00'\

_index =\
b'\x00\x00\x11\x00\x28\x00\x42\x00\x62\x00\x82\x00\xb7\x00\xe9\x00'\

try:
    from uctypes import addressof
except ImportError:
    pass

def _chr_addr(ordch):
    offset = 2 * (ordch - 32)
    return int.from_bytes(_index[offset:offset + 2], 'little')

def get_ch(ch, test=False):
    ordch = ord(ch)
    ordch = ordch if ordch >= 32 and ordch <= 126 else ord('?')
    offset = _chr_addr(ordch)
    width = int.from_bytes(_font[offset:offset + 2], 'little')
    if test:
        next_offs = _chr_addr(ordch +1)
        return _font[offset + 2:next_offs], 19, width
    return addressof(_font) + offset + 2, 19, width
```
Unless I'm missing something, the mere act of importing this file should not bring the arrays into RAM.

peterhinch, 2016-11-01:
I've now done a _reductio ad absurdum_ and frozen the following file as bigfont.py as bytecode:
```python
_font =b'\x09\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'\
```
Here is the ensuing REPL session.
```
PYB: sync filesystems
PYB: soft reboot
MicroPython v1.8.1-753-gb97c17e-dirty on 2016-11-01; PYBv1.0 with STM32F405RG
Type "help()" for more information.
>>> import micropython, gc
>>> gc.collect()
>>> micropython.mem_info()
stack: 416 out of 15360
GC: total: 102208, used: 1024, free: 101184
 No. of 1-blocks: 8, 2-blocks: 7, max blk sz: 39, max free sz: 6309
>>> import bigfont
>>> gc.collect()
>>> micropython.mem_info()
stack: 416 out of 15360
GC: total: 102208, used: 5216, free: 96992
 No. of 1-blocks: 13, 2-blocks: 8, max blk sz: 247, max free sz: 6008
>>>
>>> micropython.mem_info(1)
stack: 424 out of 15360
GC: total: 102208, used: 5312, free: 96896
 No. of 1-blocks: 15, 2-blocks: 10, max blk sz: 247, max free sz: 6008
GC memory layout; from 200030c0:
00000: h=hhhhh======================================h=h=hh=h=h=hBMDhh==
00400: hh=hB...Bh=h......h=.h=......................................h==
00800: =====h==========================================================
00c00: ================================================================
01000: ================================================================
01400: ============================================================....
       (93 lines all free)
18c00: ....................................................
>>> len(bigfont._font)
16
>>> 
```
5216-1024 = 4192 bytes being consumed by a 16 byte array? I'm convinced this did not occur in the past.

dpgeorge, 2016-11-02:
On unix port it works fine, only uses about 200 bytes to import bigfont.py regardless of the size of the bytearray in that file.

Those big chunks of contiguous heap in your mem_info(1) output are strange, but it's hard to guess what they are.

peterhinch, 2016-11-02:
I don't know if this helps but if I subsequently import another font the memory usage only increases by 200 bytes or so. It's only the first import which guzzles RAM.

dpgeorge, 2018-08-01:
I had a look at this and I'm pretty sure it is down to qstr allocation.  I was able to reproduce the problem under the following conditions:
1. have a lot of frozen modules, with lots of qstrs (hundreds of them)
2. have nothing in boot.py or main.py (so running them doesn't allocate any memory)
3. import a frozen module (or do anything that would create a new qstr)

This would lead to a few kb of allocated memory, and a large chunk in the output of `micropython.mem_info(1)`, which corresponds to the large qstr pool allocated by the first new qstr.  Then subsequent imports would not increase memory usage by much at all.

The way to see that this is definitely the issue is to use `micropython.qstr_info(1)` which will report how many bytes are allocated to qstrs.

This issue was fixed by #3962, which was merged in 6e5a40cf3ca7bb5de9819d49ff0b7483a44d525c

closed 2018-08-01 by dpgeorge

peterhinch, 2018-08-01:
Test conditions are as you described. I can confirm the problem is now fixed.