micropython/micropython #2575: esp8266: error in building image

robert-hh, 2016-10-29:
When trying to run make with the most recent statem I get an error:
```
CC modutime.c
modutime.c:122:5: error: initialization discards 'const' qualifier from pointer target type [-Werror]
     { MP_ROM_QSTR(MP_QSTR_ticks_add), MP_ROM_PTR(&mp_utime_ticks_add_obj) },
     ^
cc1: all warnings being treated as errors
../py/mkrules.mk:47: recipe for target 'build/modutime.o' failed
```

pfalcon, 2016-10-29:
This should be fixed in the current master.

closed 2016-10-29 by pfalcon

robert-hh, 2016-10-29:
Indeed. I'm just puzzled, because I'm almost sure I did a 'git pull' before the 'make' (and a make in open-esp-sdk, make clean, ....) before I confirmed wrote my note about it.