micropython/micropython #2576: MQTT: c.connect() -> IndexError: bytes index out of range

BigMan200, 2016-10-30:
On a raspian (IP=10.0.0.32) I installed mosquitto:
# service mosquitto status
gives me: [ ok ] mosquitto is running.

An ESP8266 is in the same network:
**FW: esp8266-20161017-v1.8.5.bin**
>>> sta_if.ifconfig()
gives me: ('10.0.0.105', ...)
The ESP8266 is flashed with 

On the raspian-terminal I can also ping my ESP8266:
# ping 10.0.0.105
gives me: 64 bytes from 10.0.0.105: icmp_req=1 ttl=255 time=2.62 ms ...

But I am failing to connect with the MQTT:
>>> from umqtt.simple import MQTTClient
>>> c = MQTTClient("client", "10.0.0.32")
>>> c.connect()
gives me:
Traceback (most recent call last):
File "<stdin>", line 1, in <module>
File "umqtt/simple.py", line 84, in connect
IndexError: bytes index out of range

pfalcon, 2016-10-30:
Well, this one for sure should be raised against https://github.com/micropython/micropython-lib . In the meantime, you can look at the source and see what's wrong: https://github.com/micropython/micropython-lib/blob/v1.8.4/umqtt.simple/umqtt/simple.py#L84

BigMan200, 2016-10-30:
I think the problem is solved:
I have to change  '>>> c = MQTTClient("client", "10.0.0.32")
With  '>>> c = MQTTClient(b"client", b"10.0.0.32")

pfalcon, 2016-10-30:
Well, that might make a difference for CPython, but not for MicroPython, we have relaxed rules regarding str vs bytes, and in many contexts where bytes are required, str can be passed (assumed to have utf-8 encoding).

As https://github.com/micropython/micropython-lib/blob/v1.8.4/umqtt.simple/umqtt/simple.py#L84, the problem is that your server doesn't return valid MQTT response. So, you would need to diagnose your server why it does so (it may be a network problem too, etc.)

pfalcon, 2016-10-30:
From my perspective as a maintainer, the content of this bug is: should this error be caught and and MQTTException be thrown instead? That won't change much - there will still be one error instead of another, and it still will be numerically coded, so user would need to look at the source to figure out what's wrong, just like they need to do now. But that will increase code size, eating out resources from user application (so it will start to fail with a higher probability). Is it worth it? I can't decide myself, experienced users, please comment and vote.

pfalcon, 2016-10-30:
This was resubmitted as https://github.com/micropython/micropython-lib/issues/110

closed 2016-10-30 by pfalcon

sunilsahu39, 2017-07-01:
On a linux machine(ip:192.168.1.8) I have installed the mosquitto broker.
mosquitto broker is working fine on this machine

On esp8266 nodemcu in the same network(ip:192.168.1.6,.....) I tried configure as a client and tried to publish simple message.Still I am getting unexpected error message as below:
network config: ('192.168.1.6', '255.255.255.0', '192.168.1.1', '192.168.1.1')
Traceback (most recent call last):
  File "<stdin>", line 36, in <module>
  File "<stdin>", line 34, in main
  File "<stdin>", line 25, in clientpublish
  File "umqtt/simple.py", line 95, in connect
IndexError: bytes index out of range

The code is:
```
from umqtt.simple import MQTTClient
import dht
import ubinascii
import machine
import network
import time
def do_connect():
    wlan = network.WLAN(network.STA_IF)
    if not wlan.isconnected():
        print('connecting to network...')
        wlan.active(True)
        wlan.connect('dezynetech', '20596169')
        while not wlan.isconnected():
            pass
    print('network config:', wlan.ifconfig())

#d = dht.DHT11(machine.Pin(4))
#d.measure()
#pin=machine.Pin(2,machine.Pin.OUT)
client_id="192.168.1.6"
broker_id="192.168.1.8"
TOPIC1=b"Hello_Msg"
def clientpublish(server=broker_id, topic=b"/foo", data=None):
    c = MQTTClient(client_id, server)
    c.connect()
    c.publish(topic, data)
    sleep_ms(200)
    c.disconnect()
       

def main():
    while True:
        do_connect()
        clientpublish(broker_id,TOPIC1,b"Hello")
if __name__=="__main__":
    main()

```
What should I do to avoid error

peterhinch, 2017-07-01:
This would be better raised in the forum than in the MicroPython bug tracker.

I can see three things wrong here.
1. ``do_connect()`` should just be called once (if that - see below).
 2. I would include a delay between publications.
 3. You need a delay after power up and before your code runs.

If you always connect to the same network you can remove the call to ``do_connect()`` from the code. The ESP8266 remembers the network details and automatically connects on boot. So you only need to call ``do_connect()`` once ever, or after erasing flash. In this case your ``main()`` just needs to loop waiting for ``wlan.isconnected()``, pause for a few seconds, then run a loop making publications with a delay between each.

I've found that where a program runs on power up you need a delay (I use four seconds) before doing anything. I guess the underlying RTOS of the ESP8266 takes a little time to start up.

Finally if you're going to make rapid regular publications I see no reason to keep disconnecting from the broker. I would connect once and leave the connection in place. But this depends on how frequently you publish. If you publish at long intervals (> the broker's keepalive time) then your approach is correct.

Note I'm a user of ``umqtt`` rather than a developer so these suggestions are not official - they are merely based on my empirical experience of using it.

dpgeorge, 2017-07-02:
@sunilsahu39 did you read the full issue at micropython-lib, linked above?  In particular this post about mosquitto version: http://diy.viktak.com/2015/05/installing-mosquitto-for-use-with.html

Tomaubier, 2020-12-31:
I get the same error using a wired version of the ESP32 (the wESP32) when trying to connect from the `main.py` file or the serial prompt (using `screen` or `rshell`). However, executing the same command from the webREPL runs flawlessly. I find this a bit embarrassing, have you ever encountered something like this?
I've provided more detail on this issue on this post: https://stackoverflow.com/q/65520905/9645937?sem=2

dsaad68, 2021-09-29:
> This would be better raised in the forum than in the MicroPython bug tracker.
>
> I can see three things wrong here.
>
> 1. `do_connect()` should just be called once (if that - see below).
> 2. I would [...]

Thanks a lot. This comment helped me immensely.

Saranya-karan, 2024-08-25:
> This would be better raised in the forum than in the MicroPython bug tracker.
>
> I can see three things wrong here.
>
>     1. `do_connect()` should just be called once (if that - see below).
>
>     2. I [...]

iam also facing a same issue..iam trying to send data to aws iot..my error code is  
Traceback (most recent call last):
  File "<stdin>", line 84, in <module>
  File "simple1.py", line 92, in connect
IndexError: bytes index out of range
>>>