micropython/micropython #2578: docs: Introduce "generic" target

pfalcon, 2016-10-30:
I'd like to start experimenting with an idea of having a single, common docset for MicroPython. To not regress existing ports' docs, I propose to introduce a new target for the common doc. Let's think about good name - "generic", "full" or whatever.

hosaka, 2016-10-31:
"Generic" or "common" is probably a good idea!

If I may suggest on the docs refactoring, the navigation column looks frankly cluttered and hard to read for a document format. I kept meaning to mention it on the forum.
[image]

Can we please tidy it up? Anything along the lines of:

- Quick Reference
- General Information
- Tutorials
- Libraries
- Hardware
- Datasheets
- License
- Contents

peterhinch, 2016-10-31:
@hosaka Agreed. The tutorials/language notes in particular are currently rather well hidden.

chrismas9, 2016-11-01:
> Let's think about good name - "generic", "full" or whatever.

What about "Standard" as in:

Standard Documentation
Documentation for Standard MicroPython Vx.x
The port to MyBoard is compliant with the MicroPython Standard Vx.x

This implies that any port that is not compliant with the Standard Documentation is either substandard (eg incomplete) or non-standard (deviates from the standard).

I also think "Official" should be used for ports that are officially supported by Damien, ie pyboard & ESP

dpgeorge, 2016-11-02:
I think "generic" is a bit generic and sounds like it's missing things.  So "full" or "common" or "standard" would be better.  Or even just "docs".

@hosaka if you want to look into how the nav column is generated/specified, and how to improve it, please do :)

deshipu, 2016-11-02:
It would be great if we could gradually move away from the makefile magic for the docs, so that eventually  they could be put on services such as readthedocs (it's not possible right now, due to the need for setting the environment variables). Not necessarily for the main micropython docs, of course, but it could be useful for forks.

pfalcon, 2016-11-02:
> So "full" or "common" or "standard" would be better. Or even just "docs".

I agree that "full" looks the best from user perspective. Anything else is more ambiguous.

pfalcon, 2016-11-02:
> navigation column looks frankly cluttered 

I keep repeating the issue of reworking ToC (I specifically mention PDF version). Please open a separate ticket for that.

chrismas9, 2016-11-03:
> I agree that "full" looks the best from user perspective. Anything else is more ambiguous

The USB IF got into a pickle using "full". They now have Low, Full, High and Super Speed where Full is the second slowest. Once you have "full" it's hard to find a logical name for a bigger, better, faster, etc version.

I anticipate the time when MicroPython will have multiple versions, eg Light, Standard and Advanced, loosely targeting Cortex M0, M4 and M7 class processors or small, medium and large memory devices.

I think this split will help broaden the appeal for commercial use. For example adding streaming I2S to machine in the advanced version would be attractive for Audio and High speed data acquisition applications where I would be happy to use an M7 but I wouldn't want to burden a simple M0 sensor or control node with the extra memory requirements. Also some MCU's don't have some interfaces (eg SD card), so they would not be compliant with Standard MicroPython, but would  be compliant with Light MicroPython. Note that SD card is part of machine and discussions to date indicate that no feature of the current machine should be optional.

If the common Documentation lists whether each feature is mandatory or optional for each level then I don't think it matters whether it is just Documentation or Full Documentation. For example SPI would be mandatory for all levels but I2S might only be mandatory for Advanced. Some features such as CANbus support should probably be optional at all levels due to unavailability on so many MCU's. Please don't create "Full Documentation" in such a way that ports for low end MCU's can't be compliant because they can't implement all mandatory features of the Full Documentation.

I would expect everything currently in machine to be mandatory for Standard MicroPython.

I agree about generic. Premium brands are not associated with generic. Generic reminds me of the plain wrap items on the bottom shelf of the supermarket or something of questionable origin on AliExpress or Ebay

mcauser, 2016-11-08:
Having `Quick reference for the ESP8266` and `MicroPython libraries` pinned to the top makes it a little confusing / "jumpy", when you navigate through the `MicroPython documentation contents` menu.

If you expand MicroPython documentation contents and click on quick ref, both submenus open and you may lose your place in the contents.

Can everything inside the `MicroPython documentation contents` menu be moved to the top level, to avoid pinning/duplication? Not sure how this would impact ports / pdf generation.

In the esp8266 docs, 1/2 of the items are pinned/duplicated.
http://docs.micropython.org/en/latest/esp8266/esp8266_contents.html

```
Quick reference for the ESP8266 <-----\
MicroPython libraries <---------------|-------\
MicroPython license information <-----|-------|---\
MicroPython documentation contents    |       |   |
> Quick reference for the ESP8266 <---/       |   |
> General information about the ESP8266 port  |   |               duplicates
> MicroPython tutorial for ESP8266            |   |
> MicroPython libraries <---------------------/   |
> The MicroPython language                        |
> MicroPython license information <---------------/
```

In the pyboard docs, it behaves differently.
eg. http://docs.micropython.org/en/latest/pyboard/pyboard/quickref.html

Clicking on `MicroPython documentation contents` > `Quick reference for the pyboard` only opens the pinned menu. Not both as with the esp8266 and wipy docs.

pfalcon, 2016-11-11:
> I keep repeating the issue of reworking ToC (I specifically mention PDF version). Please open a separate ticket for that.

@hosaka , Did you have a chance to create a separate ticket for this? This issue will be closed once its subject is resolves, and all unrelated, but important discussion will be lost otherwise. Thanks.

pfalcon, 2017-04-05:
As an update on the original idea: actual practice shows there's never enough time to dedicate to docs, so apparently, we should avoid round-about way of introducing "generic" doc target, and just make the main docset generic (by removing port conditionals, not adding yet another one).