micropython/micropython #2583: unix: SSL client socket, exists unexpectedly upon wrap_sockets().

hosaka, 2016-10-31:
Running a unix port:
```
MicroPython v1.8.5-62-g964fb24-dirty on 2016-10-31; linux version
```
The following code throws me back to my terminal without any messages, a quick look suggest that `mp_stream_posix_read` doesn't return.
```python
import usocket
import ussl

s = usocket.socket(usocket.AF_INET, usocket.SOCK_STREAM)
ss = ussl.wrap_socket(s)
```
Wrapping with server_size=True however, works as expected.

pfalcon, 2016-11-01:
What does standard http_client_ssl.py example give you?

I can only think that you might be experiencing https://github.com/micropython/micropython/issues/2216 problem.

hosaka, 2016-11-02:
The example/http_client_ssl.py seems to work just fine. I get the socket reading back using the stream.

If I runfrom a command line calling `micropython ussl.py` it works, but was testing this from REPL before. It certainly does look like the SIGPIPE case you've linked. I can at least continue writing the test for now.

hosaka, 2016-11-02:
However if I don't connect the socket before wrapping it, I get the same problem as initially described. This is possible with cpython however.

pfalcon, 2017-05-14:
This should be fixed now. Feel free to reopen with additional info.

closed 2017-05-14 by pfalcon