micropython/micropython #2589: micropython -m upip install not working?

bunschot, 2016-11-02:
Hi,

micropython -m upip install appears not to work on my vagrant (https://github.com/pfalcon/esp-open-sdk) based system. I've followed the build instructions provided in the README.md (https://github.com/micropython/micropython/blob/master/README.md) to the letter. 

Running upip:

vagrant@vagrant-ubuntu-trusty-64:~/micropython/unix$ ./micropython -m upip install micropython-binascii
Installing to: /home/vagrant/.micropython/lib/

The directory /home/vagrant/.micropython/lib/ remains empty...

Note: SSL is installed:
MicroPython v1.8.5-100-g10bde69-dirty on 2016-11-02; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import ussl
>>> dir(ussl)
['__name__', 'wrap_socket']

Any suggestions?

pfalcon, 2016-11-02:
I can't reproduce this. For reference, here's how working session should look:
~~~
$ micropython -m upip install micropython-binascii
Installing to: /home/pfalcon/.micropython/lib/
Warning: pypi.python.org SSL certificate is not validated
Installing micropython-binascii 2.4.0-4 from https://pypi.python.org/packages/99/5f/73356a4b492ce53a3fa394389b8b6bdfaa11735f813a6c7bd77bffa97c80/micropython-binascii-2.4.0-4.tar.gz
~~~

Please report the exact output of upip, including command prompt before and after it, using github formatting to make it clear. Immediately after upip, please run `echo $?` and report its result too.

hosaka, 2016-11-02:
Not running vagrand, but works fine for me in arch linux.

bunschot, 2016-11-02:
I get the following output on the command prompt:
```
vagrant@vagrant-ubuntu-trusty-64:~/micropython/unix$ ./micropython -m upip install micropython-binascii
Installing to: /home/vagrant/.micropython/lib/
Warning: pypi.python.org SSL certificate is not validated
vagrant@vagrant-ubuntu-trusty-64:~/micropython/unix$ echo $?
141
vagrant@vagrant-ubuntu-trusty-64:~/micropython/unix$
```

And, when running pip from within micropython:
```
vagrant@vagrant-ubuntu-trusty-64:~/micropython/unix$ ./micropython
MicroPython v1.8.5-100-g10bde69-dirty on 2016-11-02; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import upip
>>> upip.install('micropython-binascii')
Installing to: /home/vagrant/.micropython/lib/
Warning: pypi.python.org SSL certificate is not validated
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "upip.py", line 205, in install
  File "upip.py", line 195, in install
  File "upip.py", line 152, in install_pkg
  File "upip.py", line 144, in get_pkg_metadata
ValueError: syntax error in JSON
>>>
```

pfalcon, 2016-11-02:
~~~
$ echo $?
141
~~~

So, unclear why, but you're hit by https://github.com/micropython/micropython/issues/2216 .

upip is here: https://github.com/micropython/micropython-lib/tree/master/upip, you would need to add debug output to see what goes wrong. (Be sure to rename upip.py to something like upip_.py to make sure it will execute, not builtin upip).

I'd of course start with checking networking configuration of your VM.

bunschot, 2016-11-02:
Not quite sure where or how to start with checking my networking configuration... 

Below the steps I've taken to run upip with debugging option enabled:

```
vagrant@vagrant-ubuntu-trusty-64:~/micropython-lib/upip$ git pull
Already up-to-date.
vagrant@vagrant-ubuntu-trusty-64:~/micropython-lib/upip$ cp upip.py myupip.py
vagrant@vagrant-ubuntu-trusty-64:~/micropython-lib/upip$ vi myupip.py
vagrant@vagrant-ubuntu-trusty-64:~/micropython-lib/upip$ diff upip.py myupip.py 11c11
< debug = False
---
> debug = True
vagrant@vagrant-ubuntu-trusty-64:~/micropython-lib/upip$ micropython
MicroPython v1.8.5-95-gb97c17e-dirty on 2016-11-02; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import myupip
>>> myupip.install('micropython-binascii')
Installing to: /home/vagrant/.micropython/lib/
Queue: ['micropython-binascii']
Warning: pypi.python.org SSL certificate is not validated
vagrant@vagrant-ubuntu-trusty-64:~/micropython-lib/upip$ echo $?
141
```
Note that the micropython session died, I did not press Crtl-D to end the session.

Other than the `Queue: ['micropython-binascii']` message no new info...

bunschot, 2016-11-04:
Did some simple networking tests:
- `ping google` works 
- `ping pypi.python.org` works
- `pip install` works

pfalcon, 2016-11-04:
> Note that the micropython session died, I did not press Crtl-D to end the session.

Yes, as explained in #2216.

> Did some simple networking tests:

Ok, now you'll need to add debug prints to upip_.py and see where it dies and why. I.e. debug it like any other python app, except that upip is very simple app, comparing e.g. to django.

bunschot, 2016-11-04:
Some progress. I've added some debugging statements.  The `syntax error in JSON` is fair; not all data seems to be coming trough while fetching the data with `s = f.read()` in `get_pkg_metadata()`.  

```vagrant@vagrant-ubuntu-trusty-64:~/micropython-lib/upip$ micropython -m myupip install micropython-logging
install(['micropython-logging'],None)
get_install_path()
expandhome(/home/vagrant/.micropython/lib)
Installing to: /home/vagrant/.micropython/lib/
Queue: ['micropython-logging']
install_pkg(micropython-logging,/home/vagrant/.micropython/lib/)
get_pkg_metadata(micropython-logging)
url_open(https://pypi.python.org/pypi/micropython-logging/json)
  Address infos: [(2, 1, 6, None, bytearray(b'\x02\x00\x01\xbb\x97e$\xdf\x00\x00\x00\x00\x00\x00\x00\x00')), (2, 2, 17, None, bytearray(b'\x02\x00\x01\xbb\x97e$\xdf\x00\x00\x00\x00\x00\x00\x00\x00')), (2, 3, 0, None, bytearray(b'\x02\x00\x01\xbb\x97e$\xdf\x00\x00\x00\x00\x00\x00\x00\x00')), (10, 1, 6, None, bytearray(b'\n\x00\x01\xbb\x00\x00\x00\x00*\x04NB\x00\t\x00\x00\x00\x00\x00\x00\x00\x00\x02#\x00\x00\x00\x00')), (10, 2, 17, None, bytearray(b'\n\x00\x01\xbb\x00\x00\x00\x00*\x04NB\x00\t\x00\x00\x00\x00\x00\x00\x00\x00\x02#\x00\x00\x00\x00')), (10, 3, 0, None, bytearray(b'\n\x00\x01\xbb\x00\x00\x00\x00*\x04NB\x00\t\x00\x00\x00\x00\x00\x00\x00\x00\x02#\x00\x00\x00\x00'))]
Connect address: bytearray(b'\x02\x00\x01\xbb\x97e$\xdf\x00\x00\x00\x00\x00\x00\x00\x00')
Warning: pypi.python.org SSL certificate is not validated
GET https://pypi.python.org/pypi/micropython-logging/json HTTP/1.0
Host: pypi.python.org

{
    "info": {
        "maintainer": null,
        "docs_url": null,
        "requires_python": null,
        "maintainer_email": null,
        "cheesecake_code_kwalitee_id": null,
        "keywords": null,
        "package_url": "http://pypi.python.org/pypi/micropython-logging",
        "author": "MicroPython Developers",
        "author_email": "micro-python@googlegroups.com",
        "download_url": "UNKNOWN",
        "platform": "UNKNOWN",
        "version": "0.1.3",
        "cheesecake_documentation_id": null,
        "_pypi_hidden": false,
        "description": "This is a module reimplemented specifically for MicroPython standard library,\nwith efficient and lean design in mind. Note that this module is likely work\nin progress and likely supports just a subset of CPython's corresponding\nmodule. Please help with the development if you are interested in this\nmodule.",
        "release_url": "http://pypi.python.org/pypi/micropython-logging/0.1.3",
        "downloads": {
            "last_month": 0,
            "last_week": 0,
            "last_day": 0
        },
        "_pypi_ordering": 8,
        "classifiers": [],
        "bugtrack_url": null,
        "name": "micropython-logging",
        "license": "MIT",
        "summary": "logging module for MicroPython",
        "home_page": "https://github.com/micropython/micropython-lib",
        "cheesecake_installability_id": null
    },
    "releases": {
        "0.0.4": [
            {
                "has_sig": false,
                "upload_time": "2014-11-14T00:51:35",
                "comment_text": "",
                "python_version": "source",
                "url": "https://pypi.python.org/packages/1b/ea/875abe6e94ba6c6af7c8a1191f7da17832904f56c8b04d13478e55c55449/micropython-logging-0.0.4.tar.gz",
                "md5_digest": "06ce2f2cc504ee6ef493601958c00156",
                "downloads": 1718,
                "filename": "micropython-logging-0.0.4.tar.gz",
                "packagetype": "sdist",
                "path": "1b/ea/875abe6e94ba6c6af7c8a1191f7da17832904f56c8b04d13478e55c55449/micropython-logging-0.0.4.tar.gz",
                "size": 1619
            }
        ],
        "0.0.1": [
            {
                "has_sig": false,
                "upload_time": "2014-04-16T22:37:27",
                "comment_text": "",
[... 77 lines cut ...]
                "md5_digest": "b7d84b9a84845da8350ce47e87c89bcb",
                "downloads": 935,
                "filename": "micropython-logging-0.1.1.tar.gz",
                "packagetype": "sdist",
                "path": "03/3c/e5d59034ad2130c9b93825c1e5018e1a757fd2595e7a2ccecb598f432cc4/micropython-logging-0.1.1.tar.gz",
                "size": 1726
            }
        ],
        "0.1.2": [
            {
                "has_sig": false,
                "upload_time": "2016-01-31T00:00:41",
                "comment_text": "",
                "python_version": "source",
                "url": "https://pypi.python.org/packages/ef/48/b74ebec1dfde3257288d77ab1b06583e2b0b92928bf7362
Traceback (most recent call last):
  File "myupip.py", line 326, in <module>
  File "myupip.py", line 319, in main
  File "myupip.py", line 239, in install
  File "myupip.py", line 229, in install
  File "myupip.py", line 184, in install_pkg
  File "myupip.py", line 171, in get_pkg_metadata
ValueError: syntax error in JSON
vagrant@vagrant-ubuntu-trusty-64:~/micropython-lib/upip$
```

pfalcon, 2017-05-13:
Please retest with the latest MicroPython and latest upip. All issues should have been fixed. If you still experience problems, please open another bug.

closed 2017-05-13 by pfalcon