micropython/micropython #2592: Prepare for v1.8.6 release

dpgeorge, 2016-11-02:
It's the start of the month so time for the next release.  There doesn't need to be anything special with/for this release, the main thing is to have it out so that work can proceed with updating cc3200 port (eg #2559 would be merged after the release is tagged).

pfalcon, 2016-11-03:
Actually, there should be special things - new non-networking setup script for WebREPL yet needs to be written https://github.com/micropython/micropython/issues/2538

robert-hh, 2016-11-03:
For CC3200 there is also a fix needed, since it is not recognised any more by pymakr. Daniel told that he would supply that fix.

dpgeorge, 2016-11-03:
Ok, first comment updated to include things that need doing.  Let's say 1 week until the release (10th Nov), or earlier if things are all done by then.

pfalcon, 2016-11-03:
Sounds good, but another thing to do is thorough testing of esp8266 with sdk2.0 and increased heap size. I did first part of it (http://forum.micropython.org/viewtopic.php?f=16&t=2593), need to do more. Would need to collect some feedback from users.

But Nov, 10 sounds good. And next version may be 1.9, and we may do it a bigger release cycle (but not necessarily).

pfalcon, 2016-11-05:
Ok, I performed 2nd of integration tests - HTTP serving against ab/siege. With siege, got beyond 10000 requests with concurrency 2. It died from TIME_WAIT invasion, and can sustain 41 such socket. All figures aren't bad at all.

puuu, 2016-11-07:
Since the semantic of `utime.ticks_diff()` was changed (#2217), which is IMHO a big API change, I would suggest to call this version 1.9. For me, this change breaks a bunch of scripts. IMHO, it will be more easy to remember that this change was introduced in 1.9 than in 1.8.6 and it can be easier pointed out on user questions at the forum.

BTW, what software versioning process does MicroPython use?

Another suggestion is to consider, if it makes sense to move more files from `esp8266/scripts/` to `esp8266/modules/` in the `esp-extra-scripts` branch. For example, I would like to see `umqtt/simple.py` as frozen in the next release.

pfalcon, 2016-11-07:
> Since the semantic of utime.ticks_diff() was changed (#2217), which is IMHO a big API change, I would suggest to call this version 1.9. 

Your suggestion is too late, the 1.8.6 release plan has been set.

> For me, this change breaks a bunch of scripts.

Then you want to read and comment on http://forum.micropython.org/viewtopic.php?f=2&t=2589 .

> For example, I would like to see umqtt/simple.py as frozen in the next release.

Please test it in such config for a couple of days and if it's ok, it may yet fit into 1.8.6.

puuu, 2016-11-08:
> Please test it in such config for a couple of days and if it's ok, it may yet fit into 1.8.6.

Ok, I moved the following files from `esp8266/scripts/` to `esp8266/modules/` and compiled it with the recent git version. I use them in two projects. Just flashed it to the ESP8266 devices. They are running without problems. I will tell you tomorrow, if I observe some problems or not.
- `ssd1306.py`
- `websocket_helper.py` (via webrepl)
- `ntptime.py`
- `neopixel.py`
- `umqtt/robust.py`
- `umqtt/simple.py` (e.g. saves about 2900 byte and reduces fragmentation when used as frozen bytecode)

Interesting to move, but I can not test:`apa102.py`, `dht.py`, `urequests.py`, `urllib/urequest.py`

puuu, 2016-11-08:
@pfalcon wouldn't it be good to have a hint to `webrepl_setup` in the `help()` function of the esp8266 port?

pfalcon, 2016-11-08:
> Interesting to move, but I can not test: apa102.py, dht.py, urequests.py, urllib/urequest.py

Let's not haste anywhere and do it step by step. Frozen bytecode is beta (if not alpha), and there're already enough confusing issues being reported.

pfalcon, 2016-11-08:
> @pfalcon wouldn't it be good to have a hint to webrepl_setup in the help() function of the esp8266 port?

No idea. I'm sure that it would be good to have entire docs in help(), but we can't.

robert-hh, 2016-11-08:
I can test dht.py, and I have a faint memory that I did that already. But better repeat that with a recent build.

pfalcon, 2016-11-08:
@dpgeorge : I believe I have everything in from my side.

peterhinch, 2016-11-09:
Just FYI in my recent efforts with umqtt mqtt.py and robust.py were frozen as bytecode, along with other modules. I didn't encounter any issues traceable to the fact of their being frozen.

pfalcon, 2016-11-09:
I moved uqmtt, urllib.urequest and urequests to modules/ . That's all that we move for this release, done. Please retest esp-extra-scripts branch.

puuu, 2016-11-09:
>> @pfalcon wouldn't it be good to have a hint to webrepl_setup in the help() function of the esp8266 port?

> No idea. I'm sure that it would be good to have entire docs in help(), but we can't.

Is just a feeling. `help()` provides a nice copy-and-paste-like instruction for the first setup. Since now,  webrepl needs `import webrepl_setup` at the first setup, IMHO it will fit.

puuu, 2016-11-09:
Tested `ssd1306.py`, `websocket_helper.py`, `ntptime.py`, `neopixel.py`, `umqtt/robust.py`, `umqtt/simple.py` for about 24 hours as frozen byte code and it woks fine.

> I moved uqmtt, urllib.urequest and urequests to modules/ . That's all that we move for this release, done. 

This sounds good. Thanks. Looking forward to a 1.8.6 release with more free RAM.

Is there a reason, why `websocket_helper.py` is not frozen byte code? It leads to heap fragmentation when webrepl is enabled.

> Please retest esp-extra-scripts branch.

Yes, tested. Runs fine for me.

kfricke, 2016-11-09:
I have also not had any issues with `umqtt/robust.py` and `umqtt/simple.py` over the last weeks.
It is frozen bytecode in the `modules/` directory of my self-built and always updated 1.8.5 firmwares.

Websocket and WebREPL stuff is completely untested by me. And i have not yet had time to use and test the esp-extra-scripts branch.

robert-hh, 2016-11-09:
Tested a dht22 compatible (AM2302) with the dht.py module in frozen bytecode. No complaints.
Also port_diag.py runs well from frozen bytecode.

dpgeorge, 2016-11-10:
I've had no word from @danicampora about fixing the cc3200 port so it works with the new PyMakr.  So I'll go ahead later today with the release and we can make a cc3200 branch (from v1.8.6) with the fix if needed.

robert-hh, 2016-11-10:
I also approached abilio via the pycom board, and he mentioned your call to danicampora. I offered toport the change myself. That would be possible after the sources get public.

pfalcon, 2016-11-10:
> So I'll go ahead later today with the release and we can make a cc3200 branch (from v1.8.6) with the fix if needed.

Thanks. Indeed, 1.8.6.1 can be made from a branch at any time.

pfalcon, 2016-11-10:
> I offered toport the change myself. That would be possible after the sources get public.

Please be careful about this stuff - the source you're waiting for are GPL sources. And if the whole resolution of that problem depends on GPL sources, then what you would definitely look like importing GPL code into MicroPython codebase, which makes it in whole GPL. So you'd better wait for the vendor to support their products integration. An alternative is to do your own provably independent engineering to resolve this problem. And any resolution of something in MicroPython starts with detailed description of what problem MicroPython has - i.e. with bug report.

My suggestion: we'll need a lot of help with cc3200 port after 1.8.6 release. Spare your time for that, rather than working on pretty niche features like pymakr support (it can be done later if someone else don't do it).

robert-hh, 2016-11-10:
OK. Obviously this change for cc3200 is not time critical. pymakr is not close to be finished anyhow.

dpgeorge, 2016-11-10:
v1.8.6 was tagged in 5a1d63fc14dae788f705403a43c2d8639b7dd9cd

dpgeorge, 2016-11-10:
esp-extra-scripts branch was rebased on v1.8.6 and force-pushed.

closed 2016-11-10 by dpgeorge