micropython/micropython #2593: Create guide for writing drivers

ralienpp, 2016-11-03:
I want to add support for the `dp83848` Ethernet module from Texas Instruments, but after searching through the forum and the docs - I found no description of what is necessary to accomplish that.

I believe that the documentation should contain a section for driver writers, that would explain the workflow, the requirements and best practices, such that potential contributors could see the bigger picture.

Here are some of the questions that the guide could address:

- is such code generic for MicroPython or is the driver implementation different for each board?
- what language could and should it be written in?
- what information from the device manufacturer is usually necessary for a driver developer to get started?
- where should the driver files be kept within MicroPython's directory layout?
- what licensing constraints must be taken into account when using code provided by the device vendor?

deshipu, 2016-11-03:
So there is an attempt to document at least a part of that at https://github.com/micropython/micropython/wiki/Driver-API-Design-Guidelines

deshipu, 2016-11-03:
> Here are some of the questions that the guide could address:
> 
> - is such code generic for MicroPython or is the driver implementation different for each board?

There is a pull request (https://github.com/micropython/micropython/pull/2571) documenting good practices that make it easier to write generic code for all ports.

> - what language could and should it be written in?

MicroPython.

> - what information from the device manufacturer is usually necessary for a driver developer to get started?

The datasheet for that device.

> - where should the driver files be kept within MicroPython's directory layout?

Outside, in separate repositories.

> - what licensing constraints must be taken into account when using code provided by the device vendor?

It's entirely up to what license the vendor released them under, and completely unrelated to the MicroPython project.

ralienpp, 2016-11-03:
Thank you for the fast reaction, I have examined the docs and things are a bit clearer now, though some questions remain:

Concerning languages, I checked this driver and it is in C: https://github.com/micropython/micropython/tree/master/drivers/wiznet5k/ethernet. I was expecting something similar to be required for the dp83848, given that the vendor (Texas Instruments) provides a driver: https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/tree/drivers/net/phy/dp83848.c. The Wiznet samples indicate that you have to add a dependency when compiling MicroPython itself (before flashing it onto the board), i.e. `make -B BOARD=STM32F4DISC MICROPY_PY_WIZNET5K=1`

- Why are some drivers written in C and others in MicroPython?

There is a reference to db83848 in https://github.com/micropython/micropython/blob/master/stmhal/boards/HYDRABUS/stm32f4xx_hal_conf.h#L183, but it is not in the driver directory, it appears to be something specific to a particular board.

- If this code is not a driver, what is it?

When you say "Outside, in separate repositories.", I suppose you mean something other than https://github.com/micropython/micropython/tree/master/drivers.

- Why are some drivers a part of the project, while others ought to be separate?

deshipu, 2016-11-03:
> Why are some drivers written in C and others in MicroPython?
Some drivers, such as the one-wire, dht11 or neopixel ones, require a part of them to be written in C, because they need precise timing. The best way to do that is to only write in C a function that does the demanding part, and write the rest in MicroPython.

I'm not familiar with the driver you linked to, but my guess is that it plugs into some low-level machinery of MicroPython. I don't think this would be necessary for a standard SPI device such as dp83848.

> If this code is not a driver, what is it?

No idea, maybe someone else can answer.

> Why are some drivers a part of the project, while others ought to be separate?

Historical heritage. They should be all separated into individual repositories eventually.

dhylands, 2016-11-03:
The driver you linked to is a linux kernel driver which will rely on linux kernel services which are not available in MicroPython.

It's probably possible to write any of the drivers in MicroPython instead of in C, but there are tradeoffs. The C version will generally be smaller and faster. The C version also has access to things that MicroPython hasn't exposed to the python side of things (like DMA).

pfalcon, 2016-11-06:
@ralienpp: You're welcome to write such a guide.

In the meantime, please answer yourself a simple question - when you submitted this ticket, you were presented a link to Contribution Guidelines. Have you read it? Because if you have had, there wouldn't be  questions like:

> * what language could and should it be written in?
> * where should the driver files be kept within MicroPython's directory layout?

because they're well covered in that doc.

That leads to a thought - writing more documentation isn't going to help, because every documentation has a big drawback - it needs to *be read*. And the problem is not writing more documentation, but making people read what's already available. And available is a lot, a whole day probably won't be enough to read all (forum is also a documentation). At the same time, we have a whole subforum for drivers with new ones being posted regularly: http://forum.micropython.org/viewforum.php?f=14 . So, somehow many people found that they can proceed with the current documentation and examples. (Some actually write more docs based on their experience!)

Summing up: everyone missing documentation is welcome to write it. If there's an expectation that someone else will write it for you, well, that can easily be a long wait, or a straight deadlock, because "someone else" expects you to read what's available first.

deshipu, 2016-11-06:
> And the problem is not writing more documentation, but making people read what's already available.

I think a good start on solving *that* problem would be reducing the number of places one has to look for things, and making it a little better organized. For instance, I have been directed to read the contributor guidelines many times (with a silent condescending assumption that I haven't), and sometimes I found new things in there that was added just after I actually asked the question... Because of such practices it has became a jumbled mess of everything, and I'm not surprised that people fail to find what they are looking for there even after 4 or 5 re-readings. It reads like a stream of consciousness, and it's very easy to see why if you look at the list of authors at https://github.com/micropython/micropython/wiki/ContributorGuidelines/_history -- basically, this *is* a write-only dump of someone's good wishes about how this project should work.

ralienpp, 2016-11-06:
Thank you for your thoughtful message. My intention is to write that guide, but I must first create a clear mental model of how things work.

Attempting to avoid creating unnecessary discussions, I reviewed the contribution guidelines and explored some forum topics that appeared to be relevant. As I examined the source code I stumbled upon the Wiznet driver. It is written in C, contrary to the conclusions I've made after reading the guide.

Since I'm still at the very beginning of the learning curve, it is difficult to determine what exactly is going on. Does empirical evidence (a working example in C) outweigh what the theory says (write in Python)? Is it just another case of "the code was updated but the doc wasn't"? These things are obvious to veteran contributors because they've got everything cached in their heads; whereas to a beginner these appear as pieces that probably belong to another puzzle.

While I agree that documentation is often not read, sometimes one of the reasons is because it is scattered across multiple sources (Github issues, Github wiki, forum, source code, readthedocs). For instance, I asked about it on the forum you linked to, but there was no reaction; in one of the other threads someone suggested that Github is more appropriate (sure enough, here the reactions were much swifter).

pfalcon, 2016-11-06:
> the number of places one has to look for things, and making it a little better organized

When I read that, I tell *myself*: RFCs and patches welcome. That's why I consider it's fair to tell the same the same to you too. Let's see if my self-mantra works:

~~~
$ git log docs | grep "Author: Paul Sokolovsky" | wc -l
168
$ git log docs | grep "Author: Damien George" | wc -l
123
$ git log docs | grep "Author: Daniel Campora" | wc -l
31
$ git log docs | grep "Author: Radomir Dopieralski" | wc -l
10
~~~

> For instance, I have been directed to read the contributor guidelines many times (with a silent condescending assumption that I haven't)

Nothing "condescending", at most "surprised". And you cleared that part too, explaining that you have bad memory. Well, that's why that doc exists, at a specific designated place - so everyone can return to it regularly.

> and sometimes I found new things in there that was added just after I actually asked the question...

That may sound really surprising (??), but when I suggested someone to read it, I take a chance to do it too, at least to check if point in question is really covered. And I usually to see something to fix, usually typos/grammar, and that's what I do. This happened this time too, I added couple of articles. I also had a fresh look at your https://github.com/micropython/micropython/wiki/Driver-API-Design-Guidelines and added something there.

> It reads like a stream of consciousness, and it's very easy to see why if you look at the list of authors at https://github.com/micropython/micropython/wiki/ContributorGuidelines/_history -- basically, this is a write-only dump of someone's good wishes about how this project should work.

Thanks for your usual kind words. You'd be even more frightened to look at https://github.com/micropython/micropython/graphs/contributors - good 40% of changes to the project over 3 years are done by me too.

pfalcon, 2016-11-06:
@ralienpp:

> Does empirical evidence (a working example in C) outweigh what the theory says (write in Python)?

Where do you see "write in Python"? The Contributor Guidelines say clearly:

> This should be good rule of thumb - anything which can be written in Python, should be written in Python, unless there're really good reasons to write it in C.

You can go a long way in understanding from there. For example, someone who wrote wiznet driver *had good reasons* to write in in C. And folks who post drivers on forum didn't have reasons for C, so wrote in Python.

Now you probably wonder if *you* have reasons for writing in C, and yeah, that's good question which you should answer to yourself. How do you answer it? Well... that's hard question, Computer Science is tought in universities for 5 years, after which a student is expected to either be able to decide, or at least understand what involves in such decision making. I wouldn't be able to explain that "in few words" to a stranger on the street. But I can try: you should write Ethernet drivers C, and wiznet driver gives a good hint of that. But if you write it in Python, you will be the first one to do that (in MicroPython world), so it would be really cool.

> These things are obvious to veteran contributors because they've got everything cached in their heads; whereas to a beginner these appear as pieces that probably belong to another puzzle.

Sorry, as pointed above, this has little to do with MicroPython, and is just the general software development decision making. You learn it by practice. Chose one way, go it, fail, backtrack, and chose another way. Repeat until you "feel" which ways a dead-ends.

deshipu, 2016-11-06:
> Thanks for your usual kind words. You'd be even more frightened to look at https://github.com/micropython/micropython/graphs/contributors - good 40% of changes to the project over 3 years are done by me too.

Well, that's only expected from one of two core contributors (though it does seem like there is an unhealthy lack of new contributors for a project this size). I would be very disappointed with that last Kickstarter campaign if it wasn't the case, in fact -- it's good to see my money having an effect. In the mean time, I have a great day job developing open source software with very nice and friendly people, which unfortunately doesn't involve MicroPython, and a hobby that I enjoy very much, which does sometimes involve a little of MicroPython, but not all of it (for instance, the unix and windows ports are completely useless to me). So I contribute here and there, whenever I see an opportunity, but since the resistance is rather high, I recently switched to working on external documentation, such as workshops, tutorials, examples of using device drivers and an unofficial developer guide. Of course I will always be happy to help with the official documentation when there is a consensus about what needs to be done and when I can be reasonably sure my contributions won't get rejected or trampled over in the very next commit, which will change what just has been documented, without updating the docs.

pfalcon, 2016-11-06:
@ralienpp: Sorry for complete offtopic in your ticket.

> Well, that's only expected from one of two core contributors

Correction: from a *maintainer*.

> (though it does seem like there is an unhealthy lack of new contributors for a project this size)

There's no lack of *contributors*. Since KS campaign, we have 45 new people having contributed to uPy, that's 50% increase from pre-KS level. And we have 77 pending pull requests beyond that.

There's however lack of maintainers, and queue of 77 pending PRs shows that. There're reasons for that - the audience to which MicroPython catered initially is makers/hobbyist, and that's still its driving force. That's great, we do the revolution "from the bottom up", but there's little money there for big growth, and thus no big company which could afford many maintainers. And from community, there're not many who'd be able to do sustained maintenance, and before that, there should be common goals with the project (== current maintainers). And uPy is a special project here. How most user contributions to most projects go, it's for making "more", "bigger", etc. While uPy tries to stay small, and that approach seems to hard for people to understand.

> I would be very disappointed with that last Kickstarter campaign if it wasn't the case, in fact -- it's good to see my money having an effect.

Right, and that's exactly how it should be. Just keep in mind that 80% or so of what MicroPython is done by 2 maintainers, so if you like MicroPython, it makes no sense to go out of way to criticize them (or one of them).

> In the mean time, I have a great day job developing open source software with very nice and friendly people, which unfortunately doesn't involve MicroPython, and a hobby that I enjoy very much, which does sometimes involve a little of MicroPython

Same here, and that's what we're working on with Damien, trying to grow project beyond "hobby" level (there's some progress). That's one of the problem with maintainership - we try to maintain project professionally (in software development manner) whose target audience are makers and hobbyists. A lot of misunderstanding come from this mismatch. But again, your thinking should be: "if these guys led project to such point so far, they know where to lead it further". That's the same approach which everyone of us has to use about CPython, Linux, your Linux distro (close people, government, ...). There's enough disappointment along the way, and well, there's always possibility to switch to something else, but will it be for better?

> (for instance, the unix and windows ports are completely useless to me)

Hope you have enough insight to don't think that if they're useless for you, they're useless at all?

> I recently switched to working on external documentation, such as workshops, tutorials, examples of using device drivers and an unofficial developer guide.

Hey, cool to see that, and that it's a conscious decision. If only you'd also understand that it doesn't make sense to argue (like, all the time) with a maintainer. Instead, you'd better understand why one does this or that (like, think what you'd do in various cases, not bombard with questions).

> when I can be reasonably sure my contributions won't get rejected or trampled over in the very next commit, 

Well, you will be surprised again, but docs like https://github.com/micropython/micropython/wiki/ContributorGuidelines are written exactly to communicate how to write patches which wouldn't be rejected. Ditto for tickets like https://github.com/micropython/micropython/issues/1331 . And those things were put together by someone who a) cares that people's patches are not rejected; b) has time (or carves it out, because it's important) to write it. After passing review, it becomes the official project documentation. So again, instead of arguing with it, you'd better try to understand it.

> which will change what just has been documented, without updating the docs.

You treat many things with very narrow view. Above is given statistics of who and how frequently updates docs. So, you can either grow (being prepared to learn how to do that well!) your part if you're concerned, or just wait until others get to it. The start is always easy - we have gazillion of typos there which need to be fixed.

Anyway, this is offtopic here. Soon, there will be a chance to discuss project organization issues, if there's something left to discuss.

deshipu, 2016-11-06:
> Correction: from a maintainer.
Every contributor of an open source project should be doing maintenance work as much as any other work, eventually. I think it's a mistake to create roles solely responsible for this, and to remove the responsibility from the whole community.

> There's no lack of contributors. Since KS campaign, we have 45 new people having contributed to uPy, that's 50% increase from pre-KS level. And we have 77 pending pull requests beyond that.

How many of them are here to stay, though? How many of them will gradually become more involved with the project, and take upon themselves more responsibilities? How many will get scared away or discouraged? How do you plan to get more cores on board?

> Just keep in mind that 80% or so of what MicroPython is done by 2 maintainers, so if you like MicroPython, it makes no sense to go out of way to criticize them (or one of them).

You misunderstand me. I never criticize people, and in fact even if I did, I would never criticize you or Damien, for I have a very high regard for both of you. Even though I have never met you in person (I wish I will, sooner or later), I think you are a great guy, really. I do try to criticize particular behaviors and particular states of affairs, where I think they are detrimental to the project or to civil communication in general. So please don't take my criticism personally.

> But again, your thinking should be: "if these guys led project to such point so far, they know where to lead it further".

Thank you very much indeed for telling me what I should be thinking. Even the best natural-born leader is helpless without information and feedback, and I'm giving that feedback in hope that it will indeed help lead the project. In fact, I think that the project's leadership has improved in the recent months somewhat, and I like to think (a bit self-centered), that at least a tiny bit of that is because of my voice, among the hundreds other voices out there. I do trust that you will make the right long-term decisions eventually, but I will keep trying to help getting them right faster.

>> (for instance, the unix and windows ports are completely useless to me)

> Hope you have enough insight to don't think that if they're useless for you, they're useless at all?

That goes without saying. However, it also means that I have very little expertise for working on them (since I don't use them and don't even know the use cases), and even lesser motivation. I like to think of it as a natural balancing inside the project -- the things that are actually being used a lot getting more attention.

> Anyway, this is offtopic here. Soon, there will be a chance to discuss project organization issues, if there's something left to discuss.

Thanks, can't wait for that.

mcauser, 2016-11-08:
> ... we have gazillion of typos there which need to be fixed.

I fixed a lot of them #2281, but _got in trouble_ for submitting a large pull request. ¯\\\_(ツ)\_/¯

dpgeorge, 2016-11-08:
@ralienpp the majority of drivers that already exist are written in (Micro)Python and are relatively simple, controlling sensors and displays.  These are usually maintained by the respective authors in separate repos.  Looking at existing drivers and reading https://github.com/micropython/micropython/wiki/Driver-API-Design-Guidelines is a good way to get started in this area.

But writing an ethernet driver is very different to writing simple sensor drivers.  I assume you'll want to expose it as a proper socket object, and that requires a lot of work.  It could be done in Python, eg if the ethernet chip is SPI-based, and you may want to start with a proof of concept that way, and convert to C once it's working.

jonnor, 2024-07-19:
Is there anything in particular missing from the above linked documentation? If not, I propose that we close this issue.

closed 2025-03-03 by jonnor