micropython/micropython #2594: Add support for function __name__

spac3-monkey, 2016-11-03:
```python
def my_function():
    pass

my_function.__name__
```

hosaka, 2016-11-04:
What's your micropython build and port that you're running on?

e.g unix:
```python
MicroPython v1.8.5-94-g498d8b7-dirty on 2016-11-04; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== def my_function():
===     pass
=== 
=== print(my_function.__name__)
my_function
>>> 
```

spac3-monkey, 2016-11-04:
I'm running `ESP8266` port, version: `v1.8.5-108-g7bb0f7b on 2016-11-03; ESP module with ESP8266`.

prusnak, 2016-11-08:
esp8266 port is built without `MICROPY_PY_FUNCTION_ATTRS` macro defined in `mpconfigport.h` as opposed to unix or stmhal ports where this is defined to 1.

My guess is that this is because of the firmware size, but it might be that it was just forgotten.

Try rebuilding the binary with this macro defined if you need this functionality.

spac3-monkey, 2016-11-08:
thanks! @prusnak, will give it a try.

closed 2022-06-24 by dpgeorge