micropython/micropython #2595: Convert machine module and others into a soc and board package

torwag, 2016-11-04:
Hi,
Now we have the modules
"machine" which as written in the docs contains - functions related to the board
and we have a special modules e.g.
"esp" - functions related to the ESP8266
"wipy" - WiPy specific features
"pyb" - contains functions related to the board

This naming is already kind of twisting my brain. Why do we have two libs which are basically address both the board itself (wipy, pyb)? If we project this into the future with much more boards and much more chips, thing will get more and more complicated.

Thus, I thought it would be a good idea to make a more clear separation. Having a package-duo called
soc (system on chip) which contain basically all the supported soc family names and a board package, which is basically contain the board names.
The structure would look like this:
soc.foo contains all and everything which based on the actual chip (foo). foo could be STM32F4, ESP8266, nRF51, ESP32, etc. This includes Pin, ADC, DAC, Timer, SPI, I2C. For certain models within a family we could call an init function
```
import soc.stm32f4 as stm
stm.init("STM32F405RG")
```
which set limits and issues errors for possible out of range indices to address the different amounts and lacks of  I2C , SPI, etc.  interfaces, timers, etc. 
Users identify the chip of a certain board and if there is a soc-module, chances are high that this board can be used directly with micropython,

The second package would contain board modules, which contains all functions to address peripherals on the board itself. LEDs, switches, other chips, etc. This would help to let people now whether a certain board has already full support. It would also be easier for people to write a board-module for a new board and hence enlarge the board family by time. It basically wraps many of the soc module stuff and make it easier and direct accessible for the user. E.g. an additional chip connected via a SPI port could be directly addressed since we know this is hard wired on that particular board. Thus users, do not have to deal with the "low level" SPI stuff but can directly use functions to utilise the chip itself.

The distinction between boards and chips sounds rather logical to me. Micropython can easily support hundred of boards based on the same chip, without much effort, taken that the board vendors provide board modules. Vice versa different iterations of a board could use different chips (e.g. of the same family) without the need to rewrite much code. 

An example:  The soc package contains a esp8266 module and in the board package you find modules for the huzzah, feather, d1mini, etc. The esp8266 module would remain the same, whereas the board modules cover the features of the individual boards. This clear separation allows easily extendability to other boards (by vendors and users and not MP core devs) .

Coding would look like:
```
import soc.esp8266 as esp
import board.d1mini as d1
#just as an example, in case there are several deviates of the esp8266
esp.init("esp8266ex")
#direct access of board functions no pin init needed
d1.init()  #board init for the usual soc
# d1.init(esp) #init with an explicit soc
d1.led(1)
stat = d1.button.status()
#access of the soc functions to turn off the same led
led = esp.Pin(0, Pin.OUT)
led.value(0)
```
I believe that this kind of code is much more self explaining and more easy to understand by beginners. Esp. porting code from one board to another might become much more easy. It clearly separates soc  and board functions. I know this sort of discussion is rather late, but I would like to share the idea.

deshipu, 2016-11-04:
So basically what you propose is to call all different board-specific modules "board", but to call all mcu-specific modules differently based on the mcu they support. This really looks to me like a change for the sake of changing, and doesn't really bring any value.

On any given running instance of micropython you are going to ever have only one of each of board-specific and mcu-specific modules. So naming them differently to avoid name conflict doesn't make sense -- there is no conflict possible, as there is ever only one of each of them.

However, naming the mcu-specific modules differently makes it very difficult to write tutorials and documentation for any peripherals and driver libraries. It would basically require me to provide example programs for all the different platforms that I want to support. Naming them all "machine" and having a common API for all of them lets me write example code and tutorials for my drivers easily.

Conversely, since no two boards are the same, there is no way to have a common API for the board-specific modules. And each board has different connections, different built-in hardware, often even different markings on the pins. The documentation and tutorials for the boards and their internals need to be written specifically for the given board, so there is no advantage in naming them the same. In fact, since every "board" module would necessarily have different content and different API, naming them all the same will make users confused. Finally, when writing an application that runs on different boards, you can provide example configurations for each of them, and even detect them by trying to import the right module.

I agree that some of the names for the board modules are unfortunate. The `pyb` should really be called `pyboard`. The `esp` module is actually super-weird, because it in fact contains mcu-specific, not board-specific things. It should probably be merged with the `machine` module, and instead there should be things like `huzzah` (Or how is that Adafruit board actually called? There seem to be a dozen of "Huzzah" boards, each with a different mcu, very confusing.), `nodemcu`, `d1mini`, etc. But that's a historical thing, and it's too late to go around changing it all. That ship has sailed.

In the end, I think that we have a lot of much more important problems (like actually making the APIs consistent), and we won't solve them by just renaming everything every time someone gets a new idea.

torwag, 2016-11-04:
Not exactly,
I propose to use packages named soc and board which contains modules named after the soc names and board names.
Thus you have e.g. soc.STM32F4 and board.pyboard
The benefit is not to avoid name conflicts, it is about the clean separation between soc functions and board functions, which as you already pointed out is right now sometimes confusingly mixed.

It would also help people to understand which mcus are generically supported (used on other boards) and hence help them to bring micropython to new boards of the same mcu family, by simply contributing a board module. I guess it comes along with the current folder structure as well, as it was discussed elsewhere. 

I believe that this is needed if micropython continuous to grow.  Sure ship is sailed and it comes late, I wrote that already. 
I know core devs can argue that there are more urgent problems to be solved. True! But if there is a chance to make more people to contribute, e.g. by writing board modules for already existing soc modules, we should at least discuss this.
On a personal note (not particular related to your post), I feel that the recent communication level started to drift towards a more rough tone. I would be happy, if we all could keep by the initial "be nice to each other"-motto.

kamikaze, 2016-11-04:
yeah. Upvoting for a soc and board separation. Let's bring more freedom to board/port creators via pluggable functionality! )

pfalcon, 2017-05-14:
No further progress (no patches, no nothing), while machine module continues to be maintained and elaborated. Closing. Feel free to reopen with suitable backing information.

closed 2017-05-14 by pfalcon