micropython/micropython #2596: unix: Ctrl+C while input() produced empty line

pfalcon, 2016-11-04:
Instead of raising KeyboardInterrupt.

Resolving this would include completely getting rid of GNU readline code.

dpgeorge, 2016-11-05:
Would be good to use a common input() builtin for all ports, to make them all work the same.

I'm happy to remove GNU readline support.

prusnak, 2016-11-17:
The following fixes the issue for me:

```
diff --git a/unix/input.c b/unix/input.c
index 15ee60e..7c05568 100644
--- a/unix/input.c
+++ b/unix/input.c
@@ -51,7 +51,9 @@ char *prompt(char *p) {
     mp_hal_stdio_mode_orig();
     if (ret != 0) {
         vstr_clear(&vstr);
-        if (ret == CHAR_CTRL_D) {
+        if (ret == CHAR_CTRL_C) {
+            nlr_raise(mp_obj_new_exception(&mp_type_KeyboardInterrupt));
+        } else if (ret == CHAR_CTRL_D) {
             // EOF
             printf("\n");
             return NULL;
diff --git a/unix/main.c b/unix/main.c
index 482d194..40f9557 100644
--- a/unix/main.c
+++ b/unix/main.c
@@ -181,7 +181,7 @@ STATIC int do_repl(void) {
 
         if (ret == CHAR_CTRL_C) {
             // cancel input
-            mp_hal_stdout_tx_str("\r\n");
+            mp_hal_stdout_tx_str("\r\nKeyboardInterrupt\r\n");
             goto input_restart;
         } else if (ret == CHAR_CTRL_D) {
             // EOF
```

dpgeorge, 2017-05-28:
Since the built-in help() function has recently been moved to the core py/ directory, it's probably sensible to also provide a standard version of input() in the core.  In fact, most (all?) bare-metal ports are currently using stmhal/input.c as their input() implementation so I propose to move this file to py/builtininput.c (or even just put the code in py/modbuiltins.c since it's pretty small) and then provide a config option to include this function.  After that, the ctrl-C issue can be fixed and it'll then be fixed for all ports.

pfalcon, 2017-05-29:
@dpgeorge : Probably sounds good.

dpgeorge, 2017-05-29:
Looking at this in more detail I remember that the unix port has the option to not use mp-readline, ie it just uses `fgets()`.  To have a core-provided input() function that doesn't use mp-readline will complicate things.  Options are:
1. Remove fgets() option from unix port and support only mp-readline based prompt/input (this will grow unix minimal build by a bit).
2. Provide (another) config option in the core to configure the input source for the built-in input() function (eg it's either mp-readline or something custom).
3. Not enable input() for unix minimal build.

I'd prefer option 1 or 3.

stinos, 2017-05-29:
I'd be leaning towards 1: there are enough configuration options and ifdefs alerady. Then again I'm not 100% sure what the intent of the minimal port really is - if it must be truly minimal both in size and number of enabled options then 3 might be the better choice

dhylands, 2017-05-29:
The typical way to do any system call in linux is to do something like this (off the top of my head):
```
do {
    ret = some_system_call()
} while (ret < 0 && errno == EINTR);
```
This applies to almost all blocking system calls: open/close/read/write/whatever.
The interrupt could be any type of interrupt (SIGINT for Control-C, SIGALARM if you're using alarms, and a whole plethora of other signals.

If the signal handler is registered using sigaction with the SA_RESTART option, then the OS will automatically do what amounts to the above loop for many calls (including read).

dpgeorge, 2017-06-01:
Ok, it turns out that unix minimal build never had the built-in input() function, so that makes things simpler.  But the non-minimal build (that uses mp-readline) still needs special handling (cf bare-metal ports) because it needs to disable/enable raw stdio mode around the call to mp-readline.

So what's been done now is:
1. In bc76302eab56368967af4f1cf8813ac816b9c22f a core-provided version of input() is added (copied from stmhal), with proper handling of ctrl-C (different to how @prusnak suggests above, which won't work correctly for all cases).  This is controlled by MICROPY_PY_BUILTINS_INPUT, disabled by default.
2. In 6ff0ecfffcea4e7aa6301c54870ae97aa4a0cf72 all bare-metal ports are converted to use the core-provided input.  This cleans things up by making them less dependent on stmhal's code.
3. In d92898a35acdb6fc665be75f7c53d0fec107c4c4 unix is converted to use the core-provided input(), with a wrapper (mp_hal_readline) to handle raw stdio mode correctly.

closed 2017-06-01 by dpgeorge