micropython/micropython #2598: FatFs's os.rename() is not POSIX complient

pfalcon, 2016-11-05:
Trying to do os.rename("old", "new") with FatFs fails if "new" already exists with `OSError: [Errno 17] EEXIST`. This is not POSIX compliant:

> If newpath already exists it will be atomically replaced (subject to a few conditions; see ERRORS below), so that there is no       point at which another process attempting to access newpath will find it missing.

(from man 2 rename)

https://docs.python.org/3/library/os.html#os.rename says on this:

> Rename the file or directory src to dst. On Unix, if dst exists and is a file, it will be replaced silently if the user has permission. On Windows, if dst already exists, OSError will be raised even if it is a file.

So, we could leave it raising OSError, but that would be emulating Windows, and we should emulate POSIX apparently.

hosaka, 2016-11-06:
I think adhere to posix is the best option. I remember noticing this not long ago, but forgot to mention this.

pfalcon, 2016-11-06:
I think the same - like, VFS interface should adhere to POSIX definition of operations, and each particular FS should implement it.

dpgeorge, 2016-11-07:
+1 for POSIX behaviour.

robert-hh, 2016-11-07:
In principle I am in favour of a POSIX compliance, although  I do not like the idea of a silent replace.

deshipu, 2016-11-07:
The Windows way makes it very hard to update file's contents atomically -- which may become problematic on systems that have threads or other ways of doing concurrency.

robert-hh, 2016-11-07:
After writing, I considered the use cases. A typical application which uses the os.rename() call would/could check anyhow beforehand if the target exists, and act accordingly. Given that, the POSIX behaviour is fine.

dpgeorge, 2016-12-02:
POSIX behaviour for rename() was implemented in b7df3e541a49a1cb72862c4015b6a883a63e1e29.

Small caveat: renaming a directory to an existing file will give an error for POSIX but will succeed for the vfs fatfs implementation.  It would require an additional stat (of the old directory) to fix this and I don't think it's worth it.

closed 2016-12-02 by dpgeorge