micropython/micropython #2602: ANNOUNCEMENT: Clearing the backlog

dpgeorge, 2016-11-06:
Hello all.

MicroPython has grown into a piece of software that is used by many people in many different systems for many different purposes.  There are also lots of people who have contributed to the project in various ways, not only through code.  This is fantastic and is what open source is all about.

But, with such a large scope, the maintenance of the code and managing of the community and their contributions has reached a point where it takes more time than is available.  There are currently 2 core developers/maintainers of this project (@pfalcon and myself, @dpgeorge) and we have become overwhelmed by the volume of work.  The number of issues and pull-requests grows faster than we can deal with them.

To get back on top of things we have decided that for the next few months (at least until start of 2017) we will focus primarily on:
  1) reducing technical debt and making things more efficient to maintain;
  2) addressing existing issues and PRs to clear the backlog;
  3) finalising the machine API and improving documentation;
  4) working through the ESP8266 Kickstarter stretch goals.

As such, any new issues or PRs opened after this one may take a while to be reviewed and responded to until we have dealt with the backlog.  We will try to process new bugs and security issues for the Python core and active ports, so please do file bug reports if you have found something.

This issue is not meant to discourage submission of new bugs or patches but rather gives an update on the current state of the project and plans for moving forward.  As usual, contributions for items in #1331 are always welcome.

Thanks for your understanding!

prusnak, 2016-11-08:
Is there a way how other people can help you with the project and its backlog so you don't feel overwhelmed?

dpgeorge, 2020-06-09:
There's still a backlog, but I don't think it's helping to keep this issue open, so closing.

closed 2020-06-09 by dpgeorge