micropython/micropython #2603: RFC: Introducing machine.Signal class

pfalcon, 2016-11-06:
This RFC comes from a problem posed in https://github.com/micropython/micropython/pull/2571 : while application in it, while should work the same way on all devices, show different behavior depending whether there's an active-low, or active-high LED. It (RFC) is also a result of initial discussion of the problem done in http://forum.micropython.org/viewtopic.php?f=3&t=2545 .

So, what's proposed here is to add "Signal" object to handle a hardware signal inversion feature, while leaving Pin object intact. To avoid complex constructors like:

~~~
led = Signal(Pin("A0"), inverted=True)
~~~

It's proposed Signal to accept all the same parameters as Pin (and internally use them in the call to Pin constructor), and additionally "inverted" keyword arg (name can be discussed).

Signal object can have following methods: assert(), deassert(), is_asserted() - likely, in addition to standard Pin methods.

Formulated like the above, Signal class has implementation benefits comparing to adding "inverted" kwarg directly to Pin:

* Simplest implementations of Pin which don't have internal state, can continue not having it.
* Signal class can be implemented once in port-independent manner and work for any port.

deshipu, 2016-11-06:
Why does it need to live in `machine`? It looks like a perfect candidate for a wrapper class written in Python. It could live in the same place as the i2c/spi endpoint wrappers, for instance -- as it is in fact a kind of an endpoint, just for a single pin. It would be used in a similar way the endpoint is -- each driver would create its own, as different drivers may have different expectations for whether it is inverted or not.

pfalcon, 2016-11-06:
Because we want standard hardware API, and efficient too (i.e. done in C). Note that while current suggestion i2c/spi endpoint classes is to be implemented in Python, we'd move them to machine and C implementation once proven to be really useful.

Signal class is supposed to replace all the LED, Relay, Beeper and hundred of other classes, none of which actually even exists. Like not replace, but make optional and not needed generally (one can write one for themselves, not expected to be part of HW API).

dhylands, 2016-11-06:
+1

I like it. As @pfalcon suggested, Signal needs to have internal state, where Pin typically doesn't, which allows Pin objects to exist in flash, and Signal objects need to live in RAM.

pfalcon, 2016-11-06:
Right. Majority of Pin (and other HW objects) implementations would be (and *are*) singletons, just as noted by @deshipu in https://github.com/micropython/micropython/issues/1430#issuecomment-253200785 . However, in a general case that's not possible (current Zephyr port, and likely any other OS/RTOS port). And Signal by definition will be dynamically allocated. (This necessitate scheme for ensuring singleton-ness, and it's given in https://github.com/micropython/micropython/pull/2571)

peterhinch, 2016-11-07:
+1 I take it ``assert()``, ``deassert()``, ``is_asserted()`` take no args. So I'd suggest a ``__call__()`` method allowing a boolean variable to be assigned to a Signal:
```python
my_led(is_lit)
mysignal(not mysignal()) # toggle
```

pfalcon, 2016-11-07:
>  I take it assert(), deassert(), is_asserted() take no args.

Yes.

> So I'd suggest a __call__() method 

So, @dpgeorge said he's ready to give up on `__call__` for Pin. That really needs to be discussed (likely, practically tested), but we for sure not remove it from Pin, but add to Signal - there should be consistency.

As written above:

> likely, in addition to standard Pin methods.

I.e. Signal would have .value(x), .value(). If Pin keeps `__call__`, Signal would have it too.

dpgeorge, 2016-11-08:
In general I like the approach of having a flatter machine API, ie more classes each with specific and minimal/precise functionality, rather than less classes with each having more depth (more methods, more parameters, more options).  Signal fits into that idea of a flatter API.

I don't particularly like the name "Signal" because it reminds me of unix signals and interrupts, but I don't have a better suggestion right now.

> It's proposed Signal to accept all the same parameters as Pin (and internally use them in the call to Pin constructor), and additionally "inverted" keyword arg (name can be discussed).

So you'd do something like: `machine.Signal(12, machine.Pin.OUT, drive=machine.Pin.MED_POWER)`, ie need to use constants from the Pin class?

IMO it's more composable (and mirrors things like I2C and SPI) to just make Signal take a pin object, like in the very first example above.  What if the user wants to create a signal from a pre-existing pin object?

> Signal object can have following methods: assert(), deassert(), is_asserted() - likely, in addition to standard Pin methods.

How about on() and off()?  That would make code very readable, eg `led.on()` cf `led.assert()`.

But why are these methods needed at all if there's also value([x])?  Wouldn't generic code need to use value() anyway to support both pins and signals?

Will it be allowed to pass signals to constructors like I2C and SPI to create instances of those buses using inverted pins?

I assume that signals are readable if the underlying pin is in IN mode, and that signal objects invert the return value from a signal.value() call.

dhylands, 2016-11-08:
If Signal implements all of the same API as Pin, then you can pass a Signal object anyplace you can pass a Pin object. For Signal I would expect assert to synonymous with value(1) and deassert to be synonymous with value(0) (for the Signal class). I think on and off are definitely clearer than assert/deassert. From a HW perspective, I think of signals as being active high and active low, and you acivate or deactivate a signal, but I would still expect on/off to be clearer from the users perspective.

deshipu, 2016-11-08:
If the only purpose of this is to reverse the low and high signals, why don't we explicitly call it `ReversedPin` or `InversePin`, and make it identical to `Pin`, except for the inversed `value`?

As for making it from an existing `Pin`, perhaps the `reversed` builtin could do it?

pfalcon, 2016-11-08:
> I don't particularly like the name "Signal" because it reminds me of unix signals and interrupts, but I don't have a better suggestion right now.

Well, it's in `machine`, so thinking should go along those lines. Do you have better/other ideas for naming? I don't see anything beyond colloquial/amateurish "Wire".

> So you'd do something like: machine.Signal(12, machine.Pin.OUT, drive=machine.Pin.MED_POWER), ie need to use constants from the Pin class?

Constants in a Pin class is an issue on its own. It's of course right from OO perspective, helpful for completion, etc., but leads to long stuff like `machine.Pin.OUT`, just imagine it could have been `machine.OUT`.

But answering your question, yes, as Signal is extension of Pin, it uses its constants. There's even way to address that, e.g. implement proper inheritance support for native classes, but will machine.Signal.OUT make anything clearer?

> MO it's more composable (and mirrors things like I2C and SPI) to just make Signal take a pin object, like in the very first example above. What if the user wants to create a signal from a pre-existing pin object?

Of course we can and would do that, the point is not forcing users to do it like that all the time, but providing clearer and more light-weight syntax.

 > How about on() and off()?

Sure. But the biggest problem aren't they, but  a long is_asserted(). is_on()? Works, but not ideal.

> But why are these methods needed at all if there's also value([x])?

To not confusion. While concept of inversion is simple and *should* be understood by every beginner (unlike for example open-drain), actually working with .value(1) which will set pin to 0 may cause confusion.

> Wouldn't generic code need to use value() anyway to support both pins and signals?

What generic code? Let me quote from the forum discussion: "LEDs and Pins are different objects, physically and logically." Signal is a generalization of all those "LEDs", and in general, Pins and Signals would be worked on by different algorithms. Of course, there's a big deal of correlation, and I'd personally like to see a way to do "inverted I2C", but that's of research practical value.

Btw, one of the reason for going with a separate Signal object is to not slow down bitbanging I2C, SPI, etc, which work on Pins.

> I assume that signals are readable if the underlying pin is in IN mode, and that signal objects invert the return value from a signal.value() call.

Sure. And to signal.value() call. But primary conceptual interface here is assert/deassert.

pfalcon, 2016-11-08:
> Let me quote from the forum discussion: "LEDs and Pins are different objects, physically and logically." 

To clarify, that was @peterhinch's argument against adding inversion into Pin class.

dpgeorge, 2016-11-08:
> What generic code?

How about https://github.com/micropython/micropython/blob/master/examples/hwapi/soft_pwm.py#L13-L21 ... surely that function should remain as-is (calling .value) so it works with pin objects.

And what about a driver that has a CS pin: that would also need to use .value because you should be able to pass a pin into it.

Are there any examples of algorithms that would accept only a signal and not a pin?

> Signal is a generalization of all those "LEDs"

Is there any scope then to encompass the existing pyb.LED class and allow for .intensity() method?

My primary concern with signal is this: on the one hand it's trying to be exactly like a pin (same constructor, inherits all methods), but on the other hand it's trying to be different: you're encouraged to use assert/deassert methods and algorithms will be written that accept only signal objects.

pfalcon, 2016-11-08:
> How about https://github.com/micropython/micropython/blob/master/examples/hwapi/soft_pwm.py#L13-L21 ... surely that function should remain as-is (calling .value) so it works with pin objects.

Well, nope, with Signal availability that's expected to use its native interface.

> And what about a driver that has a CS pin: that would also need to use .value because you should be able to pass a pin into it.

That's better example, CS is really a Signal, but we won't force it (I mean, suggest users to always use Signal for CS) for practical reasons.

> Is there any scope then to encompass the existing pyb.LED class and allow for .intensity() method?

No, that would be a true LED class if someone will ever want to RFC and implement it.

> My primary concern with signal is this: on the one hand it's trying to be exactly like a pin (same constructor, inherits all methods), but on the other hand it's trying to be different: you're encouraged to use assert/deassert methods and algorithms will be written that accept only signal objects.

That's because pins and signals are close, but different types of objects, so all is good. Implementation-wise the difference is clear: Signal maintains extra state beyond "pure hardware", Pin doesn't have to. Whether their interfaces purposedly overlapped or purposedly disjoint (to avoid confusion) is also an implementation question.

> and algorithms will be written that accept only signal objects.

To clarify, the algorithms on signals are expected to be generally simple and application-level. Like, "turn on heater for N minutes". But it's important to provide a model where any heater can be turned on, and when it's actually off initially, then on for N minutes, then off. Not on initially, off for some, then starts to burn the environment - because user mixed up something, or didn't have a chance to configure their heater properly with a pre-existing algorithm.

dpgeorge, 2016-11-08:
> How about https://github.com/micropython/micropython/blob/master/examples/hwapi/soft_pwm.py#L13-L21 ... surely that function should remain as-is (calling .value) so it works with pin objects.
> 
> Well, nope, with Signal availability that's expected to use its native interface.

But that's reducing the generality of the pwm function.  Why would you want to make it less general?  Using .value() will make it work with pin and signal objects, and do the right thing in both cases.

> That's better example, CS is really a Signal, but we won't force it

So in the code that controls CS, will it do cs.value(0) to *assert* the line (since it's active low), or will it do cs.value(1) to assert, assuming that it's an inverted signal?

pfalcon, 2016-11-08:
> But that's reducing the generality of the pwm function. Why would you want to make it less general? Using .value() will make it work with pin and signal objects, and do the right thing in both cases.

Because specifically that is *example* of *fading an LED*. It's PWM isn't supposed to be general, it's supposed to show how write portable apps supporting variety of user hardware well (for the latter, you need to use higher-level abstraction than just Pin, namely Signal).

> So in the code that controls CS, will it do cs.value(0) to assert the line (since it's active low), or will it do cs.value(1) to assert, assuming that it's an inverted signal?

We'll keep it like it is now: CS is a Pin, and we assume the most common case for it: it's inverted. But if we'll make Signal to be passed there, good for us.

pfalcon, 2016-11-09:
Btw, linux provides inversion support for its /sys/class/gpio interface (attribute name is `active_low`, now seen in more than one place, and perhaps we should use it instead of `inverted`).

pfalcon, 2016-11-09:
FYI, initial implementation of machine.Pin for Linux is made: https://github.com/micropython/micropython-lib/commit/7ac2cf662e8c66eb8bc333dc5554714ceef07fd5 . It was tested only for output so far, using soft_pwm.py example:  https://github.com/micropython/micropython/commit/29f3f84fbd90e5d43ba5d6c621d19021a45d184d

This all naturally leads us to https://github.com/micropython/micropython/issues/2615 (though we already have been there with https://github.com/micropython/micropython/pull/1611)

dpgeorge, 2016-11-10:
> Btw, linux provides inversion support for its /sys/class/gpio interface (attribute name is active_low, now seen in more than one place, and perhaps we should use it instead of inverted).

But what if the input object to the Signal constructor is a user-defined Pin that is already inverted, for example?  Signal provides inversion on the underlying object so "inverted" seems a better choice for the param.

A related point: will one be able to wrap a Signal in a Signal to do double inversion?  It seems logical to be able to do that, especially if you get passed an object which you don't know if it's a Pin or Signal and just want to invert the digital logic.

pfalcon, 2016-11-10:
> Signal provides inversion on the underlying object so "inverted" seems a better choice for the param.

I'm in favor of "inverted", just mentioned "active_low" as a possibility.

> A related point: will one be able to wrap a Signal in a Signal to do double inversion? It seems logical to be able to do that, especially if you get passed an object which you don't know if it's a Pin or Signal and just want to invert the digital logic.

That depends on whether we treat Signal as a true subclass of Pin, or as a wrapper with somewhat disjoint interface. That in turn depends on whether we want to "policy" users to use only Signal objects in some contexts, not Pins. I never was a proponent of such "policying" - users who will want to do it wrong will find a way to do it, and for doing it right, good documentation and examples are should be easily available.

I guess it makes sense to make a summary - are there serious objections to having Signal class, or viable alternative proposals? If no, we can discuss the API of Signal, and (almost) finalize API of Pin (because Signal will definitely affect "broadness" of Pin API, and that's the whole reason for bringing it into picture).

chrismas9, 2016-11-10:
> are there serious objections to having Signal class, or viable alternative proposals?

Yes, but I won't be able to elaborate until tomorrow

hoihu, 2016-11-10:
It seems to me that this addition rather adds confusion than actually clarify things.

If I have an LED wirh active low driven, then why not just use a small LED wrapper class and invert the logic there? Am I missing here something?

dhylands, 2016-11-10:
> It seems to me that this addition rather adds confusion than actually clarify things.
> 
> If I have an LED wirh active low driven, then why not just use a small LED wrapper class and invert the logic there? Am I missing here something?

The signal class wouldn't only be for LEDs. There are lots of other devices which also have active low interfaces.

chrismas9, 2016-11-11:
I think pin has been around too long and its use is too entrenched to expect a wholesale change to signal for GPIO interfacing.

I do think adding invert would be very useful for code and library portability. But it's not just needed for pin. PWM needs it for two reasons - active low timer pins (eg pyboard pin Y6 (PB13)) and active low devices such as blue LEDs driven from 5V with a open drain driver. In the case of PWM invert would do  Duty Cycle = (100-n)%. Uart needs invert for Kinetis MCU's which have polarity inversion in hardware. Active high UART pins are useful for certain interfaces.

I don't think assert, deassert, on, off, etc are necessary (users can define them if they really want to). Just adding "inverted=true" as an optional parameter is enough. This could be added to pin and pwm, plus uart and any other classes that need it for a particular MCU. Other than poking at MCU registers there is currently no way to select hardware polarity on MCUs that support it.

My suggestion to keep things simple is:

1. Add "inverted = true" option to pin class. This alone provides a way to support active low devices without breaking current usage of pin.
2. Add "inverted = true" option to pwm class. It's needed for all the reasons pin needs inverted.
3. Add "inverted = true" option to uart class. @dhylands could support it for Teensy
4. Add a pin_raw or pin_fast class for the few power users and for soft I2 and neopixel that need fast IO.

Basically I am suggesting rename pin to pin_raw and signal to pin. This achieves the purpose of supporting active low devices with minimum of disruption and confusion.

> Why does it need to live in  machine?

Some MCUs support hardware inversion of pins (eg Kinetis Uart). So using the same argument as for Open Drain (It's provided in hardware on some MCUs and can be emulated on others) invert can be included in machine and implemented in hardware or by emulation. For MCUs with hardware polarity inversion the support should be in machine, as it is a machine option.

> @dpgeorge I don't particularly like the name "Signal" because it reminds me of unix signals and interrupts, but I don't have a better suggestion right now.

If you want to stick with pin and signal what about GPIO, UserIO, DeviceIO or DevicePin or something else that indicates it is controlling a 1 pin device such as LED or Relay. I don't associate signal with physical devices either.

> Is there any scope then to encompass the existing pyb.LED class and allow for .intensity() method?

That's trying to mix two interfaces (pin and pwm) into a higher level driver and I don't think complex drivers should live in machine. I will discuss my thoughts on the fate of LED, ACCEL and SERVO elsewhere.

dhylands, 2016-11-11:
The issue with adding the inverted flag to the Pin class is then the class can't be stored in flash since it would have non-register state.

Actually, thinking about it some more, it could be done by having a uint16_t in RAM for each PORT which would store the inversion state for every pin (so about 20 bytes of storage to cover ports A-K)

pfalcon, 2016-11-11:
> Add "inverted = true" option to pin class.

Surprisingly, that's what I proposed in http://forum.micropython.org/viewtopic.php?f=3&t=2545 . That didn't receive *single* understanding response from anybody else. I've regrouped and now with the current proposal. You're welcome to move forward with my old one though.

> 1.
> 2.
> 3.
> 4.

I've shared my suggestions about such things in http://forum.micropython.org/viewtopic.php?f=3&t=2545&start=50#p15477 , the part in bold:

> Choose 3 (three) things you would like to change in the current Hardware API the most, and submit them on github.

So, that would be separate tickets, except to keep us from stack overflow, only 3 per person are proposed. And if 3 submitted now, then apparently more can't be submitted later (until next HW API iteration, but we're stuck at the current iteration last 1-1.5 years).

> Please choose carefully

Finally, submitting proposal doesn't mean it's going to be implemented, so, besides technical side (which is worth very careful thinking out - is it possible? is it plausible? is it feasible?), please consider organization part - who's going to implement that? Needless to say, I'm implementing this proposal if it's accepted.

chrismas9, 2016-11-13:
Some of you may wonder why I pop up here from time to time, but don't contribute any code. I am a hardware design engineer. I have designed pyboard and MicroPython into commercial products for clients. I expect to be using MicroPythonn myself next year and employing a software engineer to contribute ports and drivers. I am very keen to see MicroPython succeed in the commercial world. This will lead to more companies like Pycom, Adafruit and mine contributing to microPython and hopefully reducing Damien's and Paul's workload. The only way I can contribute to MicroPython right now is to give all the developers and contributors a perspective of what's important to commercial hardware designers and end users. I feel that the direction of many software projects is sometimes directed towards whats pure and beautiful or sometimes easy (that's not referring to MicroPython) from a programmers point of view to the detriment of what's needed to make the whole software / hardware / user experience right. I do lots of things in hardware that some people may not know about or don't think is important. Everything I suggest is based on real world experience and is carefully considered. I don't expect every suggestion to be implemented, but by putting them out there they can be considered. 

Recently I spent 2 days designing a resistor out of a circuit so I could fit the design into a standard shielding can instead of a custom one. Removing that $0.001 resistor saved $1. So when I suggest allowing for adding "inverted=True" to the uart class by someone in the future it's not just so I can save a $0.05 inverter IC, but maybe so I can fit a QFN MCU instead of a chip scale BGA and microvias. Likewise my suggestions about open drain and inverted pwm.

In ultra reliable safety related systems there a 3 processors. Two are redundant controllers, the third is a supervisor. The main processors are programmed in different languages (eg C and ADA). One is programmed with positive logic (if A is high and B is high do C) and the other in negative logic (if A is low or B is low stop doing C). The supervisor is like a smart watchdog but it can't reset a controller without agreement from the other controller. There are several FPGAs or CPLDs which merge the controller outputs. The're programmed in VHDL and Verilog, again one with negative logic. They have their own watchdogs on IO signals. So when I suggest that pin.value() should be able to read the pin in PIN.out mode it lets you compare pin.out_value() with pin.value() to detect contention or hardware failure. Why not C and MicroPython? Wouldn't it be good if ESA or Boeing or Tesla or someone adopted MicroPython and worked on say heap defrag so MicroPython could run for 20 years without a reset?

> So, that would be separate tickets, except to keep us from stack overflow, only 3 per person are proposed. And if 3 submitted now, then apparently more can't be submitted later (until next HW API iteration, but we're stuck at the current iteration last 1-1.5 years).

Of particular concern is that there is talk that the hardware API will be locked at v2 with no way to add modes, methods or classes to support more advanced hardware or needs of users. If I submit a PR to add inverted to Uart for Kinetis next year is it going to be automatically rejected for 1.5 years? That's going to lead to forks and incomopatible APIs. MicroPython has to have aa way to grow as new features are needed.

from @dhylands:

> Some other examples where you need to specifically initialize pins:
•Quadrature decoding needs 2 or 3 pins. The Timer API only allows a single pin to be specified - which covers off most non-quadrature usages. You need to use the machine.Pin to set the appropriate alternate functions for the other pins not initialized by the TImer API.
•ADC sampling using differential signals needs 2 pins. The ADC API only allows a single pin to be specified - which covers the more typical single-ended use cases. You need to use the machine.Pin API to set the alternate function for the pin not initialized by the ADC API.

Will the developers / maintainers even consider adding Quadrature decoding or differential ADC to machine before v3 if there is a good argument for it? I'm not talking about who does the development, but policy of allowing the machine API to evolve (in a backward compatible way so as not to break code and libs).

#####################################################

Now back to Signal and the specifics of my last post. All of my comments were in response to two specific questions asked in this thread:

> Q1. Do you have better/other ideas for naming? 

> Q2. are there serious objections to having Signal class, or viable alternative proposals?

Q1. For an active high device, eg a LED with its cathode to GND there will be two equally viable methods of control - pin.value(x) and signal.value(x). I don't think you will get wholesale conversion of users to signal. I think this will lead to confusion and break portability. I think others agree. So my answer to Q1, a better name, is to call signal pin and pin something else. I'm not telling you how to write the signal class, just give it a different name. This simple name change has two consequences. 1. 100% of users will start using the signal class because they will keep using pin It will work the same way and they won't notice any difference. This is what we want for portability. 2. A few sped freaks may have performance issues with t [... cut ...]
> Finally, submitting proposal doesn't mean it's going to be implemented, so, besides technical side (which is worth very careful thinking out - is it possible? is it plausible? is it feasible?), [...]

I am well aware of that and appreciate all the effort you are putting into MicroPython. That's why my proposals were intended to have minimum demand on developers for maximum benefit. 1. a name change only, 2. invert swaps the duty cycle value, 3. one line in the documentation.

I am working very long hours on a commercial project. Unfortunately it means MicroPython usually only gets a look in after midnight and I can't always respond in a timely manner.

pfalcon, 2016-11-13:
> I am working very long hours on a commercial project. Unfortunately it means MicroPython usually only gets a look in after midnight and I can't always respond in a timely manner.

Reading such things keep me perplexed... The most obvious approach for someone who is in the position above is to assume that everybody else is in the same position. But then why tell busy people that you're busy? Everyone in the room already understands that.

That makes me wondering is something else is being meant by sayings like the above. I may imagine a few things could be possibly hinted:

1. You hint that you're the only one in the room doing "serious business", and everyone else ... well, maybe cheerful students, giggling, sitting on top of a synchrophasotron and refusing to add Kinetis UART support to it.
2. You hint that you're very overworked person, while everyone else is apparently a millionaire who works on MicroPython for fun, between chasing new car and fashion models.

We'd need to reconcile our positions, because otherwise there will never be understanding on the other questions. And I may assure you that there're no either students or millionaires in active MicroPython contributors.

deshipu, 2016-11-13:
@pfalcon no need to attack people personally again.

deshipu, 2016-11-13:
> So my answer to Q1, a better name, is to call signal pin and pin something else. I'm not telling you how to write the signal class, just give it a different name. This simple name change has two [...]

This will not work like you imagine, for the simple reason that the peripherals that provide PWM, UART, I2C, SPI and even NeoPixel and OneWire are not implemented in Python, and will not call the public functions of the `Signal` class. In fact, most of them will not even call any C functions -- all they need from the pin is on what port it is and which bit on that port, and then they will just configure the microcontroller's hardware to use those bits on those ports for those functions. With a few exceptions (notably, the SPI modes and some rare UART peripherals on some boards) you don't even have any ways to tell the hardware to invert the signal.

The only case where using `Signal` makes any practical sense at all is when you are toggling an individual gpio pin from Python code, and even then it makes much more sense to do the inversion in the device driver code.

chrismas9, 2016-11-14:
> Needless to say, I'm implementing this proposal if it's accepted.

My comment about work was actually an apology for my late submission. You are obviously ready to start now and I was apologising for not being able to review and comment on this in detail a few weeks ago. I didn't expect a big rant about millionaires. If I could get my head around C programming (I have tried) I would be submitting PR's instead of comments. 

I feel that some of the decisions about MicroPython features need a better balance of catering to hardware, system and user needs. I am just trying to give the decision makers who are all skilled programmers a better idea of the other side of the world. Some of the ideas, only a programmer could love. As an example when machine was first proposed the architects were adamant it should be machine centric. For example a 12 year old with their first pyboard or micro:bit, or an artist, biologist, etc had to understand the insides of an MCU, pick a timer from the available alternatives (TIM2), then an MCU pin (PA0) then a board pin (X0). I argued for a board centric option and now you can instantiate a PWM object directly onto a board pin. Its nearly as easy as AnalogWrite. Some of the implementations don't make sense when you look at how the hardware is used. Pin.value() can't read the pin state of an output. Why not? The MCU vendors didn't add that capability to their GPIO logic just so they could use more silicon and make their MCU more expensive. They did it because there is a need for it that currently MicroPython doesn't support.

If you want to design MicroPython to only appeal to skilled programmers then there is no point in my contributing. If you want MicroPython and pyboard to maintain the same low entry barrier as Python then I think I can contribute a useful perspective from the other side. Obviously I don't understand software as well as you guys do, but maybe it's a bit the other way with hardware. Can't we try to work together to get the best outcome for MicroPython and its users.

I will try to avoid the how too's. I don't understand enough of the internal workings of the software so I will just briefly reiterate  my concerns and bow out.

1. Signal is meant to be a device driver for all the single wire things that need digital control and may be active low, including replacing LED. It doesn't achieve that aim because LED, RELAY, BUZZER, etc all need intensity (pwm) as provided by pyb.LED. I don't see signal as a complete solution without also adding invert to pwm.

2. The easiest way to handle invert is once when a pin / pwm/ uart / etc is instantiated. This provides best portability for user code, tutorials and higher level libraries. Signal only does this for pin, and not all the other classes that need invert.

3. Having two identical ways (from a user perspective) to the same thing  - pin.value(x) and signal.value(x) - will lead to lead to confusion and reduction of code portability.

4. It appears that machine will be frozen at v2 for 1 to 1.5 years, with no acceptance of change requests of PR's. If I need invert on a Kinetis UART I'd be happy to get it done and submit a PR. But what's the point if you are locking the machine interface. I'm not talking about breaking the API, just extending it in a compatible way as needed. Hence my raising the issue of incomplete support for invert now.

peterhinch, 2016-11-14:
If Pin.value() doesn't read the state of, say, an open drain output it's surely broken. Many protocols rely on being able to do this and it would be wasteful to require another pin to read the state.

pfalcon, 2016-11-14:
> My comment about work was actually an apology for my late submission.

Sorry Chris, but that's not true. You've been a participant in the forum topics from which this RFC was spawned, here are you messages, and you had more: http://forum.micropython.org/viewtopic.php?f=3&t=2545&start=20#p15042 , http://forum.micropython.org/viewtopic.php?f=3&t=2545&start=30#p15054 , http://forum.micropython.org/viewtopic.php?f=3&t=2545&start=60#p15667 . Note the last one was after mine "gentle closing note" proposing everyone to move to per-topic tickets on github, so you either read the whole thread and well aware of everyone's position, or you just chose to ignore some messages which doesn't lead us anywhere.

> You are obviously ready to start now

I'm ready to start in a way which doesn't cause strife or anxiety to other MicroPython contributors, so I don't know why you specifically are. I'm ready to start following Damien's call that we should finalize hardware API, starting from simple things like Pin. I'm ready to start about a very specific feature, and I start from the very specific example which is already "broken" to show that the problem is "unavoidable". I don't know why you turn this very specific into vicious circle of FUD ("rejected for 1.5 years") and offtopic.

>  I didn't expect a big rant about millionaires.

It's only only as big as it involves millionaires. It was much shorter than your posts, which are 90% offtopic for this ticket. And I still didn't receive a response, so let me reword it: how do you perceive MicroPython developers: a) as slaves chained to the wall and condemned to implement every whim of every passer-by; b) as god-like creatures, not subject to laws of time and material economy, which are destined to implement every whim of every passer-by thanks to their divine nature, or c) just as the  industry professionals as you, with a dayjob or commercial contracts, needing to support their families, care about their health, etc, again, the same as you, with the only difference - unlike you, with all that, they also contribute code to MicroPython and let everyone (including you) to use it for free. Please do not escape the answer this time.

> I feel that some of the decisions about MicroPython features need a better balance of catering to hardware, system and user needs.

But of course! Good half of the world has such feeling about many things. It's the same about CPython, Linux, Windows, maybe you want to call Bell Labs dudes back into 1960ies and tell them they should have done Unix quite differently, or maybe that they just should have named "sed" as "eds". It goes beyond just software, to anything else in the world - e.g. how many times you've called the president to tell him that you have a feeling that he doesn't get it, and then called next week to express concern that might have not understood the message from the previous call.

> I am just trying to give the decision makers who are all skilled programmers a better idea of the other side of the world. Some of the ideas, only a programmer could love.

So, why do you do it in such a way? You were gently hinted that you don't follow the procedure, your posts are offtopic, and that doing it like that doesn't help anyone. You ignored that and continue to push thru your stuff, and large part of that stuff is you explaining that you don't have time and that you're working on commercial projects. Is it only because this is an open-source project? But that doesn't mean it works like that and you can dump unstructured stuff on it. It works like any other project, like your own operation. Can you imagine me calling you (I use phone as the example because you aren't (apparently) doing an open-source process, if you did, I could use phrase "submitting a ticket") and telling something like "I'm not a hardware engineer and I don't know much about your business (heck, it's closed!), but I still have some helpful suggestions, which I really think you should implement:
* Drop whatever proprietary EDA tool you use and start using Kicad.
* If you find some features in it lacking, implement them yourself or hire someone to do that and share with everyone.
* Design a cheap ESP8266 board for MicroPython. Make it price competitive with chinese stuff, you may lose a lot of money, but guys like Tesla and Boeing may hear about it, and they will contract - here you might think that they may contract you, but no, they will contract me, but thanks to your work! You're still in the gain - that will alleviate your load, like you won't need to work hard to implement contracts from them.
* As a final note, can you design a board for free (then maybe couple more)? Why, you do it every day for years, it should be piece of cake for you.
"

So, can you imagine me calling your office and telling something like the above, then calling in a week to check if you really understood suggestions/requests? Then why do you do it in this manner here?

> Can't we try to work together to get the best outcome for MicroPython and its users.

Yes we can. Here're the steps:

* Keep calm and stop worrying about MicroPython, it's in good hands.
* Start developing code for features you need, then refactor them in a way they're useful for everybody else (or at least for many people), and submit as patches.
* If you can't do the above, please contract MicroPython developers to implement them for you.
[... cut ...] whose job will be also respond to such proposals timely. ($50K ain't big money, but I hope it will be possible to find someone who'd love to work on MicroPython, even if salary is not exactly competitive).

In the meantime, the only reason there's a talk of "3 per person" things is to gently hint audience that MicroPython developers aren't super-humans (not slaves either) and can work only on very limited number of things at time. How did you come up with the idea of "locking" is a pure mystery.

pfalcon, 2016-11-17:
Ok, so discussions like in https://github.com/micropython/micropython/issues/2612#issuecomment-260870772 shows that even maintainers "forget"/"mix on-spot" scope and directions of HW API. It's the same issue in nature as above and as discussed over a year in https://github.com/micropython/micropython/issues/1430 . And it boils down to: different people want and expect different things from HW API. Who are these people, what are their expectations can they be grouped by their expectations to make number of choices manageable? Yes, I think it's possible to do that. So, the stakeholders (as how these groups are called in such a kind of analysis) of HW API are:

1. People who'd like to use MicroPython as expected of a scripting language: to write portable applications. They would also like to write simple applications in a simple manner.
2. People who're interested in a particular hardware (board, MCU, SoC) and interested in MicroPython to let them get the most out that hardware.

It's clear that interests are pretty different and don't conflate to a single image of API, but rather diverge beyond some initial common point.

What can be said about Group1? Perhaps a surprising question whether such a group actually exist. The answer is in the objective proofs and exact names: @danicampora , while working on cc3200 port, considered it his important contribution to MicroPython, which idea was fully supported by the MicroPython author, the maintainer(s), and 18 contributors who participated in the discussion in https://github.com/micropython/micropython/issues/1430 . Unless someone wants to undermine @danicampora's efforts personally, and all efforts of other people in this direction, it's not even correct to pose the question like the about?

What can be said about Group2? Let's quickly sample subgroups of this group, who would be peopke in it:

* Silicon vendor employees who sell their MCU/SoC/etc.
* Hardware vendors who sell their hardware designs based on a particular MCU/SoC/etc.
* Beginner users who just start with a single particular board and don't yet know that there's nothing more transient than hardware, and in some half year, if they continue to tackle at all, they'll likely tackle with another board/MCU, the another, and another.

A few things become obvious: while this group share caring about a particular MCU, it's different for everyone in this group! Moreover, nobody makes a life-time oath to a particular MCU, so interests shift from one MCU to another all the time. The conclusion is obvious: while this Group2 definitely exists, its requirements can *never* be satisfied, because they're moving target, a chaotic particle movement. Only requirement of a particular member can be satisfied, for a given period of time.

With all the above in mind, the nature of my proposals should be very clear. Trying to even account for requirements of Group2 is what made us going in squirrel circles for more than a year. So, I propose for the maintainers to make a responsible decision of gently, but firmly move requirements of Group2 into a cozy, welcoming waiting area with a TV and background music, and concentrate on the requirements of Group1. Unlike Group2's whose requirements diverge over time, Group1's converge, and thus by spending not too big, but quality time on them, we can arrive at the no-longer-can-be-practically-minimized core feature set, on top of which we can add Level1 feature set. Level0 (core) and Level1 featureset interact, that's why one needs to bring Level1 to fully finalize Level0. This particular ticket considers Level1 feature - signal inversion.

@dpgeorge : Please ACK or NAK this approach.

deshipu, 2016-11-17:
I think you are very mistaken about how you categorize people around you into neat groups, and most of all, the way you label the groups is very unfair. I think that all of the users want the language to be simple, portable and powerful, and there is no special group that somehow is against that.

If you really have to divide your world into black and white, I think the line would line more between different uses of MicroPython, not particular people (though of course many people would focus on one of those uses). Those would be:

- Using MicroPython on actual microcontroller boards either to build physical projects that have to work in real world.
- Using MicroPython to build a personal toy operating system (or just its shell), often based on existing operating systems, such as unix, zephyr or a million other obscure projects, and generally having lots of operating system services, such as elaborate server setup applications.

I think the needs of those two use cases are very different, and hence the conflict.

Personally, I think the second use case if much less important when considering machine API, since the architecture astronauts rarely need to actually touch any hardware.

dpgeorge, 2016-11-18:
> @dpgeorge : Please ACK or NAK this approach.

At this point I don't ack or nack.

What I can say is that the majority of the funding for uPy has so far been centred around group 2: KS1 (the pyboard), KS2 (esp8266), Pycom (esp32 and their contribution to threading).

hoihu, 2016-11-18:
The first KS campaign. what was so exiting was the tight integration of HW features with a modern scripting language. No operating system. Having a python IRQ handler that gets called with usec latency. Inline assembler AND python code mixed. HW Timer acess etc. Wow. That was the magic that let me support uPy and introduce it into the company I work with.

The groups you are defining are arbitrary though

I'm not changing the MCU architecture every year just for fun. I'm not a beginner either - been doing staff on msp430 mcus for 13 years now.

The "genes" of uPy are tight HW interactions IMO. The "cozy waiting area" is an affront for all those who had supported uPy for exactly this reason. Yes there are different functionalities across MCUs. but if the solution is to search for the minimal common ground of something on every system (including linux..) you're likely to get... minimal functionality.

dpgeorge, 2016-11-20:
> People who'd like to use MicroPython as expected of a scripting language: to write portable applications. They would also like to write simple applications in a simple manner.

I'd say portability is not the main benefit of a scripting language.  You can already write portable applications with C.  Scripting languages give you rapid prototyping, higher-level constructs and a more enjoyable programming experience.  Python in particular gives you a whole set of standard libraries to make it even faster to get something working.

> People who're interested in a particular hardware (board, MCU, SoC) and interested in MicroPython to let them get the most out that hardware.

Pretty much all the uses of MicroPython that I know of fall into this category.  The pyboard is used for powerful IO, the microbit port is highly tuned to that device, the ESA port to LEON is about being able to control space payloads, OpenMV Cam is to control a camera, esp8266 port is to get the most out of that chip, Pycom's boards are all about latest tech and using uPy to get the most out of that tech in an easy way.

> Btw, the scope of Level0 should be very clear either - if something can't be done on (almost *) every MCU, it's not in scope.

It's very unclear to me how this scope is set.  Shall we create a list of, say, 100 MCUs and then go though them all to find the lowest-common-denominator feature set?  Is it just low-level MCUs that set the scope, or also the features that embedded OSs provide (eg mbed, zephyr)?  For example, how to decide on the following features:

- Configuring a pin to output mode, is that a level0 feature?.  Apparently ESP32 has a bunch of pins that are input only.  

- Reading input via `pin.value()` when the pin is set to OUT mode.  As far as I know, most MCUs allow to read the input independently of the drive.  Is this then a level0 feature?

I would like to see portability in the HW API.  I would also like to see a nice separation of "levels" in the HW API of "basic" and "advanced" features (eg Pin, Timer, vs QuadratureEncoder).  But I don't want to sacrafice useful features for the sake of ultimate portability.

pfalcon, 2016-11-22:
> It's very unclear to me how this scope is set. Shall we create a list of, say, 100 MCUs and then go though them all to find the lowest-common-denominator feature set?

Yes, that's a possible way, though I'd save the effort like that to answer questions (much later questions) like "Is it worth to standardize inverted UART support?" or "Does it make sense to provide standardized API for quadrature decoders?".

It's possible to define needed features in an easier way, e.g. "what is available within classical Boolean logic". So, values are 0 and 1, period. Given a Boolean expression `z = f(x, y)`. x and y are inputs, z is output, period. The Level 0 featureset defined.

> Is it just low-level MCUs that set the scope, or also the features that embedded OSs provide (eg mbed, zephyr)?

Apparently, basics of Boolean logic define the scope, and indeed, those would be available anywhere.  

> Configuring a pin to output mode, is that a level0 feature?. Apparently ESP32 has a bunch of pins that are input only.

It's typical design or theory-making choice whether to treat something as an extension to existing notions or as a special case to them. The criteria is orderliness of a design/theory. Here the choice is obvious - ESP32's input-only (GPI) pin is a special case of just a pin. Using it for output leads to undefined behavior (implementors are encouraged to throw an exception on attempt to configure such a pin as output).

> Reading input via pin.value() when the pin is set to OUT mode. As far as I know, most MCUs allow to read the input independently of the drive. Is this then a level0 feature?

Within bounds of classical Boolean logic, reading value of a pin set to output mode doesn't make sense - it has a value last set on it. If you need this value later, save it to variable and use it. Approaching it from the other side, the simplest way to implement switchable input/output pins (again, within bounds of a classical Boolean schematics) is to use multiplexer(s) in pin's electrical scheme which would exclusively connect either input-handling block, or output handling block. Making sure that they both active at the same time and operate correctly would almost certainly require more complex machinery (again, within bounds of Boolean logic). It may be a case that there're extra-boolean magic shortcuts, and maybe all modern MCUs use them. But someone making a normal boolean CPU (e.g. on their kitchen table) wouldn't use them.

So, approaching the matter from 2 different sides, it's clear that getting input value of output pin can't be a mandatory operation for level0, thus, its result is undefined on that level. (It may be defined on later levels, likely for particular types of pins, but the whole idea is to finalize definition of level0, then level1, 2 based solely on level0's features; discussing finalization of anything beyond that is very premature).

dhylands, 2016-11-22:
> Within bounds of classical Boolean logic, reading value of a pin set to output mode doesn't make sense - it has a value last set on it. If you need this value later, save it to variable and use it

I guess I don't agree with the "doesn't make sense". That's like saying if I set a variable, I can't read it back.

On many MCUs, I can read back the value I last set a pin to without having to introduce any extra saved state (i.e. bloat). 

If the API spec says that reading back the output state of a pin is undefined it forces the caller to introduce state (to save the value last written).

If the API spec says that reading back the output state of a pin is defined (at least when in output mode), it forces the implementation to save the state if the HW is incapable of doing so.

It would also be possible to word the API spec to say that reading back the output state of a pin is defined if the underlying MCU supports it.

So that leads to the question, which processors are capable of saving the state and which ones aren't.

So let's see which ones are or aren't capable of reading back the output state:
STM32 family - can read output state
cc3200 - can read output state
atsam series - can read output state
Freesacle K-series - can read output state
esp8266 - can read output state

Does anybody have an example of an MCU which can't read the output state?

dpgeorge, 2016-11-23:
@dhylands the discussion is about pin.value() reading the input value on the pin, regardless if the pin is in IN or OUT or OPEN_DRAIN mode.  pin.out_value() exists to read the value written to the output register.

dpgeorge, 2016-11-23:
> It's possible to define needed features in an easier way, e.g. "what is available within classical Boolean logic". So, values are 0 and 1, period. Given a Boolean expression z = f(x, y). x and y are inputs, z is output, period. The Level 0 featureset defined.

Wow, this is just making me even more unclear as to the scope of level 0.  Boolean logic can do anything so I don't see how it gives restrictions, or even guidance, on what features to choose.

> Apparently, basics of Boolean logic define the scope, and indeed, those would be available anywhere.

I don't see how it can define anything sensible to do with pin objects.  What are the arguments then against adding a method like `pin.and_(other_pin)` which takes 2 pins, reads their values, and returns the logical and of the values?

> It may be a case that there're extra-boolean magic shortcuts, and maybe all modern MCUs use them.

Yes, it seems all modern MCUs do, so we should expose this functionality as a basic feature.

> But someone making a normal boolean CPU (e.g. on their kitchen table) wouldn't use them.

All classical CPUs I know of run on Boolean logic and are "normal", and have complex features.  What is the point of this argument, why are we suddenly targeting CPUs made on kitchen tables?

pfalcon, 2016-11-23:
> Wow, this is just making me even more unclear as to the scope of level 0. Boolean logic can do anything so I don't see how it gives restrictions, or even guidance, on what features to choose.

Playing broken phone doesn't help here either. Someone can stay before white board and tell that it's black and there's no intra-argument way to overpersuade him. It's the common agreement on how that particular color on a board is called what allows to move forward. So, you asked me to answer your questions and I did. I can repeat my answer adding one or two words to address your concerns, but there will be a lot of room to misunderstand still. So, the *domain* of classical Boolean logic are: values 0 and 1, and input and outputs of logical blocks (== arguments and results of Boolean functions). Inputs are suddenly only inputs, you can't output anything into them, outputs are suddenly only outputs, you can't input anything from them. Proposition: let's fully define subset of Pin API which works within bounds of this domain, leaving anything else aside (because if we can't define so simple things, we really should not go beyond that).

> What are the arguments then against adding a method like pin.and_(other_pin) which takes 2 pins, reads their values, and returns the logical and of the values?

Did word "domain" help, or will you pick at other things now? "Boolean functions" in relation to MicroPython would be arbitrary code (algorithms) in Python. What's important is that these algorithms would take inputs, each valued as 0 or 1, and output only 0 or 1 to outputs. That's how classical Boolean logic works, no other values, no outputting to input an vice versa.

> What is the point of this argument, why are we suddenly targeting CPUs made on kitchen tables?

We target all CPUs. Most of modern MCUs are crippled and don't even expose address and data bus to allow to connect basic logical devices. Find an MCUs with external address and data buses. Such definitely still exist, though one practical way to get such is to sit at kitchen table, [like real men do it](http://history-computer.com/ModernComputer/Relays/Stibitz.html), fire up kitchen-table FPGA software like https://github.com/cliffordwolf/yosys (any other is just proprietary), and load up a normal CPU, not crippled, CPU into your FPGA. Then connect to it some classy stuff like 7477, using WR signal for strobe. That's your pins. Output pins. Now try to read them. There's simply nothing to read.

> What is the point of this argument

The point of the argument is very simple and obvious: to finalize the definition of hardware API. Any argument not leading in that direction apparently not helpful. And any argument "we need somthing more" is exactly such. We need to think  not what we can add, but when there's nothing to remove. When it's reached (the proverbial "level0"), we should define API for that set, and then start to build features on top of it. In a careful way, not adding new notions until we exhausted all possibilities offered by existing. The progression could be e.g.:

* Level 0: 0/1, input-vs-output pins.
* Level 1: Inversion, easy to understand concept, fully expressable with Level 1 notions.
* Level 2: SPI, fully expressable in Level 1 notions.
* Level 3: Next on the plate I2C, but Level 0-2 notions aren't enough to implement it. There're descriptions of I2C in terms of "open drain" pins, but that's too complicated to bring in right away, a simpler concept of "high impedance" is enough to implement I2C signalling. And here we'll employ physical property that input pin is in "high impedance" (i.e. kind of disconnected) state for someone try to read it (this property (it's desirability) comes from analog domain). That allows us to still stay within bounds of Level 0, except we add 2 assumptions: a) it's possible to switch input/output mode for pins dynamically; b) input pin is in "high impedance" state. These assumptions are true for many modern CPUs/MCUs, but if any of them isn't, I2C implementation won't work. Caveat emptor, I2C is extra-boolean protocol (unlike SPI).
* Level 4-m: Other protocols expressable within Level 0, maybe with assumptions.
* Level m: Only here comes the concept of "open drain" pin. Some MCUs support it natively, for others it can be emulated using "high impedance", i.e. input, state above. 
* Level n: Here come implementations of SPI, I2C, etc. protocols using vendor blocks. These "blocks" include both hardware device blocks in MCU and software APIs of underlying OSes. As any vendor hardware/software is dumber (but faster) than reference implementation of Levels 2, 3, its API will be subset of API of Level 2, 3, and defining this subset is a separate task.
* Level q: Only here comes features like hardware UARTs where RX is open-drain and TX is inverted and other "exciting new" stuff from hardware vendor press releases.

All these levels adhere to the law of diminishing returns: Level 0-1 allow to do 50% of what's interesting to do (I'd personally think that more, but let's stay conservative at merely 50%). Adding Levels 2-3 brings it to 75%. Then each next level brings only few percents more in what can be done and what userbase is covered by those features. For example, hardware SPI driver for STM32 is of ~zero use for users of other  MCUs, and there're hundreds of MCU families. That doesn't mean support for them should not be implemented, only that it can't be implemented for them all, so users of a particular MCU [... cut ...]
What's important for me is the idea "MicroPython runs everywhere". That's why I'm interested in Levels 0-3, as they cover 75% of usecases. I don't worry much about higher levels - those levels are in good hands of people who need them (really need, up to spending their own time or money on them). What's interesting to me is that a lot of people who argue with me apparently not interested for MicroPython to run everywhere, they want to keep it limited to their own particular needs. That's the fundamental conflict of interest, as all my years of contribution to MicroPython were driven by the idea above. I'm saddened that we, @dpgeorge, came to disagreement on this fundamental point too.

dhylands, 2016-11-24:
> @dhylands the discussion is about pin.value() reading the input value on the pin, regardless if the pin is in IN or OUT or OPEN_DRAIN mode. pin.out_value() exists to read the value written to the output register.

IN or OUT seem to covered (the code can return out_value if the mode is OUT). The tricky one seems to be OPEN_DRAIN when out_value is 1. I think this boils down to MCUs which only have a single register for input and output.

Looking at the list of processors I showed before:
- STM32 can read input while in an output mode
- cc3200 - cannot read input while in an output mode (and doesn't support open-drain)
- atsam series - can read input while in an output mode
- Freesacle K-series - can read input while in an output mode
- esp8266 - can read input while in an output mode, but doesn't seem to support open-drain.

I think its still reasonable to say thay pin.value() will return the value of the input pin (while in a non-input mode) if the processor supports it, and undefined otherwise. If you want portable code, then you'd configure the pin as an input. If you want to take advantage of processor specific features then you can.

If the port is going to support open-drain by switching the pin from output to input when the output is 1, then the input value can be read.

dpgeorge, 2016-11-24:
@pfalcon thank you for taking the effort to explain your position in good detail, especially providing the overview of the different levels.  Honestly I did not know this was how you were thinking about the machine API (yes you did hint at this many times from various angles, but it was never spelt out like this).  When you mentioned levels at the start of this post I thought level 0-1 were basic HW entities like Pin, UART, I2C, SPI, ADC, RTC, WDT, Timer, etc (which do exist on most MCUs).  And then latter levels were more complex interactions between these, compositions, and advanced features (eg QuadEncoder).

When @danicampora and I did the first draft of the machine API, we were thinking at the level of MCU hardware and exposing the peripheral blocks that exist.  It says at the start of the HW API document that  "The main aim is to provide Python modules/functions/classes that abstract the hardware in a Pythonic way", and then "There should be a close mapping from functions to hardware".  You can see that I wrote those words in this version: https://github.com/micropython/micropython/wiki/Hardware-API/fb77a18828c2754f14ceb37f1bf644b7b2cfea00

From my point of view the machine module is a cleaned-up and more MCU agnostic version of the pyb module.  The machine module has a very simple goal of exposing the hardware capabilities so they can be accessed from Python.  That's why there is an "id" parameter in all the constructors, so you can specifically ask for a certain hardware block (or pin, etc).  The machine module does not (in my mind) have any grand plans of building up things from first principles.

@pfalcon I think what you outline above with the levels is a totally different approach (conceptually, in design, and in implementation) and if you want to follow through with it I think we need to start from scratch (and with a different module name).  Because otherwise the purity of your idea is going to get lost.

> What's important for me is the idea "MicroPython runs everywhere".

To me this phrase means that the uPy core runs wherever you have a C compiler.  MCUs are just one such place, and the machine module is just a way to get the most out of an MCU, with machine module limited in scope by the MCU hardware.  uPy can exist without the machine module and still run everywhere (eg in space).  uPy should not be defined by the machine module (although it is a nice and useful thing to have).

pfalcon, 2016-11-26:
> When @danicampora and I did the first draft of the machine API, we were thinking at the level of MCU hardware and exposing the peripheral blocks that exist.

Peripheral blocks exist to implement some functionality. And the functionality is what's important. Trying to think off "peripheral blocks" will lead to getting lost in thickets of silicon vendor marketing and "differentiation".

 > "The main aim is to provide Python modules/functions/classes that abstract the hardware in a Pythonic way". You can see that I wrote those words in this version: https://github.com/micropython/micropython/wiki/Hardware-API/fb77a18828c2754f14ceb37f1bf644b7b2cfea00

You apparently made a typo and it skipped a word "functionality" after "hardware". It can be easily fixed.

> @pfalcon I think what you outline above with the levels is a totally different approach (conceptually, in design, and in implementation) and if you want to follow through with it I think we need to [...]

And I don't agree that it's "totally different approach". But it's definitely an orderly, structured approach to design and implementation, vs largely spontaneous approach employed so far. And my approach doesn't change much in the existing machine module, it's the approach to *finally* *finalize* it. So, it's largely a methodological change, which calls to not starting more complex things until more simple are tackled well enough.

Moreover, that would be the natural way how someone would implement it - regardless of whether someone wants "latest tech", one would start with simple things like making an LED blink. That was how esp8266 port progressed, except you spontaneously implemented "open drain emulation" which broke progressive motion. But now it's the same with Zephyr port - third party who implemented it, implemented just input/output, 0/1 pins. Indeed, doing it differently requires more effort than doing it in a natural way. The fact that there's no finalized spec and implementations for a over a year, only scope creep, offtopic, and flame, shows that very well.

>>What's important for me is the idea "MicroPython runs everywhere".

>To me this phrase means that the uPy core runs wherever you have a C compiler.

And for me that:

* MicroPython runs on more boards than any other scripting languages for constrained systems.
* It's possible to run MicroPython in a useful way on any hardware without extraordinary efforts (whereas other scripting languages wouldn't run in a useful way, or would required extraordinary efforts to do so). This useful way can be simple enough, but pave the road for further interest in MicroPython.

pfalcon, 2017-01-31:
With basic Signal support merged, next step is deciding on its methods for asserting and deasserting a signal (beyond inherited .value() methods). Based on the previous discussion, following is proposed:

~~~
.on() - assert signal
.off() - deassert signal
~~~

It's also proposed to retrofit these methods to Pin class, where they replace previous provisional methods .high() and .low(). Thus, consistency between classes is achieved.

dpgeorge, 2017-02-01:
> With basic Signal support merged, next step is deciding on its methods for asserting and deasserting a signal ...

Adding .on/off to signal sounds good, and I guess to Pin as well to make the 2 classes more compatible (turning a Pin "on" is a sensible thing to do and is not ambiguous).  But I'd be include to keep high/low for Pin because they are conceptually different, you really want the pin to go to a specific logic level, and doing "pin.on()" may actually turn something off.

pfalcon, 2017-02-01:
> But I'd be include to keep high/low for Pin because they are conceptually different,

They are conceptually different, but we need to limit number of concepts by the Occam's blade. So, need to find one which covers the most usecases.

> you really want the pin to go to a specific logic level,

pin.value(level)

> and doing "pin.on()" may actually turn something off.

Then use Signal - that's exactly what it is for. Having consistent method will also allow to "easily" replace misused Pin with a proper Signal in 3rd-party code.

pfalcon, 2017-02-01:
P.S. We'd be discussing .value() vs `__call__()` situation next, based on the outcome of .on/.off vs .high/.low discussion.

dpgeorge, 2017-02-03:
> and doing "pin.on()" may actually turn something off.
> 
> Then use Signal - that's exactly what it is for.

In that case, pin shouldn't have on/off at all, so you are forced to use signal and not have any logical errors.

> P.S. We'd be discussing .value() vs __call__() situation next, based on the outcome of .on/.off vs .high/.low discussion.

I think we should discuss it all at once.  Without high/low I'd want to keep `__call__` as a convenient way of doing high/low.

pfalcon, 2017-02-04:
> In that case, pin shouldn't have on/off at all, so you are forced to use signal and not have any logical errors.

Yes, but nobody wants to "force" anybody, there's desire just to have good guidelines and the highest possibilities/bloat ration.

Let's look again: I'm a big proponent of high/low methods for Pin. Also, Signal should have on/off methods (naming not without issues per earlier discussion, but I already agreed to it). So, that would call to have 2 pairs of methods, which is a bit too much. Core API should have 1 feature (e.g. set by value), 2 features (e.g. assert or deassert), third is already too much. So, I, just as you with `value()` vs `__call__`, accept a compromise and say that it's ok to drop high/low and use just on/off.

 > I think we should discuss it all at once. Without high/low I'd want to keep __call__ as a convenient way of doing high/low.

"All at once" approach always lead to a vicious cycle of recursive scope creep, so we should stay wary of it. But ok, there're 2 possible outcomes.

1. We can't contain ourselves, and go with both high/low and on/off. But that would likely mean that we wouldn't be able to contain ourselves in the next question of value() vs `__call__()` either.
2. But that's not scalable approach at all. We should try to apply self-control in high/low vs on/off issue, and then *maybe* we would have moral right to relax in value() vs `__call__()` issue.

So, like that: we have 3 slots, and 4 contenders: high/low, on/off, value, `__call__`. IMHO, a fair formulation.

Extra points for exercising self-control diligently and using up just 2 slots. Note that I'm positive for keeping `__call__` ;-). So, there would be a need for good 3rd-party counter-arguments ;-).

peterhinch, 2017-02-04:
I'd suggest pins controlled only by high() or low() methods; signals only by on() or off(). This reflects physical reality and helps avoid logical errors. A motor or LED is on or off. If you're working at pin level, high or low is all you've got - it's your responsibility to understand the outcome and these names promote that.

Function call syntax on either accepts an optional bool and returns a bool meaning high for a pin and on for a signal.

A value() method brings nothing to the party and can be discarded.

dpgeorge, 2017-02-07:
> So, like that: we have 3 slots, and 4 contenders: high/low, on/off, value, __call__. IMHO, a fair formulation.

Yes that's a fair summary.  The virtues of these are:
- `__call__`: quick and easy to write for both reading and writing values.  Also efficient in terms of execution (there's no method lookup).  I use this form all the time at the REPL.
- value(): relatively easy to understand when reading code for the first time, especially for reads of a pin: `x = pin.value()`
- high/low: simple methods with obvious behaviour, very easy for beginner to understand and use, and makes for clean and readable code.
- on/off: same as for high/low but with slightly different semantics (as we already discussed).

Drawbacks:
- `__call__`: if you don't know what it does it's not obvious what it's doing; also slightly overloaded in that it can set and get the value.
- value(): verbose to write, if you know about `__call__` then you'd use that instead.
- high/low: 2 methods to set compared to just 1 method with call/value
- on/off: same as high/low

Unless we add something like read() we definitely need at least one of call/value.

> A value() method brings nothing to the party and can be discarded.

I tend to agree!  But would there be a lot of user backlash if we removed value()?  (Of course it could be kept for the time being for backwards compatibility reasons, but we'd tell people to use call instead.)

dpgeorge, 2017-02-08:
From comment https://github.com/micropython/micropython/pull/2834#issuecomment-277991529:

> We don't want [Signal] for I2C, SPI. time_pulse_us is a grey area. I initially though that yes, it should account for Signal, and created button_reaction.py sample. I now have reservations. 

Ok, so if there's no need to support arbitrary objects in I2C, SPI etc then there's no need for PinBase.

Similarly, if there's no need to support Signal in these clases/functions, then there's no need to make Signal behave like Pin (ie have the same methods).  Instead, as has been discussed, Signal should be disjoint/distinct from Pin, and should be used to implement a very simple on/off driver for a peripheral like an LED or button.

Thus, Signal can have just .on(), .off() and .is_on().  That's it.  No need for call/value/high/low.

Then Pin can stay as it is, with call/value, and we can add high/low to it.

pfalcon, 2017-02-12:
> But would there be a lot of user backlash if we removed value()?

Yes. value() is the *only* method which is needed, all other methods are conveniences. Like, if someone has space constraints, one can drop anything except for .value(). It was also likely the oldest thing in https://github.com/micropython/micropython/wiki/Hardware-API .

pfalcon, 2017-02-12:
> The virtues of these are: / Drawbacks:

Sorry, that's not what I meant with "So, there would be a need for good 3rd-party counter-arguments" above. Both you and I can provide a list of arguments pros and cons `__call__`, that doesn't count if we both feel like keeping it. There should be someone who strongly feels that having it is more wrong then removing something which existed for quite some time. As strongly as feel the need to develop small Python implementation, as strongly as the need for testsuite with almost complete coverage, as the need for OO VFS system, or that root filesystem should be mounted at root, etc.

Again, as we lack such party, here's my summary and addition to points:

* `__call__` is practical for interactive use and quick scripts.
* `__call__` is hugely impractical for maintenance of larger projects. There's simply nothing to search for when you want to see when Pin operations happen. "value" method name provides good search selectivity.

Consequently, proposal: for purely practical reasons (our current code implements internally value() in terms of `__call__`), keep `__call__`, but make it undocumented (remove documentation if already exists). In a year we can see how it goes and if we can part with it completely.

pfalcon, 2017-02-12:
> then there's no need for PinBase.

Sorry, how PinBase got into the requested discussion of on/off methods? We have a separate ticket on PinBase, let's keep its discussion there.

pfalcon, 2017-02-12:
> Similarly, if there's no need to support Signal in these classes/functions, then there's no need to make Signal behave like Pin (ie have the same methods). Instead, as has been discussed, Signal should be disjoint/distinct from Pin, and should be used to implement a very simple on/off driver for a peripheral like an LED or button.

This is NOT how Signal was conceived, designed and being implemented. The idea behind Signal is very clear, obvious, and everyone had a chance to see it born: this idea was that it would be nice to add optional inversion support to Pin. There were arguments against that in terms of efficiency, e.g. Pin can be a fully static ROMable singleton object, while Signal needs to keep RAM state (inversion status). So, it was split into a **subclass** of Pin, subclass in terms of OO design, not in the sense that issubclass(Signal, Pin), as common in uPy, we avoid that as it costs us bytes. So, Signal is ducktyping subclass of Pin, that's its definition. Consequently, it must have interface compatible with a Pin (and Pin should avoid having interface which would be truly confusing for Signal, like high/low).

That's it.

pfalcon, 2017-02-12:
> on/off: same as for high/low but with slightly different semantics (as we already discussed).

There's a *huge* semantic difference between subtracting small quantity from bigger and bigger from smaller. And yet any grown up can handle the difference between `3 - 2` and `2 - 3`. So, no worries, on/off even despite "slightly different semantics" fits well both Pin and Signal.

pfalcon, 2017-02-12:
on() and off() methods for Signal were implemented in https://github.com/micropython/micropython/commit/181f7d145002731e5baec0f2c43ac57818447f8b

I'm not sure about .is_on() - what's the difference between it and .value() (w/o args)? Just returning boolean?

[10 more replies]

closed 2017-05-28 by pfalcon