micropython/micropython #2605: esp8266 webrepl_setup crashes

deshipu, 2016-11-07:
```
MicroPython v1.8.5-122-g8f068e8 on 2016-11-07; ESP module with ESP8266
Type "help()" for more information.
>>> import webrepl_setup
WebREPL daemon auto-start status: disabled

Would you like to (E)nable or (D)isable it running on boot?
(Empty line to quit)
> E
Fatal exception 28(LoadProhibitedCause):
epc1=0x402432b2, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000633b0, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 32020, room 16 
tail 4
chksum 0x12
load 0x3ffe8000, len 1084, room 4 
tail 8
chksum 0x59
load 0x3ffe8440, len 3000, room 0 
tail 8
chksum 0xc3
csum 0xc3
l��|��s{oc��l�c�lc쌜���dc�l{lllܞ|��{{oc��ll��b�c쌜��c��l{d�l��|��{oc��l��c�c쌜��dc�lcd����o�;��g|�ddd$`��r�l�d�l`��s�l�d�l`��s�d���ll`{d��sl���#��c�bb{�rc��n�o�l��d�l��ll�Ä���l�g����cd$�rp���cd�c{d{d;�n�����g�߀�b{c��c쇜���d�lsd��ln�xsl�l��#4 ets_task(40100164, 3, 3fff8388, 4)
```

deshipu, 2016-11-07:
This is a fresh build from the master, flashed after erase_flash. It's reproducible every time on that board.

dpgeorge, 2016-11-07:
It works ok for me.  v1.8.5-122-g8f068e8 of uPy, erase_flash first, using commit e8d757b1a70a5cf19df0afe23a769739c6cff343 of esp-open-sdk.

pfalcon, 2016-11-07:
@deshipu : How did you flash it? Where esptool.py comes from?

pfalcon, 2016-11-07:
And to diagnose such exceptions, firmware.bin/elf/map should be attached to issue (please mention this information on a forum, etc. when you see such case).

deshipu, 2016-11-07:
I'm very sorry, it wasn't a clean board -- I had a `boot.py` which started webrepl in there. Here's how to reproduce the problem in a clean board:

```
>>> import webrepl
>>> webrepl.start(password="ziew")
WebREPL daemon started on ws://192.168.4.1:8266
Started webrepl in manual override mode
>>> import webrepl_setup
WebREPL daemon auto-start status: disabled

Would you like to (E)nable or (D)isable it running on boot?
(Empty line to quit)
> E
Fatal exception 28(LoadProhibitedCause):
epc1=0x402432b2, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000633b0, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)
```

deshipu, 2016-11-07:
Attaching the binaries:

[build.zip](https://github.com/micropython/micropython/files/574951/build.zip)

deshipu, 2016-11-07:
I'm using `esptool.py` from the master of https://github.com/themadinventor/esptool

pfalcon, 2016-11-07:
Ok, so you may want be interested in how to diagnose such issues (crashes with exception traces) further (for your own cases and to help other users). The first step is simple enough:

~~~
xtensa-lx106-elf-objdump -D firmware.elf | less
~~~

Then search for epc1 value from the exception (here 402432b2). Then see what function that instruction is in. Here it's mp_map_lookup(), so we know it's likely MicroPython issue, and not vendor blobs'.

pfalcon, 2016-11-07:
I can confirm that flashing provided .bin leads to exactly same CPU exception for me.

pfalcon, 2016-11-07:
Now I rebuild the same git revision (8f068e8) myself. I checked that at 0x402432b2 I has the same instructions as your .bin (which doesn't have to be true, why it's important to provide user's binaries). But - my build doesn't crash.

pfalcon, 2016-11-07:
Now, I have a bunch of cruft in my tree - my own test scripts in frozen modules, etc. So, I do a clean clone, and then build again. Surprisingly (sic!, don't count on that), I get the same binary size as yours: firmware-combined.bin . However, no wonders:

~~~
yours: 615d0f6aecfda8c7286dd8bdf27d53a8  firmware-combined.bin
mine: 8c0c699c3f99ab960d7ab05f1eb81898  firmware-combined.bin
~~~

That's why it, again, important that someone experiencing crash provided their own build, and all 3 of .bin/.elf/.map. We have all 3 too at http://micropython.org/download#esp8266 (but for some reason, now there's only one day is stored, @dpgeorge  ?).

pfalcon, 2016-11-07:
I flash my latest build, and there's no issue. Now 2 people confirmed there's no issue, so @deshipu , you have a broken/old toolchain.

dpgeorge, 2016-11-08:
I can't reproduce the latest build.  @deshipu you may have an outdated mpy-cross that needs recompiling.

> but for some reason, now there's only one day is stored

Should be fixed now.

deshipu, 2016-11-08:
Thank you. I will try it as soon as I can. For now I updated the SDK and can't get it to compile anymore.

deshipu, 2016-11-08:
I re-downloaded and re-built the whole SDK, re-compiled everything, including mpy-cross, but the problem still occurs.

puuu, 2016-11-09:
I can't reproduce this crash with latest build, too.

Only observed an error, if `boot.py` is missing, but this is another topic.

deshipu, 2016-11-09:
I tried again with a recent build, and the problem seems to have gone away. Could it be something cached that didn't get cleaned up with `make clean`?

closed 2016-11-09 by deshipu