micropython/micropython #2607: pyboard: accelerometer fails to update after UART write

peterhinch, 2016-11-07:
This arose from [this](http://forum.micropython.org/viewtopic.php?f=6&t=2578&p=15491#p15491) forum post. One user is seeing this occur without using the UART but I could not replicate this. However the following causes the accelerometer to fail to update on a Pyboard V1.0 and a Pyboard Lite AC.
```python
accel = pyb.Accel()
uart = pyb.UART(1, 9600)
print(accel.x(), accel.y(), accel.z())
uart.write( 'hello')
for _ in range(10): # Accel won't update
    pyb.delay(500)
    print(accel.x(), accel.y(), accel.z())
```
Outcome (board is moved around while loop executes)
```
MicroPython v1.8.1-505-g67d52d8 on 2016-09-26; PYBLITEv1.0 with STM32F411RE
Type "help()" for more information.
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== accel = pyb.Accel()
=== uart = pyb.UART(1, 9600)
=== print(accel.x(), accel.y(), accel.z())
=== uart.write( 'hello')
=== for _ in range(10):
===     pyb.delay(500)
===     print(accel.x(), accel.y(), accel.z())
=== 
-4 0 4
5
0 4 8
0 4 8
0 4 8
0 4 8
0 4 8
0 4 8
0 4 8
0 4 8
0 4 8
>>> 
```

dhylands, 2016-11-07:
This appears to be caused by a pin collision.

The Accelerometer is on I2C bus 1, which uses pins )B6 and PB7.
UART1 is on the same pins.

When UART1 switches the pin mux, then the accelerometer will wind up get i2c read failures (which the code currently ignores).

I believe that the data returned will just be whatever random data happens to be on the stack at the time:
https://github.com/micropython/micropython/blob/master/stmhal/accel.c#L140

dpgeorge, 2016-11-08:
Yes, this is a limitation of the board: if you use the internal accelerometer then that uses up pins X9/X10 (well, you could still have other I2C devices on this bus, that'll work).

The question is: should accelerometer raise an exception if the bus is not in I2C mode?

peterhinch, 2016-11-08:
Is this documented anywhere beyond the schematic? Perhaps a note on the [board layout](https://store.micropython.org/static/store/img/products/PYBv1_1-E.jpg) would be in order, as there is other information about the accelerometer there.

dpgeorge, 2016-11-08:
> Is this documented anywhere beyond the schematic?

I don't think so.  A note should be added to docs/libraries/pyb.Accel.rst, and probably the board pinout as well.

dpgeorge, 2018-12-06:
I have now added a note to the pyboard pinouts regarding the fact that X9/X10 will be in I2C mode when the accelerometer is in use.

closed 2018-12-06 by dpgeorge