micropython/micropython #2610: Change pyboard.py behavior when running a file and "-c" param.

pfalcon, 2016-11-07:
With a file like https://github.com/pfalcon/micropython-projs/blob/master/wifi_rssi_watch/wifi_rssi_watch.py, I'd like to run

~~~
pyboard.py --device /dev/ttyUSB0 wifi_rssi_watch.py -c 'main(b"foobar")'
~~~

And get the expected result. Currently it leads to:

~~~
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'main' is not defined
~~~

We should also promote pyboard.py's usage more widely.

dpgeorge, 2016-11-08:
Yes, I would find use for this, instead of constantly tweaking parameters in a script and saving it.

Feel free to submit a patch.  I'm not sure how to make switch arguments positional with argparse, and to allow more than one of them.

dpgeorge, 2016-12-15:
Part of the work required to support this was done in 29b58796136e3c9133cd268b163db1c53e67ac9b, where pyboard.py was refactored so that it doesn't do a soft-reset of the target between execution of scripts/commands that are passed in one go.  I.e. you can now run multiple files like `pyboard.py file1.py file2.py` and they are executed in the same REPL session.  Similarly, doing `pyboard.py -c "command" file.py` will execute the command and the file in the same REPL session.

But "-c" is an optional arg, not positional, so doing `pyboard.py file.py -c "command"` will still execute the command *before* running the file.  Despite its customisability, argparse does not support ordering of optional args.  Yes it can be hacked using custom actions, but it really needs to be hacked up a lot to get something simple like `pyboard.py file1.py -c "command" file2.py` working as expected.  And even then argparse will give confusing error/usage messages.  See eg http://stackoverflow.com/questions/5373474/multiple-positional-arguments-with-python-and-argparse

The only (?) clean way to support a general list of interspersed "-c" commands and filenames on the command line is to ditch argparse and write a custom argument parser.

jcw, 2020-06-07:
A few years later / time flies ... I'm taking the liberty of commenting anyway because this issue appears as the _only_ open one in a milestone titled "prio-high' (either it is, or it shouldn't be here?).

Can't this be solved with a combination of `if __name__ == '__main__'` and `sys.argv`, i.e. simply passing 'foobar' as argument to the script, without the `-c`?

dpgeorge, 2020-06-07:
> Can't this be solved with a combination of `if __name__ == '__main__'` and `sys.argv`, i.e. simply passing 'foobar' as argument to the script, without the `-c`?

I don't really understand what you mean by this...

The idea would be to get something like `pyboard.py file1.py -c "command" file2.py` working "as expected", which would be: execute file `file1.py` on the board, execute "command" on the board, execute `file2.py` on the board, in that order.

jcw, 2020-06-07:
Awww ... I completely missed the "pyboard" part. I read it as "python script [-c] arg".

Pls ignore my comment, it makes no sense at all.

dpgeorge, 2021-05-29:
Now that `mpremote` exists, it obviates the need for this feature.

`pyboard.py` will remain a relatively "lean" tool mainly as a low-level way to connect and control a MP instance.  `mpremote` will be promoted as the main user-facing tool.

The original request at the top of this issue can be achieved by:
```
$ mpremote u0 run wifi_rssi_watch.py exec 'main(b"foobar")'
```

closed 2021-05-29 by dpgeorge