micropython/micropython #2612: ESP8266 get pin id once set

spac3-monkey, 2016-11-08:
Is there a clean way to access the pin id having a `Pin` object?

```python
>>> from machine import Pin
>>> p = Pin(4)
>>> p.id()  # Something like this
4
```

Thanks

prusnak, 2016-11-08:
Internally this is stored in `phys_port` variable, but this seems not be exposed.

Very ugly workaround is to parse it from repr/str representation, where this variable is used:

```
>>> p = Pin(4)
>>> int(str(p)[4:-1])
4
```

spac3-monkey, 2016-11-08:
@prusnak that's what I'm doing, yes I saw `phys_port` var in code.

spac3-monkey, 2016-11-15:
I can't actually use this approach because I need to identify pins from an interrupt callback, and they can't allocate memory.

pfalcon, 2016-11-15:
@spac3-monkey : There's no problem with identifying pins and you don't need anything beyond what's already there:

~~~
pin1 = Pin(1)
pin2 = Pin(2)

if p is pin1:
    # this is pin1!!!
elif p is pin2:
    # this is pin2!!!
~~~

dpgeorge, 2016-11-16:
The hardware API specifies a pin.id() method which needs to be implemented for esp8266, and should do exactly what you need.

pfalcon, 2016-11-16:
> The hardware API specifies a pin.id() method which needs to be implemented for esp8266, and should do exactly what you need.

It's impossible to implement Pin.id() in a general case (consider Zephyr, consider a port with hundreds of GPIOs, so Pin object will be dynalloced, and will store register addr and mask, consider mapping implementations which will do `Pin("foobar")` which is actually `Pin((135, 100))`). So, it should be dropped from HW API spec. Entire HW API is made in such way that "IDs" are needed only for instantiation of objects and should not be needed afterwards. So, someone who needs them, first should elaborate why one needs them.

dpgeorge, 2016-11-16:
> It's impossible to implement Pin.id() in a general case

It's always possible: just store what the user passed in as the id.  If the pin object is statically allocated, then store the canonical id in the statically allocated object.

And I see that in zephyr, the pin print function does actually print the port and pin id, so it's there.

> So, it should be dropped from HW API spec.

We can't adopt this attitude, to drop everything that can't be done for all ports, otherwise we are left with almost no features.  Maybe pin.id() is not really needed and it can be dropped (we could go through an RFC for it...), but in general we need to have an understanding that some features/function/methods are good to have despite not being able to implement on every port.

pfalcon, 2016-11-16:
> It's always possible: just store what the user passed in as the id.

And that's impossible in a general case: storing such otherwise useless info may overflow memory, or at least cut it severely from useful user data. So it's against our (and users') interests to bet on that kind of solution.

> We can't adopt this attitude, to drop everything that can't be done for all ports, otherwise we are left with almost no features.  Maybe pin.id() is not really needed and it can be dropped (we could go through an RFC for it...), 

Please excuse me my colloquial brevity, you perfectly know that I'm the last one to drop things all around. "Drop" here meant "make it optional and stop worrying about it until we have implemented the core features". But core features aren't even defined! The more we should stop worrying about optional ones (== those which can't be implemented for each port) and concentrate on defining the core features.

> but in general we need to have an understanding that some features/function/methods are good to have despite not being able to implement on every port.

Sure. What we really need to understand are requirements for the HW API. And actually, "requirements for requirements" of HW API, i.e. define who are stakeholders for the HW API, and what are there interests in and expectations from it. It's clear there're 2 distinct groups with conflicts in interest, and until those groups and interests are pin-pointed, there's vicious circle of misunderstanding and lack of progress. I'll try to do that today, but I'm planning vacation from tomorrow, so maybe I won't have time.

pfalcon, 2017-04-05:
As discussed, id() isn't implemented because it's not possible to implement it in a general case, and to save space. An obvious workaround is shown. Also, any part can always subclass or wrap machine.Pin class and store any additional information required. Closing.

closed 2017-04-05 by pfalcon

AlexandreLarribau, 2023-03-13:
Hey ! I think I have the same issue and none of the answers has worked. 

I am listenning 5 pins. And I try to guess which has raised the interrupt. 

p1 = machine.Pin(1, machine.Pin.IN)
p2 = machine.Pin(2, machine.Pin.IN)
p4 = machine.Pin(4, machine.Pin.IN)
p5 = machine.Pin(5, machine.Pin.IN)
p6 = machine.Pin(6, machine.Pin.IN)

def handle_pwm_interrupt(pin):
  
    print("the pin is : ")
    print(pin)

p1.irq(trigger=machine.Pin.IRQ_FALLING | machine.Pin.IRQ_RISING, handler=handle_pwm_interrupt)
p2.irq(trigger=machine.Pin.IRQ_FALLING | machine.Pin.IRQ_RISING, handler=handle_pwm_interrupt)
p4.irq(trigger=machine.Pin.IRQ_FALLING | machine.Pin.IRQ_RISING, handler=handle_pwm_interrupt)
p5.irq(trigger=machine.Pin.IRQ_FALLING | machine.Pin.IRQ_RISING, handler=handle_pwm_interrupt)
p6.irq(trigger=machine.Pin.IRQ_FALLING | machine.Pin.IRQ_RISING, handler=handle_pwm_interrupt)

when I do this the display is "Pin(GPIO4, mode=IN)". 
I tried pin.name(), pin.id(), pin.id and each time i get :"File "<stdin>", line 16, in handle_pwm_interrupt
AttributeError: 'Pin' object has no attribute 'name'" 

And with id(pin) i get "268703464" which doesn't help me much to know which pin it is. 

Do you have advices ?

AlexandreLarribau, 2023-03-13:
> You could make a global list with your pins and compare `pin` in your IRQ with that list,

Something like this : "pwm_in_pins = [1, 2, 4, 5, 6]" ?

So I have to look for these number in "Pin(GPIO4, mode=IN)" ? Yeah this is a solution. 

Also I was thinking about making 5 different "handle_pwm_interrupt" fonction but I think there is a better way. I'll do this but if somenone has a better solution it would be nice !

robert-hh, 2023-03-13:
In the callback, pin IS the object which caused the IRQ. So you could prepare a lsut like:

`pin_list = [p1, p2, p4, p5, p6]`

And in the IRQ function:
```
global pin_list
pos = pin_list.index(pin)
```
pos tells, at which place in the list pin is. 

You can as well use `str(pin)` and look at the number of the GPIO, which is at pos 8.

AlexandreLarribau, 2023-03-13:
Okey, I tried pin_list = [p1, p2, p4, p5, p6] and got "Traceback (most recent call last):
  File "<stdin>", line 8, in <module>
NameError: name 'p1' isn't defined" 
Maybe I am doing it wrong... I don't understand... My imports are "from machine import Pin
import time". Thank you for your help

robert-hh, 2023-03-13:
You have to place the statement `pin_list = [p1, p2, p4, p5, p6] ` behind the lines, where you create p1, ... p6, and not into the callback:
```
p1 = machine.Pin(1, machine.Pin.IN)
p2 = machine.Pin(2, machine.Pin.IN)
p4 = machine.Pin(4, machine.Pin.IN)
p5 = machine.Pin(5, machine.Pin.IN)
p6 = machine.Pin(6, machine.Pin.IN)
pin_list = [p1, p2, p4, p5, p6]

def handle_pwm_interrupt(pin):
    global pin_list
    pos = pin_list.inded(pin)
    print("the pin index is : ", pos)

# More of your code to follow.
```

robert-hh, 2023-03-13:
But you can as well write in your callback:

def handle_pwm_interrupt(pin):
    global p1, p2, p4, p5, p6
    if pin is p1:
         # code for p1
    elif pin is p2:
        # code for p2
    # and so on

```

robert-hh, 2023-03-14:
There is a little hiccup in the code you have linked. The variable `pin_concerned` receives a value between 1 and 5. The index into the list must have a range from 0 to 4. Array indices in Python are 0-based. But maybe you have changed that already, but not the code at the web site.